tsduck/tsduck #1250: [BUG] Cross compilation tries to execute target bin

ankitp86, 2023-07-03:
#### Bug description:
Cross compile version v3.34-3197 fails due to build process trying to use target compiled bin on host PC; first cause of issue is tsxml. The host (where I'm doing the compilation) is x86, the target (where tsduck is intended to run) is arm7.

It seems that the build process first builds tools, then executes these tools to generate xml files. This would fine where host and target are the same arch. In a cross compilation environment, the tools intended to be executed on the host must be compiled using host sdk.

Note that I have been able to cross compile tsduck in similar way on older version (3.18-1236) successfully.

#### How to reproduce:
```
export SDK_ROOT=/home/ankit/sdks/sdk-1.2.1_gcc_6.x
export SDK_PREFIX=arm-unknown-linux-gnueabihf

make -j8                                        \
    CROSS_PREFIX=${SDK_ROOT}                    \
    CROSS_TARGET=${SDK_PREFIX}                  \
    TARGET_ARCH=armv7l MAIN_ARCH=arm            \
    NOTEST=1 NOCURL=1 NORIST=1 NOPCSC=1 NOSRT=1 \
    NOHIDES=1 NOVATEK=1 NODEKTEC=1 NOEDITLINE=1 \
    cross
```

#### Expected behavior:
The build process to succeed with Target binaries/library.

#### Errors and logs:
```
Activating Git LFS at user level
Activating Git LFS at repository level
  [DEP] base/xml/tsxmlUnknown.cpp
  [DEP] base/xml/tsxmlTweaks.cpp
  [DEP] base/xml/tsxmlText.cpp
...
  [GEN] tsduck.h
  [CXX] dtv/codec/tsAC3Attributes.cpp
  [CXX] crypto/tsAES.cpp
  [CXX] dtv/signalization/tsATSCMultipleString.cpp
...
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/objs-libtsduck/alltables.o
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/objs-libtsduck/alldescriptors.o
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/objs-libtsduck/allcharsets.o
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/objs-libtsduck/allplugins.o
  [AR] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/libtsduck.a
  [DTAPI] no DTAPI available
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/libtsduck.so
  [COPY] tsduck.hfbands.model.xml
  [COPY] tsduck.time.xml
  [COPY] tsduck.keytable.model.xml
...
  [GEN] tsduck.dtv.names
  [GEN] tsduck.dektec.names
  [DEP] tsplugin_tsrename.cpp
...
  [CXX] tsplugin_aes.cpp
  [CXX] tsplugin_bat.cpp
  [CXX] tsplugin_analyze.cpp
...
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsplugin_aes.so
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsplugin_analyze.so
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsplugin_bat.so
...
  [DEP] tstestecmg.cpp
  [DEP] tsswitch.cpp
  [DEP] tshides.cpp
...
  [CXX] tsanalyze.cpp
  [CXX] tsbitrate.cpp
  [CXX] tscharset.cpp
...
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsanalyze
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsbitrate
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tscharset
...
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tstestecmg
  [LD] /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsversion
  [GEN] tsduck.tables.model.xml
bash: line 1: /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsxml: cannot execute binary file: Exec format error
make[3]: *** [Makefile:67: /home/ankit/repos/tsduck/bin/release-arm-NB-ANKIT/tsduck.tables.model.xml] Error 126
make[2]: *** [Makefile:38: default] Error 2
make[1]: *** [Makefile:61: default] Error 2
make: *** [Makefile:90: cross] Error 2
```
Further more:
```
>file bin/release-arm-NB-ANKIT/tsxml
bin/release-arm-NB-ANKIT/tsxml: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 4.14.0, not stripped
```

#### Environment:
 - OS: Linux (Windows WSL)
 - OS version: Ubuntu 20.04
 - TSDuck full version:  v3.34-3197
 - Installation type: rebuilt binary, used as library
```
>uname -a
Linux NB-ANKIT 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

#### Additional information:

lelegard, 2023-07-04:
Hi @ankitp86 

I understand the issue. Cross-compiling TSDuck has always been experimental and never documented in [the build instructions](https://tsduck.io/doxy/building.html).

It has become impossible about a year ago, when the XML model files have been modularized. The utility tsxml became part of the build process, to rebuild the final XML model file for the signalization. The last part of the build process now uses the newly-built tsxml binary.

Since you are the first one to report the issue, I assume that nobody uses cross compilation any longer. I first setup the cross-compilation part of the Makefile to test TSDuck on an Arm-based Synology NAS but I later used a Raspberry Pi 3 for the 32-bit Arm compilation, using a native build. I haven't used the cross compilation for years.

If you have a Raspberry Pi with a 32-bit distro, I would recommend to use it for native build.

I have two options regarding cross-compilation in the makefiles:
1. Completely remove it, removing the illusion that it is possible when you look inside the makefile.
2. Force the usage of a natively installed tsxml during cross-compilation, meaning that a native TSDuck shall be installed first on the build system.

I would prefer option 1 since nobody use cross-compilation (except you, sorry about that).

Option 2 would be done on a best effort basis only, without guarantee, since I will never be ablle to test and support it. I no longer have any cross-compilation environment and I have no plan to install one any more.

MMachmerth, 2023-07-06:
I’m although using cross compilation feature of TSDuck. Therefore, I would prefer option 2: “Force the usage of a natively installed tsxml”. The cross compilation is integrated in a Yocto-System which handles the build of the native tsxml to and I needed an option to be flexible to set the directory where the native tsxml is located. 

pull request #1253 is a suggestion for option 2.

lelegard, 2023-07-06:
@ankitp86, @MMachmerth, OK, you win, let's fix support for cross compilation.

@MMachmerth, I keep the idea of NATIVEBINDIR, but not as you did it. First, I prefer to keep this within cross compilation and not use it in the general case. Second, I don't want to force a native build of TSDuck (it takes time) if a native TSDuck package is already installed.

I have made and pushed the corresponding fix. I tested it against a Linaro cross-compilation environment I quickly downloaded for that test, using arm-linux-gnueabihf target.

Let me know if this works for the two of you.

MMachmerth, 2023-07-06:
Hi Lelegard
Thanks for your changes. While testing it I dived into the YOCTO build process and recognised that I’m not using the cross compilation mechanism of TSDuck. Apologize for my wrong statement in my last post. The YOCTO system gives me a complete environment that handles the cross compilation and TSDuck is therefore build “native” in this environment. The only thing I need is the NATIVEBINDIR variable to get an executable tsxml while building. Is it possible to use this variable in the general case too? 
Thanks for your quick support. The TSDuck is a awesome project.

lelegard, 2023-07-06:
>  Is it possible to use this variable in the general case too?

Added in commit 8798a55

Please confirm that it works

ankitp86, 2023-07-07:
I can confirm the changes in  baaaff3 work for cross compilation. It is a two step process (as intended).

One small/minor request: when compiling for native target without providing a BINDIR, the build system generates a destination of `bin/release-{LOCAL-ARCH}-{HOST}`, is it possible for the `cross` target to search for `bin/release-{LOCAL-ARCH}-{HOST}` so that we don't even need to provide a NATIVEBINDIR? this would make the steps a little easier , ie `make && make cross`

MMachmerth, 2023-07-07:
I have tested changes in https://github.com/tsduck/tsduck/commit/8798a55aeaf9a1a7220e0ad12f10ece0bc17d101 now the NATIVEBINBIR work fine in the YOCTO-Build. Thanks a lot

ankitp86, 2023-07-10:
No problem regarding the default value of NATIVEBINDIR...

However relative paths don't quite work. Not really an issue for me as I can use absolute paths:

Following Works:
```
make -j8                                        \
    NOTEST=1 NOCURL=1 NORIST=1 NOPCSC=1 NOSRT=1 \
    NOHIDES=1 NOVATEK=1 NODEKTEC=1 NOEDITLINE=1 \
    NOGITHUB=1 NODTAPI=1                        \
    BINDIR=${PWD}/bin/native
export SDK_ROOT=/home/ankit/sdks/sdk-1.2.1_gcc_6.x
export SDK_PREFIX=arm-unknown-linux-gnueabihf
make -j8                                        \
    CROSS_PREFIX=${SDK_ROOT}                    \
    CROSS_TARGET=${SDK_PREFIX}                  \
    TARGET_ARCH=armv7l MAIN_ARCH=arm            \
    NOTEST=1 NOCURL=1 NORIST=1 NOPCSC=1 NOSRT=1 \
    NOHIDES=1 NOVATEK=1 NODEKTEC=1 NOEDITLINE=1 \
    NOGITHUB=1 NODTAPI=1                        \
    NATIVEBINDIR=${PWD}/bin/native              \
    BINDIR=${PWD}/bin/target                    \
    cross
make CROSS=true                             \
    CROSS_PREFIX=${SDK_ROOT}                    \
    CROSS_TARGET=${SDK_PREFIX}                  \
    TARGET_ARCH=armv7l MAIN_ARCH=arm            \
    NOTEST=1 NOCURL=1 NORIST=1 NOPCSC=1 NOSRT=1 \
    NOHIDES=1 NOVATEK=1 NODEKTEC=1 NOEDITLINE=1 \
    NOGITHUB=1 NODTAPI=1                        \
    NATIVEBINDIR=${PWD}/bin/native              \
    BINDIR=${PWD}/bin/target                    \
    SYSROOT=${PWD}/bin/install                  \
    install
```

but if change all the `${PWD}/bin/...` to `bin/...` it does not quite install in the correct directory...

lelegard, 2023-07-11:
Did you pull the repo as recommended?
In my post preceding yours, I mentioned that I fixed that problem.

ankitp86, 2023-07-11:
Yes, tried with hash a62d6c8

lelegard, 2023-07-12:
> but if change all the ${PWD}/bin/... to bin/... it does not quite install in the correct directory...

Could you describe precisely where it fails? Is this during the compilation, the definition of `BINDIR` and `NATIVEBINDIR`, the current issue? Or the installation, the definition of `SYSROOT`?

In the latter case, it has always behaved like that, you need to provide an absolute path.

closed 2023-12-10 by github-actions[bot]