tsduck/tsduck #1251: [Q] MPEInject usage

maceyldn, 2023-07-04:
looking to get some clarification on usage.

-P mpeinject -b 10000000 --max-queue 10000 -p 7030 -l 192.168.145.145 239.100.200.1:5001 239.199.200.2:5001 239.199.200.3:5001

I'm trying to pull 3 multicasts into pid 7030. They are approx 2Mbit each.. I see 239.100.200.1 but not the others and they are defiantly present on the interface (tcpdump proves it).

Also tried them on different port numbers to no avail!

Am i missing something?

Thanks
Andy

lelegard, 2023-07-05:
Hi @maceyldn 

It works the way you do it. On an Ubuntu 23.04 system:
~~~
$ tsp -b 2,000,000 \
      -I null \
      -P mpeinject 239.100.200.1:5001 239.199.200.2:5001 239.199.200.3:5001 -l 127.0.0.1 -p 1000 \
      -P mpe --log -p 1000 --dump-max 0
      -O drop

* mpe: PID 0x03E8 (1000), src: 127.0.0.1:49343, dest: 239.199.200.2:5001 (01:00:5E:47:C8:02), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:49343, dest: 239.199.200.2:5001 (01:00:5E:47:C8:02), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:56842, dest: 239.100.200.1:5001 (01:00:5E:64:C8:01), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:56842, dest: 239.100.200.1:5001 (01:00:5E:64:C8:01), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:36715, dest: 239.199.200.2:5001 (01:00:5E:47:C8:02), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:36715, dest: 239.199.200.2:5001 (01:00:5E:47:C8:02), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:54903, dest: 239.199.200.3:5001 (01:00:5E:47:C8:03), 1316 bytes, fragment: 0x0000
* mpe: PID 0x03E8 (1000), src: 127.0.0.1:54903, dest: 239.199.200.3:5001 (01:00:5E:47:C8:03), 1316 bytes, fragment: 0x0000
...
~~~
We see a mixture of the 3 addresses.

maceyldn, 2023-07-05:
Thanks lelegard, Could you explain the -b and and --max-queue settings,
Your example shows they ain't there, i am pushing multiple 2Mbp/s through. Is it something i need to have?

A

lelegard, 2023-07-05:
> Could you explain the -b and and --max-queue settings,

Seriously, did you read the documentation?

~~~
  -b value
  --buffer-size value
      Specify the UDP socket receive buffer size (socket option).

  --max-queue value
      Specify the maximum number of queued UDP datagrams before their insertion
      into the MPE stream. The default is 32.
~~~

These are tuning options. Just like any tuning option, there is no definitive answer. It depends on the load and responsiveness of your system, as well as the profile reception (bursts or not). This is the problem of blind copy/paste from someone else. The fact that one guy, one day, on one system, in one networking environment, needed to use some tuning options does not mean that you or anyone else need the same tuning options.