tsduck/tsduck #1257: [Q] Your help is needed UDP input

zjaffesyna, 2023-07-12:
Hello
i have a server that get multicast TS, and filter out some PIDs from the stream, then sends me just a few PIDs of the TS but with UDP.
i want to play the stream, (and add null packets to fillout the BW.

and here is what i do:
`tsp -v -I ip 10.66.168.1:9981 -O dektec -d 0 --frequency 1528000000 --symbol-rate=13000000 -r 8/9 --modulation=DVB-S2-QPSK
`
but then i get an error message saying 
*** Error: ip: address 10.66.168.1:9981 is not multicast**

please advice
thank you
Z

lelegard, 2023-07-12:
Hi @zjaffesyna 

Well, the message is pretty clear, the specified address is not a multicast one.

When you _receive_ UDP datagrams, either you received unicast for your system, in which case you specify just a port (as in `tsp -I ip 9981`), or you join a multicast group, in which case you specify the address of the multicast group and the port (as in `tsp -I ip 224.1.2.3:9981`). In the latter case, the IP address must be a multicast address, of course.

zjaffesyna, 2023-07-12:
Thank you.
But here is my challenge,
The server (Linux) , uses netcat and listens , so my client needs to connect to the server and get the TS using UDP.
Is there a way to do that?

zjaffesyna, 2023-07-13:
> Sorry, but this is unclear. You provide no information on which system sends to where, on which address, unicast or multicast, using which protocol, etc.
>
> Without a precise description of your configuration and what you want to achieve, we can't help you.
>
> -

I'm sorry, I will explain
In place X, I have a Linux server that gets multicast TS.
This server filters some PIDs from the TS, and uses netcat (nc) to transmit the new edited TS using UDP. Here is the command runs on the server:

./rx_multicast | ./keepack_pids $PIDS | mbuffer -m 10M -T /run/user/1000/tmpfile -P 10 | nc -lk 9981

In the client (Windows), I want to the new TS file (using UDP), and fill the TS BW with null packets, so to tune to this TS will remain with same parameters.

Explanation about the Linux command:
./rx_multicast | ./keepack_pids $PIDS | mbuffer -m 10M -T /run/user/1000/tmpfile -P 10 | nc -lk 9981
We get the MC | we take only the relevant pids for us | we create a 10M buffer file | we netcat to port 9981.

thanks for your kind willing to help