tsduck/tsduck #1259: [Q]

iltomma, 2023-07-14:
HI,
I was able to read a satellite multistream with windows. I'll switch to Linux as soon as I can, but in the meantime I'd like to read the bitrate of the service.
With this string (`tsp -v -I dvb -a 1 -f 12466000000 --polarity vertical --pilots off --roll-off 0.2 --delivery-system DVB-S2 -m 16-APSK -s 41950 --fec- inner 2/3 --pls-mode GOLD --pls-code 262140 --isi 12 -P until --seconds 1 -P analyze --json -O drop`) I get the report I need, but the bitrate value it is expressed in packets (I assume) and returns me about 6283
I multiplied the value by 188 (6283x188=1181204), but it doesn't coincide with what I find by analyzing the TS recorded with the command tsbitrate -v "`C:\TSDuck\Record\test.ts`" where the value is 1,435,137.25. 
Is it otherwise possible to concatenate the two commands without having to record the TS and then analyze it? 
Or how can I calculate the bitrate from the report that comes out of the tsp command?
I need to automate it.
Thank you very much and thank you

AlfonZ42, 2023-07-14:
Displayed bitrate values should be in bits/second.

I believe 1 second is too small to yield usable values. In my experiments on a captured file (of DVB-T service though), bitrate value stabilized after 5s (1s: *5073073*, 2s: *5506208*, 5s: *5503173*). For comparison, `tsbitrate -v` says *4,275,846.52*, `tsbitrate -v -a` *5,503,172.61*.

As for the value *6283*, I would double-check whether that is indeed the bitrate of the service/ts, not of a single PID (e.g. the first encountered bitrate in the json).

iltomma, 2023-07-14:
hi,
I tried to increase the time to 2 minutes, but the bitrate is still very different from the one detected by tsreader or by analyzing the file with the tsbitrate command. 
`tsp -v -I dvb -a 1 -f 12466000000 --polarity vertical --pilots off --roll-off 0.35 --delivery-system DVB-S2 -m 16-APSK -s 41950 --fec-inner 2/3 --pls-mode GOLD --pls-code 262140 --isi 12  -P until --seconds 120 -P analyze -O drop`
Here is the first part of the report of this command:
```
===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ....... 6369 (0x18E1)  |  Services: ............. 10  |
|  Bytes: ....................... 381,504,828  |  PID's: Total: ......... 35  |
|  TS packets: .................... 2,029,281  |         Clear: ......... 35  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 8  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ............................ 110,631 b/s      120,046 b/s  |
|  Estimated based on PCR's: ............... 24,882,384 b/s   27,000,034 b/s  |
|  Selected reference bitrate: ................ 110,631 b/s      120,046 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: .............................. 27587 sec (459 min 47 sec)  |
|  First TDT UTC time stamp: ........................... 2023/07/14 17:13:12  |
|  Last TDT UTC time stamp: ............................ 2023/07/14 17:15:13  |
|  First TOT local time stamp: ......................... 2023/07/14 19:13:11  |
|  Last TOT local time stamp: .......................... 2023/07/14 19:15:13  |
|  TOT country code: ................................................... ITA  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0709  RETE 7 HD .................................... C        6,259 b/s  |
|  0x070B  CANALE ITALIA ................................ C        6,684 b/s  |
|  0x070C  (unknown) .................................... C          Unknown  |
|  0x070E  TELECITY ..................................... C        6,573 b/s  |
|  0x070F  TV d'A ....................................... C        6,682 b/s  |
|  0x0712  TELECOLORVDA ................................. C        6,282 b/s  |
|  0x0713  France 24 HD ................................. C       13,365 b/s  |
|  0x0714  RTS Un HD .................................... C       13,365 b/s  |
|  0x0715  TV5 MONDE EUROPE ............................. C       13,365 b/s  |
|  0x071F  CANALE ITALIA ................................ C        6,684 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================
```
where you can see the values of all the channels and also the screenshot of the sat channel viewing program where you can see the real bitrate
[image]
and 1,21Mbps is very different from "TELECOLORVDA ................................. C        6,282 b/s  "

AlfonZ42, 2023-07-14:
It looks like services' bitrates are calculated using broadcast time of *459 min 47 sec*, not *2min* as I would expect. I'll leave explanation/further investigation on lelegard.

In the interim as a workaround, *tsbitrate* does accept standard input, so piping the commands could work: `tsp -v -I dvb ... -P until --seconds 15 -P zap TELECOLORVDA | tsbitrate -v --all`. 
I would also try `tsp -v -I dvb ... -P until --seconds 15 | tsanalyze --json` to test whether separating the analysis from the capture helps.

lelegard, 2023-07-18:
Hi,

The bitrate based on PCR's is around 24 Mb/s and is probably correct.

The "user specified" bitrate is an external value. This can be an explicit `--bitrate` in the `tsp` command or some physical evaluation from the input plugin. Some input plugins have no idea about the bitrate, such as `file` and never report a bitrate. Some other plugins have a supposedly valid value of the bitrate. This is the case of the `dvb`  plugin which computes the bitrate from the modulation parameters. The "selected" bitrate is the value which is considered the most reliable one between the value from the PCR's and the value from the input plugin, if there is one. In the case of the `dvb` input plugin, its value is considered more reliable because there is only one possible bitrate for a set of modulation parameters.

However, in the case of DVB-S2 and a multistream carrier, the bitrate computation is wrong. I will fix this later, meaning that the input plugin should say "no idea" when requested to evaluate the input bitrate.

iltomma, 2023-07-31:
> It looks like services' bitrates are calculated using broadcast time of _459 min 47 sec_, not _2min_ as I would expect. I'll leave explanation/further investigation on lelegard.
>
> In the interim as a [...]

Hi, sorry for the delay, I was out of office. I tried with command pipe and it works like a charm!!! I can read correct bitrate! Thankyou

closed 2023-12-31 by github-actions[bot]