tsduck/tsduck #1260: [Q] Please tag git branch with v3.35-3248?

zerodefect, 2023-07-15:
I noticed that the CHANGELOG.txt in the repo indicates there is a version 3.35-3248.

I don't see the corresponding version as a tagged release in the repo. There is a fix in the release notes which has my interest.

lelegard, 2023-07-15:
Hi @zerodefect 

Only major releases are tagged and pre-built binaries are made available for each of them. 

The last number after the dash is a commit number which is automatically generated during commit or merge in the master branch. Tagging all of them would mean tagging each comment, which is a nonsense. This is [explained here](https://tsduck.io/download/prerelease/). 

If you are interested in a recent fix or feature, just pull the current state of the repo, it contains everything by definition.

zerodefect, 2023-07-16:
Thanks for your response.

closed 2023-07-16 by zerodefect