tsduck/tsduck #1261: [Q] Getting notified/call callback at certain PTS

jxxp9, 2023-07-15:
Hi,
how can I get notified at a certain PTS to run a custom action? Can I get some sort of signal? (Possibly not using the C++ interface)

Thanks!

AlfonZ42, 2023-07-16:
Hello.

My naïve idea is to use *pcrextract* plugin to get all PTS values of a target PID, then keep parsing the output and do stuff when the conditions are right. 
I'm not sure how efficient that is though / whether there is a better way to do this.

jxxp9, 2023-07-19:
@AlfonZ42 So I tried `tsp -I dvb --uhf 38 -P pcrextract --pid 1159 --pts -g -O drop`
I get this:
```
PID,Packet index in TS,Packet index in PID,Type,Count in PID,Value,Value offset in PID,Offset from PCR
1159,3675,21,PTS,1,6240352553,0,
1159,6555,38,PTS,2,6240369833,17280,
1159,9373,55,PTS,3,6240387113,34560,
1159,12254,72,PTS,4,6240404393,51840,
1159,15272,89,PTS,5,6240421673,69120,
1159,18079,106,PTS,6,6240438953,86400,
1159,20837,123,PTS,7,6240456233,103680,
1159,23778,140,PTS,8,6240473513,120960,
1159,26601,157,PTS,9,6240490793,138240,
1159,29409,174,PTS,10,6240508073,155520,
1159,32363,191,PTS,11,6240525353,172800,
1159,35184,208,PTS,12,6240542633,190080,
1159,37990,225,PTS,13,6240559913,207360,
1159,40939,242,PTS,14,6240577193,224640,
1159,43692,259,PTS,15,6240594473,241920,
1159,46638,276,PTS,16,6240611753,259200,
1159,49449,293,PTS,17,6240629033,276480,
1159,52334,310,PTS,18,6240646313,293760,
1159,55147,327,PTS,19,6240663593,311040,
1159,58026,344,PTS,20,6240680873,328320,
1159,60852,361,PTS,21,6240698153,345600,
1159,63856,378,PTS,22,6240715433,362880,
1159,66551,395,PTS,23,6240732713,380160,
...
```

I've specified I only want the PTS (`--pts`) but it returs lots of stuff. Is it normal?
Which is the PTS column?
Thanks!

AlfonZ42, 2023-07-20:
The first line is the header with description of that column. The value is the 6th column. 
The output is every packet that contains a PTS value. 

I've made a thingy which parses the output and writes a line every second (configurable):
```python
#!/usr/bin/env python3

def formatpts(pts):
    s1 = int(pts//90000)
    s = s1 % 60
    m1 = s1 // 60
    m = m1 % 60
    h = m1 // 60
    return f"{h:d}:{m:02d}:{s:02d}"

time_wrap = 1<<33 # wrap-around time of timestamps
time_diff = 1*90000 # trigger each 1s

# read and validate header line
header_line = input()
header_col = header_line.split(',')[5]
if header_col != 'Value':
    print(f"Header line column 6 expected to be 'Value', got '{header_col}'.")
    exit(1)

time = -1
while True:
    line = input()
    value = int(line.split(',')[5])
    if time < 0:
        time = value
    elif value > time:
        print(f"Trigger at {value} ({formatpts(value)})")
        time += time_diff
        if time > time_wrap:
            time -= time_wrap
```

For experiments with it, `tsp -I dvb --uhf 38 -P pcrextract --pid 1159 --pts -g -O drop 2>&1 | python thingy.py` should suffice. 
In the end, since you are processing the stream with *tables* plugin, forking the stream might be needed as well:
`tsp -I dvb --uhf 38 -P fork 'sh -c "tsp -r -P pcrextract --pid 1159 --pts -g -O drop 2>&1 | python thingy.py"' -P tables ...`

lelegard, 2023-07-20:
Hi @jxxp9 

> I've specified I only want the PTS (--pts) but it returns lots of stuff. Is it normal?

This plugin reports _time data_ (PCR, OPCR, PTS, DTS) and corresponding _metadata_ (PID, position in the stream, etc). 

The option `--pts` means that you need only the PTS _data_. The _metadata_ (what you call "lots of stuff") are always there. 

> Which is the PTS column?

Look at the header line.

closed 2024-01-21 by github-actions[bot]