tsduck/tsduck #1262: How to Rename the Output Bin Files

M-Bala-2612, 2023-07-17:
```
   https://github.com/tsduck/tsduck/archive/refs/tags/v3.32-2983.tar.gz
   cd tsduck-3.32-2983
   scripts/install-prerequisites.sh
   make installer
   sudo dpkg -i installers/tsduck_3.34-3197.ubuntu20_amd64.deb
```
   
@lelegard 
I was Compile the TSDuck Using this Commands , 
How to Change the Output Bin Files name. which directory file was output bin name allocated files
Like tsp name change to StreamingBin

Thankyou!!!

lelegard, 2023-07-17:
A TSDuck installation contains many files, binaries and configuration files, in a well-defined organization. 

You can change the installation root during `make install` but the file names and organizations remain the same. 

You use `make installer` which creates a standard package, rpm or deb, which is even more "standard". 

If you want to customize the file names of the binaries, feel free to try but there is no supported way and there won't be any in the future. 

TSDuck is open source and you are free to modify anything (within the liberal limits of the BSD-2 license). But this becomes your project and if you break something, you won't have support to fix it.

M-Bala-2612, 2023-07-17:
Okey @lelegard .
I was Changed some bin files name manually, 
But Changed bin file was not worked , any other way to Change the Bin file name in the Source Code

M-Bala-2612, 2023-07-18:
Okey @lelegard .

I was Changed some bin files name manually,
But Changed bin file was not worked , any other way to Change the Bin file name in the Source Code.

which directory file was Output Bin name allocated 

Thankyou !!!

udhayNuk, 2023-07-19:
@M-Bala-2612 
its your wish dont ask any more fucking questions.

M-Bala-2612, 2023-07-19:
@lelegard 
Its Possible to Combine the all Bin file to Single Bin file ?

Thankyou !

closed 2023-07-25 by M-Bala-2612