tsduck/tsduck #1263: HOW TO STORE THE UDP CONTENT WHEN STREAMING ONE UDP TO ANOTHER UDP?

udhayNuk, 2023-07-19:
@lelegard 

tsp -v -m --synchronous-log -I ip 224.2.2.1:5000 -P regulate --pcr-synchronous -O ip -l 127.0.0.1 235.1.1.1:1234 

this is my udp to udp conversion.

i want to store the output udp content in STORE.TS in ts file or any other idea please give me.

if its possible means tell me the how to store when streaming udp to udp content.

thank you in advance.

M-Bala-2612, 2023-07-19:
Are you crazy @udhayNuk 
Read the User guide .

AlfonZ42, 2023-07-19:
Hello.

Does *file* plugin not work in your case, i.e. `... -P file STORE.TS ...`?

udhayNuk, 2023-07-19:
@AlfonZ42 
i have already this method.
but i want any another method.

AlfonZ42, 2023-07-19:
I'm not sure what you mean.

Are you saying that you have tried the *file* plugin, it worked, but you want to do one thing via two different methods? 

Or are you saying you have already tried the *file* plugin but it doesn't work for what you wan to achieve? In which case could you please describe what is not working for your case?

udhayNuk, 2023-07-19:
@AlfonZ42 
i have tried it has been working i want another method.

lelegard, 2023-07-19:
The `file` plugin, as recommended by @AlfonZ42, is _the_ method. 

That being said, TSDuck is open source and you may invent and implement any other method for your own usage.

closed 2023-07-26 by udhayNuk

M-Bala-2612, 2023-07-26:
Fuck your Question @udhayNuk  ...
Read the User guide