tsduck/tsduck #1264: How to scramble the udp to udp when conversion using partial scrambling without using service id and pids[Q]

udhayNuk, 2023-07-19:
tsp -v -m --synchronous-log -I ip 224.2.2.1:5000 -P scrambler --cw 0000000000000001 --partial-scrambling 7 -O ip -l 10.10.10.10 235.1.104.55:1234
 this is my script.
How to scramble the udp to udp when conversion using partial scrambling without using service id and pids.
give me the idea to solve\
Thanking you in advance

lelegard, 2023-07-19:
If you specify neither a service nor pids, what do you expect the plugin to do?

Please RTFM. 

> If no service is specified, a list of PID's to scramble must be provided using --pid options. 

That's all.

udhayNuk, 2023-07-19:
@lelegard 
tsp -v -m --synchronous-log -I ip 224.2.2.1:5000 -P scrambler --cw 0000000000000001 --partial-scrambling 7 101 -O ip -l 10.10.10.10 235.1.104.55:1234 
when iam using the scrabling using the service id or pids, i runned the streams udp to udp using partial scrambling with service id after few days the udp service id or pids changed by provider my streams get stopped.
how to overcome from this to scramble get the default service id or pids to scramble.

udhayNuk, 2023-07-22:
@lelegard 
how to scramble the udp without mentioning the service id or name.
with getting the default service id or name automatically to scramble the udp.
please help me.

lelegard, 2023-07-22:
There is no way to do this currently. 

I recommend to write a script. In the first phase, use `-P tables -p 0 -x 1` and either `--xml` or `--json` (select your favorite format). The initial command extracts the PAT and terminates. Use your favorite XML or JSON analyzer to get the id of the unique service. Then start the scrambling command with that service id. 

TSDuck was designed to be scripting-friendly. So, take advantage of this.

udhayNuk, 2023-08-11:
> 

@lelegard 
thank you

closed 2024-01-14 by github-actions[bot]