tsduck/tsduck #1268: Is too hard to find so many request[Q]

zouji2012, 2023-07-21:
[root@localhost soft]# rpm -Uvh tsduck-devel-3.34-3197.el9.x86_64.rpm
error: Failed dependencies:
        libcurl-devel is needed by tsduck-devel-3.34-3197.el9.x86_64
        libedit-devel is needed by tsduck-devel-3.34-3197.el9.x86_64
        pcsc-lite-devel is needed by tsduck-devel-3.34-3197.el9.x86_64
        pkgconfig(libcurl) is needed by tsduck-devel-3.34-3197.el9.x86_64
        pkgconfig(libedit) is needed by tsduck-devel-3.34-3197.el9.x86_64
        pkgconfig(libpcsclite) is needed by tsduck-devel-3.34-3197.el9.x86_64
        pkgconfig(libusb-1.0) is needed by tsduck-devel-3.34-3197.el9.x86_64
        pkgconfig(srt) is needed by tsduck-devel-3.34-3197.el9.x86_64
        rpmlib(PayloadIsZstd) <= 5.4.18-1 is needed by tsduck-devel-3.34-3197.el9.x86_64
        srt-devel is needed by tsduck-devel-3.34-3197.el9.x86_64
        tsduck = 3.34-3197.el9 is needed by tsduck-devel-3.34-3197.el9.x86_64

When i m install tsduck  So many request need 。。
anyone canhelp package them 。
try my best can not find all.

lelegard, 2023-07-21:
Hi @zouji2012 

First, you must have noticed that the package file name contains `.el9`. So, the fist question is: do you use Red Hat Enterprise Linux 9 or one of its clones ? 

If the answer is no, then this package cannot be installed. 

If the answer is yes, then simply `dnf install` all requested packages. Except the `tsduck` package of course which must be manually installed.

zouji2012, 2023-07-24:
> Hi @zouji2012
>
> First, you must have noticed that the package file name contains `.el9`. So, the fist question is: do you use Red Hat Enterprise Linux 9 or one of its clones ?
>
> If the answer is no, [...]

Thank you very much for your reply 。。

[image]
I tried Centos 9 ,some of the request installed ,  still  can not  use DNF to install these .
i also tey to search in the internet  ,, even after i install  pcsc-lite-devel-1.9.1-4.el9.aarch64.rpm
[image]
got more issue ,   
i am noob
do you have some suggention to me ?
thank you!

lelegard, 2023-07-24:
Run this:
```
dnf install libcurl-devel libedit-devel pcsc-lite-devel srt-devel
```

And then use rpm to install TSDuck and tsduck-devel

zouji2012, 2023-07-24:
[image]
maybe i need check my repo ？

zouji2012, 2023-07-24:
[root@localhost soft]# rpm -Uvh tsduck-devel-3.34-3197.el9.x86_64.rpm
error: Failed dependencies:
        tsduck = 3.34-3197.el9 is needed by tsduck-devel-3.34-3197.el9.x86_64
  finally i found all the request ，but new alarm 。。
[image]

zouji2012, 2023-07-24:
> Run this:
> 
> ```
> dnf install libcurl-devel libedit-devel pcsc-lite-devel srt-devel
> ```
> 
> And then use rpm to install TSDuck and tsduck-devel

thank you again i have finished install all the  request !!

closed 2023-08-02 by zouji2012