tsduck/tsduck #1269: [Q] TSP capturing 2 streams at the same time (getting distortion)

claudinei-evertz, 2023-07-25:
Hi there,

I am trying to use the TSDUCK TSP command to capture two IPTS streams at the same time. However, doing this, I am getting distortion and all sorts of imperfections on the captures.

**Example of the capture:**
[image]

**What is the goal?**
I have a system that is generating 2 identical IPTS streams. The goal is to make sure they are identical because I am testing an `SMPTE 2022-7` implementation. Later I will dump the packets to analyze it.

**How I am doing it:**
I am sending both IPTS streams to the server where I have TSDUCK installed. Each stream is using a different port.

ORIGIN SERVER:
Source IP: `192.168.1.110`
IPTS Sink:
```
    udp://192.168.1.252:1234
    udp://192.168.1.252:1235
```
On the TSDUCK Server, I am running the captures like this:
```
tsp -I ip :1234 --source 192.168.1.110 --receive-timeout 30 -P until --seconds 25 -O file /home/tsduck_test.ts &
tsp -I ip :1235 --source 192.168.1.110 --receive-timeout 30 -P until --seconds 25 -O file /home/tsduck_test_2.ts &
```

Note: I tried with "-v" but there is not much to see from this tsp command.

The source is streaming 1080p/25, however, I tried with 720, 1080i, 4k, and multiple frame rates.

If I capture one at a time (while receiving multiple streams on the ethernet port) no problem is identified in the `.ts` file. The problem happens when I try to capture multiple streams at the same time.

**Environment:**
I have two installations of TSDUCK (the Raspberry Pi was to make sure we didn't have any problem with the VMware server)
```
ESXi VMware:
    ESXi version: 6.5.0 build-17477841 (update 3)
    Server specs: 20CPUs@2.8GHz , 6.6TB disk, 384GB memory
        TSDUCK VM: 4 vCPUs, 300GB disk, 32GB memory
        TSDUCK VM OS: Ubuntu 22.10
        TSDUCK VERSION: 3.34-3197
	TSDUCK VM NETWORK ADAPTER:
            description: Ethernet interface
            product: VMXNET3 Ethernet Controller
            vendor: VMware
            size: 10Gbit/s
            capacity: 10Gbit/s

Raspberry Pi:
    Model: Raspberry Pi 4 Model B Rev 1.1
    OS: Debian GNU/Linux 11 (bullseye)
    TSDUCK version: 3.34-3197
    Network Adapter:
       size: 1Gbit/s
       capacity: 1Gbit/s
```
I've checked and CPU, Memory, Network, and Disks are not a concern during the captions. Also checked the TSDUCK documentation and couldn't find a better way to issue the TSP command.
Finally, I also tested with two separate origins (coming from two different servers) and the problem persists.

**Question:**
Am I doing something wrong with the way I am trying to capture the IPTS streams?
Is there a better way to achieve it?
Did anyone encounter the same problem trying to capture more than one IPTS stream at the same time?

Thank you and I appreciate any help in this endeavor.

lelegard, 2023-07-25:
Hi @claudinei-evertz 

The reception commands are fine. I see no problem in principle. 

Do you have the same dual reception problem on the Ubuntu VM and the Raspberry Pi?

claudinei-evertz, 2023-07-25:
Thanks @lelegard, and yes, the exact same problem.

lelegard, 2023-07-25:
Did you try to increase the UDP socket receive buffer size (option`--buffer-size`)?

claudinei-evertz, 2023-07-25:
Tried `--buffer-size-mb 32` and `--buffer-size-mb 64` and they didn't make any difference, still having distortion in the capture.

lelegard, 2023-07-25:
> Tried --buffer-size-mb 32 and --buffer-size-mb 64 and they didn't make any difference, still having distortion in the capture.

No, this is a global `tsp` option which sizes the global packet buffer. I meant the UDP socket receive buffer size in the `ip` input plugin. Please read the user's guide for more details.

claudinei-evertz, 2023-07-26:
My bad, I didn't get that at first. I spent some time running the captures to identify what would be the correct buffer size.

With `--buffer-size 1048576` I was able to get some good captures, however, from time to time, I still get distortion in the captures. By the way, I went from `65536 bytes` all the way to `8388608`, and `1048576` seems to be the sweet spot, with `2097152` I was getting much more distortion, and from `4194304` onwards, using the `--buffer-size` does more harm than good.

Do you think that there is any other argument that I could pass to the TSP command that would improve the quality of the capture?

**Note:** I also tried to push the envelope and capture 3 streams at the same time. It almost max out the ethernet interface, however, there is no value in the --`buffer-size` that would help, and pretty much all the captures were distorted.

Thank you and I really appreciate the help.

lelegard, 2023-07-26:
This is really a socket and system tuning issue, not a specific TSDuck issue.

It depends on several factors:

- The bitrates of all received streams. 
- The bandwidth of the Ethernet interface. 
- The burst profile of each incoming stream. 
- The outgoing bandwidth of the disks where you store the file (especially if this a NAS which reuses the Ethernet interface). 

We may also add the responsiveness of the application. But tsp is specially designed to avoid this. Each plugin executes in its own thread. Therefore, the `ip` input plugin is simply a loop on receive a UDP datagram, extract the TS packets from the datagram, store them in a large global buffer, signal a condition that there is something to do, and loop back. So, tsp should be the last bottleneck. This is why this is a socket and system tuning issue.

claudinei-evertz, 2023-07-27:
Thank you very much for the clarification. I was able to tune (enough) the system to receive 2 streams at the same time. Still have some tuning to do and your help was crucial.

closed 2023-07-27 by claudinei-evertz