tsduck/tsduck #1270: How to Maintain standard Audio quality in UDP to UDP Live stream

M-Bala-2612, 2023-07-26:
```
tsp -v -m --synchronous-log -I ip -r $INPUT_URL \
 -P analyze -i 30 -o report.txt \
 -O ip --ttl 2 -l $OUTPUT_INTERFACE $OUTPUT_URL \
</dev/null > /dev/null 2>&1 2>file.txt & echo $! > $STREAM_PID_PATH  
```

Hello @lelegard 
How to Maintain standard Audio quality in UDP to UDP Live stream.
sometimes automatically audio sound was increased how to Maintain standard Audio quality ,
(Ex ; Movie playing the Audio quality was normal after few mins NEWS playing in the same channel Audio quality was increased automatically.)
Note ; The volume was 27 only both times , but automatically audio sound was increased

udhayNuk, 2023-07-26:
@M-Bala-2612 

its not possible the control the volume 

FUCK THE RFMD

lelegard, 2023-07-26:
TSDuck is a transport stream processor. You would like to transform the content of an audio elementary stream. This is a codec-dependent topic that TSDuck cannot address. You should use audio-processing tools such as ffmpeg.

M-Bala-2612, 2023-07-26:
` if you're talking about varying audio volume of different types of content` 
Yes @AlfonZ42 

Thankyou @lelegard 
Thankyou @AlfonZ42

closed 2023-07-26 by M-Bala-2612