tsduck/tsduck #1271: [Q] Having issues with SCTE-35 PID not showing on listener end of SRT stream.

kalegarner, 2023-07-27:
Hello,

I am having an issue where I am sending an SRT to a remote endpoint with SCTE-35 markers. They said that they are unable to see the PID (600). When doing a test over LAN with two instances of tsp I can see the PID and with splicemonitor on the listener end I can see the SCTE PID using analysis. I am likely sending to a professional encoder. I know in the past you have said you aren't the best person to speak to on SRT since the plugin was created by someone else. Just seeing if you have any advice.

Here is my command.

`ffmpeg -txt_format text -raw_format yuv422p10 -format_code ntsc -teletext_lines all -format_code hp59 -f decklink -i "DeckLink Studio 2" -f mpegts -b:v 5M - | tsp --verbose --initial-input-packets 7 --max-input-packets 7 --bitrate-adjust-interval 1 --max-flushed-packets 14 -v --add-input-stuffing 1/10 -P pmt --service 0001 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject --service 0001 --files C:\FFmpeg\xml.xml --wait-first-batch -P splicemonitor --all-commands -P analyze -O srt --caller xx.xx.xx.xx:xxxx --transtype live --messageapi --multiple`

lelegard, 2023-07-27:
Hi @kalegarner 

This is not an SRT issue. If you just replace `-O srt ...` with `-O ip ...` and you see the splice PID, then the exact same content is sent when using SRT. The output plugin does not change the content, it just outputs it. 

You should check what "they" expect as signalisation for the splice PID and make sure you are compliant.

kalegarner, 2023-07-27:
Thank you for your timely response! As far as signalization is concerned, can you think of anything in your experience that is different between TSDuck and professional decoders? They have seen the PID show up for a brief period of time, but not for a while. 

As far as IP, I am using the same method for another professional decoder with no issues. That one is an ip multicast over LAN.

kalegarner, 2023-07-28:
Something that I forgot to mention is that they are receiving continuity errors. Would that affect the recognition of a SCTE PID?

lelegard, 2023-07-28:
Discontinuities may affect the signalisation only if there are lots of them, really a lot, up to the point where the PMT is never correctly read. 

Tsp alone will never introduce discontinuities. Discontinuities are either:

- preexisting in the input stream
- due to loss of packets in input plugin
- due to loss of packets in output link

I think that SRT is a reliable link. No loss of packets is expected here. 

The input is a pipe. No loss of packets here. 

I don't think that ffmpeg creates discontinuities. 

Try to use the `continuity` plugin in you tsp command to see if there are some (RTFM for details). If there is none, then the problem is on the decoder side. 

Finally, don't be impressed by the "professional" decoder. Some issues here where raised as TSDuck bugs which were reported by a "professional" tool while the bug was actually in the "professional" tool.

kalegarner, 2023-08-01:
Thank you for your reply. I think the issue has partially been nailed down. (Wrong IP analysis)Thank you for your commitment to open source software. I might have one more question related to this, so please don’t close it just yet.

closed 2024-01-21 by github-actions[bot]