tsduck/tsduck #1272: MPEG2-TS stream  Demux & Remux

sanyan97, 2023-07-28:
Hi:  
    Could you tell me where to find the TSDuck .cpp source source code for our project application?   
    The input streams is from the 4x channels Encoder TS over IP output and has been processing through the  python socket program, 
    and looking the TSDuck C++ code to de-mux this TS stream -> extract the TS Video stream (ES) , and keep the Audio/Data (ES) as
     it is  -> then send it back to the python program for the processing 

    After python finish the processing (dynamic change the Encoders bit-rate) re-send it TS Video stream back to the TSDuck for the 
    Mux with the existing Audio/Data (ES) -> then output for the final TS

    Your early reply is highly appreciated

AlfonZ42, 2023-07-28:
Hello.

The source code for TSDuck is in the code portion of this very repository. 
I'm not sure whether you are looking for how to use TSDuck as an utility/executable, or as a library.

Could you please confirm you've read the documentation: https://github.com/tsduck/tsduck#readme, https://tsduck.io/download/docs/tsduck.pdf, https://tsduck.io/doxy/.

Thanks.

sanyan97, 2023-07-28:
Thanks the reply soon,  
I did read that document before,  however it can't tell to find what the sample code we need?  
Kindly tell me exactly which ???.cpp  ???.h files to implement the TS demux -remux

lelegard, 2023-07-28:
On the demux side, it depends on what you want to demux. ES, right, but on which form? TS payload? PES packets? Demuxing TS payload is trivial, just read TS packets, check the PID value and get the payload address and size (TSPacket C++ class). Demuxing PES packet is done using PESDemux. An sample usage is the `pes` plugin. 

On the remux side, well, TSDuck is not a MUX as mentioned multiple times here. However, the simplest way is to replace null packets with the packets of your PID, based on various policies, as used in all injection plugins.

sanyan97, 2023-07-28:
On the demux , in our project, we receive TS over IP streams from 6x Encoders. The TS packets contain video, audio, and data streams. To demux these streams, we need to extract the Elementary Streams (ES) from the TS packets.

In the Python code integration, we are review to using the TSDuck C++ library to handle the demuxing process. Specifically, reading the incoming TS packets, checking the PID values, and extracting the payload data to get the Elementary Streams. This is achieved by using the TSPacket C++ class, which provides the necessary methods to access the payload of the TS packets.

For demuxing PES packets, we can use the PESDemux class provided by TSDuck. The PESDemux class allows us to parse PES packets and extract the individual video, audio, and data streams from them.

On the remux side, for our specific application, we are using TSDuck C++ code to perform bit-rate adjustments on the extracted video stream. After making the necessary bit-rate adjustments, we can then use TSDuck C++ code to remux the modified video stream back into the TS packets.

The remuxing process involves replacing null packets with packets of the modified video PID, based on the required policies for bit-rate adjustments. This is achieved by manipulating the payload of the TS packets and inserting the modified video stream back into the appropriate PIDs. with the existing Audio/Data stream  ES 

I already build TSDuck  development environment (116 projects), with vs2019, tell me exactly which .cpp .h code is for our  requirement??

lelegard, 2023-07-28:
> tell me exactly which .cpp .h code is for our requirement??

There is no such file which does exactly what you want. 

It is up to you to develop your own application from your own specifications. You may explore the TSDuck code base to find inspiration and similar use cases but do not expect an already cooked application.

sanyan97, 2023-07-29:
I  knew there is NO exactly same code for the requirement. 

What I  am asking is the similar template demux/remux sample code and we can develop/modify it from that sample code.
You follow me ??

closed 2023-12-31 by github-actions[bot]