tsduck/tsduck #1275: [Q] How to enter the sat position for DiSEqC 1.1?

andimik, 2023-08-01:
In the PDF, I read

> --satellite-number
value
Used for satellite tuners only.
Satellite/dish number. Must be 0 to 3 with DiSEqC switches and 0 to 1 for non-DiSEqC
switches. The default is zero.

But a friend of mine wants to use DiSEqC 1.1 as he has up to 16 different sat positions.

https://en.wikipedia.org/wiki/DiSEqC

lelegard, 2023-08-01:
Hi @andimik 

There is currently no support for more recent versions of DiSEqC. The current limit is 4 dishes. 

I keep this suggestion for improvement. However, I will not be able to test this kind of reception. So, the improvement will be implemented only if there is a commitment from a user to test it and provide feedback. 

The improvement will be implemented on Linux only. On Linux, the DiSEqC commands are explicitly built by the application (TSDuck) and updates are possible. On Windows, this support is built inside DirectShow and BDA, the Windows framework for broadcasting which is more or less abandoned by Microsoft and I do not expect any improvement.

andimik, 2023-08-02:
Thank you for the fast response.

I forgot to tell that the OS is Ubuntu.

> However, I will not be able to test this kind of reception.

Me too, I will only install `tsduck` on his laptop, but he needs to test it.

But we think we have a workaround: A DVB-S(2) receiver with DiSEqC 1.1 support and LNB loop-through might help here. So the receiver tunes to that satellite and I expect that there is no need to use any DiSEqC parameters under Ubuntu.

andimik, 2023-08-07:
https://www.digitalfernsehen.de/ratgeber/workshop-bis-zu-64-sat-positionen-mit-diseqc-1-1-empfangen-555892/

The author of this article (and also of many other sat books) is the person where I wanted to try that and it's his sat antenna.

I'm not a programmer, just a user, but maybe that helps: https://github.com/Barracuda09/SATPI/issues/150

Or https://www.i-have-a-dreambox.com/wbb5/forum/index.php?thread/106589-probleme-mit-diseqc/&postID=911621#post911621

massweet, 2025-07-16:
Quite how DiSEqC tone burst, 1.0 and 1.1 switches "stack" or "cascade" is left as a user/installer choice.

The DiSEqC specifications permit cascading, but do not give guidance as to "approved configurations", indeed with DiSEqC bus 2.x capable devices and receivers the receiver can discover what is present, and change the slave address of devices leading to very complex installations.

- The "tone burst A" and "tone burst B" commands address one bit of satellite number. They were only ever supposed to be used on their own as cheap devices where the user would only ever want to switch between two dishes;
- The command 0x38 "write to port group 0" is a DiSEqC 1.0 command and addresses 4 bits, which control high/low; polarity; and two bits of satellite number. DiSEqC 1.0 switches are ofte called "committed" switches, I think because the low two bits are "committed" to controlling polarity/range.
- The command 0x39 "write to port group 1" is a DiSEqC 1.1 command that also addresses 4 bits, allowing for the selection of up to 16 "things". This is often interpreted as 4/8/16 ports (depending on how many bits the device supports), with the receiver controlling the LNB with voltage/tone after sending the command.

CEI_EN_61319-1:1996 (sorry, behind a paywall) table ZB.1 suggest that the A/B tone-burst switch should be regarded as (satellite-number & 0x01), and the DiSEqC 1.0 command "0x38" should be regarded as (satellite-number & 0x03). The table also suggests that the Voltage/Tone should always be set appropriately for the LNB.

CEI_EN_61319-1 §ZB.9 explicitly makes it clear that this specification makes no recommendation about cascading DiSEqC devices.

EN50607 (Unicable II) (sorry, also behind a paywall) in §7.1.1 suggests that the DiSEqC 1.1 switch ports and the DiSEqC 1.0 switch ports can be combined as nibbles, with the uncommitted ports being the high nibble.

This suggests that you could model:

- DiSEqC 1.1 switches as being ((satellite_number & 0x3c) >> 2);
- DiSEqC 1.0 switches as being (satellite_number & 0x03).

Note that you would have to set up the switch nearest to the receiver first, wait for 100ms (in the DiSEqC spec v4.2 §5.4) for the next switch to power up, and then send the setup command for the second switch. This means that the physical order of the devices between the receiver and dishes is important to the sequence in which you send the commands.

lelegard, 2025-07-19:
Hi @massweet 

We can extend the DiSEqC support to DiSEqC 1.1, with a single switch. Could you provide the exact spec of what should be done, or a PR? Do you have the appropriate equipment to test this? We cannot write the code without the equipment to test it.

I do not see cascading of DiSEqC switches as a real use case for TSDuck. It's too specific and complicated. I do not see the value for the project. But I may be wrong.

massweet, 2025-07-21:
I do have a DiSEqC 1.1 switch at work, but no use for this at work. I'll have to code it at home & test it at work after hours.
I'm rather busy for the next two weeks.