tsduck/tsduck #1276: [Q] Help me, I don't understand what the reason is

KradinKYu, 2023-08-01:
When receiving a stream over hls and transmitting over udp, there is a slight freezing of the image in vlc
I have already tried the --regulate option and small adjustments, but nothing helps.
Please tell me what could be the reason

tsp -v --debug -I http://online.video.rbc.ru/online/rbctv_576p/index.m3u8 -O ip 224.100.1.1:1234

Latest version, windows 10, standard from the website

hls: downloading from https://e8-online-video.rbc.ru/online2/rbctvhd_576p/1690909364000.ts
* hls: MIME type: video/mp2t, expected size: 1308104 bytes
* Debug: hls: downloading http://online-video.rbc.ru/online/rbctvhd_576p/index.m3u8?e=e8&t=iELf69
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx
* Debug: hls: HTTP header: Date: Tue, 01 Aug 2023 17:02:53 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Transfer-Encoding: chunked
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Last-Modified: Tue, 01 Aug 2023 17:02:49 GMT
* Debug: hls: HTTP header: ETag: W/"64c93ab9-19b"
* Debug: hls: HTTP header: Expires: Tue, 01 Aug 2023 17:02:53 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header:
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* Debug: hls: playlist media sequence: old: 1949965/0, new: 1949961/4
* Debug: hls: no new segment in playlist
* Debug: hls: downloading http://online-video.rbc.ru/online/rbctvhd_576p/index.m3u8?e=e8&t=iELf69
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx
* Debug: hls: HTTP header: Date: Tue, 01 Aug 2023 17:02:56 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Transfer-Encoding: chunked
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Last-Modified: Tue, 01 Aug 2023 17:02:49 GMT
* Debug: hls: HTTP header: ETag: W/"64c93ab9-19b"
* Debug: hls: HTTP header: Expires: Tue, 01 Aug 2023 17:02:56 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header:
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* Debug: hls: playlist media sequence: old: 1949965/0, new: 1949961/4
* Debug: hls: no new segment in playlist

lelegard, 2023-08-01:
Hi @KradinKYu 

HLS is bursty by design (each segment is read at the speed of the network). Therefore, with `ip` output, `regulate` is required. Also make sure to pass 7 packets at a time (see the many topics on `ip` output about regulation and buffering).

KradinKYu, 2023-08-05:
Sorry, I tried different options:

tsp -I hls https://hls-03-video.webcaramba.com/rutv/live_720/index.m3u8 -O ip 224.100.1.1:1234&pkt_size=1316&fifo_size=1000000&burst_bits=188 -P regulate

tsp --realtime=off --add-input-stuffing 15/25 --buffer-size-mb 50 -I hls https://hls-03-video.webcaramba.com/rutv/live_720/index.m3u8 -P regulate -O ip 224.100.1.1:1234&fifo_size=5000000 -s 188 -p 7 -P filter -n -p 0x1FFF

tsp -I hls https://hls-03-video.webcaramba.com/rutv/live_720/index.m3u8 -P regulate --pcr-synchronous -O ip 224.100.1.1:1234 -s 188 -p 7

I can't adjust the bitrate in any way....

please tell me, maybe there is a universal recipe how can I do this?

lelegard, 2023-08-07:
> `-O ip 224.100.1.1:1234&pkt_size=1316&fifo_size=1000000&burst_bits=188`

Where did you get that kind of syntax from ?

lelegard, 2023-08-07:
Try something like this:
~~~
tsp --max-input-packets 7 --max-flushed-packets 7 \
    -I hls ... \
    -P regulate --pcr-synchronous --packet-burst 7 \
    -O ip 224.100.1.1:1234 --enforce-burst 
~~~

KradinKYu, 2023-08-12:
I took your advice, similar errors began to appear, please tell me what is the reason for this?

tsp --max-input-packets 7 --max-flushed-packets 7 --buffer-size-mb 5 --max-output-packets 7 -I hls https://hls-03-video.webcaramba.com/rutv/live_720/index.m3u8 -P regulate --packet-burst 7 -P bitrate_monitor -O ip 224.100.1.1:1234 --enforce-burst

Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (4,866,643 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (1,433,914 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (4,783,923 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (4,202,477 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (5,130,144 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (4,955,981 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (4,525,536 bits/s) is back in allowed range (10-4,294,967,295 bits/s)
* Warning: bitrate_monitor: TS bitrate (0 bits/s) is lower than allowed minimum (10 bits/s)
* Warning: bitrate_monitor: TS bitrate (5,029,677 bits/s) is back in allowed range (10-4,294,967,295 bits/s)

lelegard, 2023-08-13:
I looked at the URL you mention. The stream has some unusual characteristics:

- Very few PCR, only 0.6 per second, while the usage is typically one PCR per frame, i.e. 25 or 30 PCR per second.
- The PCR rate is not stable. Sometimes, two consecutive PCR's are distant by more than 2 seconds.
- The bitrate is highly variable, with large variations between lower and higher values.

This means that it is difficult to evaluate in real time the instant bitrate. Evaluating the instant bitrate from PCR requires that the PCR samples arrive at a higher rate than the bitrate variation (this is some kind of application of the Shannon's theorem on sampling).

Regulating a stream requires a reasonably valid evaluation of the instant bitrate. Full accuracy is not necessary when receivers have some buffering capabilities. However, here, we have zero accuracy.

For highly variable bitrates, regulating using `--pcr-synchronous` may be a solution. However, it requires sufficiently close PCR samples. It generates errors and loses synchronization when two consecutive PCR's are distant by more than 2 seconds, which is the case here on a regular basis.

This makes this stream impossible to regulate. This is a stream which is typically encoded for HLS, where the media player or receiver can receive and play at different rates, with proper buffering in between.

I see no solution to regulate this stream for a reliable HLS -> UDP conversion.

closed 2024-01-14 by github-actions[bot]