tsduck/tsduck #1277: [Q] how to select particular network interface to push rtp streaming data (eth0,eth1,wifi..)

sivashakthip, 2023-08-02:
currently, I am using the below command for RTP streaming from mpeg ts file
tsp -v -I file -i file.mpg -P regulate -O ip -r 234.100.200.1:5001 

is there a way to push only to a particular network interface (eth0,eth1, wifi..)?

sivashakthip, 2023-09-22:
@lelegard I couldn't find the answer in the document, could you please mention the topic/title name from the document where i can find the interface selection

lelegard, 2023-09-22:
TSDuck User's Guide from https://tsduck.io/ => Documentation => User's guide => tsp plugins => ip output plugin.

Or simply:
~~~
$ tsp -O ip --help
...
  -l value
  --local-address value
      When the destination is a multicast address, specify the IP address of
      the outgoing local interface. It can be also a host name that translates
      to a local address.
...
~~~

closed 2023-10-01 by sivashakthip