tsduck/tsduck #1279: [Q] How to use packet processing versions of IP and file outputs with C++ library?

GrapheneCt, 2023-08-02:
When using standalone tsp application, I can do something like this: `tsp.exe -I file file_in.ts -P regulate -O file file_out.ts -P t2mi --log` to utilize passthrough version of file output as described in docs.
However when attempting to push `file` or `ip` plugins into the `PluginOptionsVector` of `ts::TSProcessorArgs`, tsp doesn't recognize them as valid plugins.
What would be the proper way to use passthrough versions of outputs with the library?

GrapheneCt, 2023-08-02:
That's just a basic usage example. I already have full working app that integrates tsduck as a library.

It appears that I've been confused by the User's Guide pdf because it contains two separate descriptions for "file (output)" and "file (packet processing)" and also has a typo in the latter, stating `-P file` instead of `-O file`, which leaves the impression that you can have multiple outputs, with one of them forking the packets from the middle of the plugin chain and other one being normal final output which is not actually possible.

Because of that I don't really see a point of "file (packet processing)" and "Ip (packet processing)". Correct me if I'm wrong, but it seems to me that those two sections just mislead user into thinking that they can have multiple outputs in the middle of the plugin chain.

AlfonZ42, 2023-08-03:
There are indeed three `file` plugins - input, packet processing, and output. The selection of which one is used is done via `-I`, `-P`, or `-O`. 
So `-P file` refers to the packet processing plugin and `-O file` to the output plugin. I don't see a typo in the doc in regards to this.

`tsp -I file file_in.ts -P file file_out1.ts -O file file_out2.ts` is a valid command, albeit a silly one, and does produce two output files.

Applying
```c
    opt.input = {u"file", {u"file_in.ts"}};
    opt.plugins = {
        {u"file", {u"file_out1.ts"}}
    };
    opt.output = {u"file", {u"file_out2.ts"}};
```
into _sample-app/myexec.cpp_ worked for me as well, and produced two output files as expected.

GrapheneCt, 2023-08-03:
Your tsp example does indeed work but `tsp.exe -I file dile_in.ts -P regulate -P ip 127.0.0.1:4322 -O file file_out2.ts` does not. Is there just no -P ip plugin?

lelegard, 2023-08-03:
There is indeed a `-P ip` plugin but it was recently introduced and is not in version 3.34. 

You need to rebuild from the latest state of the repo. If you use Windows, you may download the latest nightly build from here: https://tsduck.io/download/prerelease/

GrapheneCt, 2023-08-03:
Thanks, that clears things up for me.

closed 2023-08-03 by GrapheneCt