tsduck/tsduck #1280: [Q] Can you scramble a PID without fixed CW

xfiles2007, 2023-08-05:
Is it possible to scramble a PID without a fixed CW input file.

I have transport stream and I wish to encrypt a specific PID (Video file only, not audio for example) but want to connect it to my ecmg 

When I specify the the PID I get this error : 

* Error: scrambler: specify control words to scramble an explicit list of PID's

xfiles2007, 2023-08-05:
Encrypting the whole service with ecmg is no issue but I just want 1 specific PID.

lelegard, 2023-08-07:
Typically, an ECMG is used to scramble a service, meaning the video PID and, most of the time, all audio PID's. The options `--no-audio` and `--no-video` can leave audio or video PID's in the clear. Do you have a different use case?

closed 2024-01-07 by github-actions[bot]