tsduck/tsduck #1282: [Q] Playing out at specified bitrate with -P regulate

AWhateley, 2023-08-09:
I am playing out a ts with target bitrate 9000900 and for some reason the bitrate monitor is playing out at 9024000.
I'm sure I must be misunderstanding something. I have tried changing the packet-burst value.
Find below some of my attempts.

```
tsp -I file playout.ts -P regulate -b 9000900 -P bitrate_monitor -p 5  -O ip 239.
30.52.33:1234 -t 7
* bitrate_monitor: 2023/08/09 20:01:20, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,947 bits/s
* bitrate_monitor: 2023/08/09 20:01:25, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,947 bits/s
* bitrate_monitor: 2023/08/09 20:01:30, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,646 bits/s
* bitrate_monitor: 2023/08/09 20:01:35, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,646 bits/s
* bitrate_monitor: 2023/08/09 20:01:40, TS bitrate: 9,024,000 bits/s, net bitrate: 7,801,248 bits/s
* bitrate_monitor: 2023/08/09 20:01:45, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,646 bits/s
* bitrate_monitor: 2023/08/09 20:01:50, TS bitrate: 9,024,000 bits/s, net bitrate: 7,801,549 bits/s
* bitrate_monitor: 2023/08/09 20:01:55, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,646 bits/s

tsbitrate playout.ts
TS bitrate: 9,000,948 b/s (188-byte), 9,766,987 b/s (204-byte)

tsp -I file AlMamlaka.ts -P regulate  -P bitrate_monitor -p 5  -O ip 239.30.52.33:1
234 -t 7
* bitrate_monitor: 2023/08/09 20:10:21, TS bitrate: 9,072,128 bits/s, net bitrate: 7,842,458 bits/s
* bitrate_monitor: 2023/08/09 20:10:26, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,947 bits/s
* bitrate_monitor: 2023/08/09 20:10:31, TS bitrate: 9,072,128 bits/s, net bitrate: 7,842,157 bits/s
* bitrate_monitor: 2023/08/09 20:10:36, TS bitrate: 9,024,000 bits/s, net bitrate: 7,800,646 bits/s
* bitrate_monitor: 2023/08/09 20:10:41, TS bitrate: 9,024,000 bits/s, net bitrate: 7,801,248 bits/s

tsp --version=bitrate
64-bit floating-point value
```

lelegard, 2023-08-11:
Hi @AWhateley 

You point a limitation of the current rudimentary implementation of the `regulate` plugin.

Let me explain this. First, there are two constraints:
- `tsp` works on TS packets, not bits. A TS packet is 8 * 188 = 1504 bits.
- Each operating system has a minimum wait time for a thread. You cannot make a thread wait for less than a certain amount of time. On traditional Linux configuration, this is 10 ms. It can be tuned at kernel level but `tsp` does not count on it. On Windows, the default time granularity is 20 ms. You can ask for less and Windows will return the best it can do, regarding your request.

So, assuming you work on Linux, the minimum wait is 10 ms. If you request a bitrate of 9000900 b/s, that means that the `regulate` plugin shall let 90009 bits pass every 10 ms. 9009 bits means 59 TS packets plus 1273 extra bits. Since a plugin can pass only entire TS packets, we round this to 60 packets. And 60 packets every 10 ms exactly make 9024000 bits per second.

Generally speaking, the higher the bitrate is, the worse the regulation is. Specifically when we reach the minimum wait time of the operating system. On Linux, this limit is reached between 2 and 3 Mb/s. Additionally, the threshold effect depends on the extra bits at each wait time (+1273 bits in your case). Doing some tests, we can see that regulating at 5 Mb/s is not that bad (+ 1kb/s) while regulating at 4 Mb/s is the worst case (+ 60kb/s). At your bitrate, you get +23kb/s.

I agree that this is not a satisfactory implementation. I will try to improve this, based on the actual number of passed packets, and compensate for the extra packets, reducing the number of packets on later bursts.

Sorry for the inconvenience. I will post an update here when a better implementation is available.

AWhateley, 2023-08-11:
Thank you for your detailed explanation. 
Do you have any recommendation to work around this with other plugins?
I got closer by using pcr-synchronous but still suffered some level of introduced PCR drift compared to other playout methods.

lelegard, 2023-08-11:
I have no other option for the moment inside TSDuck.

TSDuck was originally developed for lab experimentations on content transformation (encryption and signalization) and CAS integration. The plugin `regulate` was a minimal implementation to play a TS file over unregulated outputs such as UDP and do some tests on the received contents.

I need to improve this.

AWhateley, 2023-08-11:
That's fair, thank you again. I will use other tools for this in the meantime, appreciate you improving this in future.

lelegard, 2023-08-11:
Hi @AWhateley 

I was a bit upset by the current implementation and I decided to fix it right away. The fix is applied to the bitrate-based regulation only, not to the PCR-synchronous one.

The amount of passed bits is more closely tracked over pairs of measurement periods. Since entire TS packets (1504 bits) must be passed, missing bits or excess bits from a period are credited or extracted from the next period.

The result seems satisfactory. Example on macOS:
~~~
$ tsp  -I null -P until -s 20 -P regulate -b 9000900 -P bitrate_monitor -p 2 -s -O drop
* bitrate_monitor: 2023/08/12 00:43:23, TS bitrate: 9,000,749 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:25, TS bitrate: 9,000,633 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:27, TS bitrate: 9,001,777 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:29, TS bitrate: 9,002,316 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:31, TS bitrate: 9,001,668 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:33, TS bitrate: 9,004,601 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:35, TS bitrate: 8,999,281 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:37, TS bitrate: 9,001,855 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:39, TS bitrate: 9,000,697 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:41, TS bitrate: 9,000,468 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:43, TS bitrate: 8,999,308 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:45, TS bitrate: 8,999,546 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:47, TS bitrate: 9,003,571 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: 2023/08/12 00:43:49, TS bitrate: 8,998,403 bits/s, net bitrate: 0 bits/s
* bitrate_monitor: TS average bitrate: 9,000,913 bits/s, average net bitrate: 0 bits/s
~~~

Again, since packets are passed by multiples of 1504 bits, the bitrate which is evaluated over periods of 2 seconds cannot be _exactly_ the requested value but the final average seems correct (the option `-s` or `--summary` has just been added to `bitrate_monitor` for that purpose).

You need to pull the latest state of the repo and rebuild (see the [details here](https://tsduck.io/doxy/building.html)).

AWhateley, 2023-08-12:
That's great, I wasn't expecting a fix so soon, thanks.
I will test this next week and comment here if there is still a noticeable PCR drift or if it goes within standards.

lelegard, 2023-08-12:
Yes, please, let us know the result.

I don't think that PCR-synchronous regulation need any fix. In the following example, I artificially generate PCR for a 9 Mb/s CBR stream and the regulation seems correct:
~~~
tsp -I null \
    -P filter --every 150 --set-label 1 \
    -P craft --only-label 1 --pid 100 --no-payload --pcr 0 \
    -P pcradjust -b 9,000,000 \
    -P regulate --pcr-synchronous \
    -P bitrate_monitor --periodic-bitrate 2 --summary \
    -P until --seconds 30 \
    -O drop
* bitrate_monitor: 2023/08/12 13:42:44, TS bitrate: 9,001,780 bits/s, net bitrate: 60,012 bits/s
* bitrate_monitor: 2023/08/12 13:42:46, TS bitrate: 9,000,065 bits/s, net bitrate: 60,000 bits/s
* bitrate_monitor: 2023/08/12 13:42:48, TS bitrate: 9,000,051 bits/s, net bitrate: 60,000 bits/s
* bitrate_monitor: 2023/08/12 13:42:50, TS bitrate: 8,999,388 bits/s, net bitrate: 59,996 bits/s
* bitrate_monitor: 2023/08/12 13:42:52, TS bitrate: 9,000,496 bits/s, net bitrate: 60,003 bits/s
* bitrate_monitor: 2023/08/12 13:42:54, TS bitrate: 9,001,807 bits/s, net bitrate: 60,012 bits/s
* bitrate_monitor: 2023/08/12 13:42:56, TS bitrate: 9,001,423 bits/s, net bitrate: 60,009 bits/s
* bitrate_monitor: 2023/08/12 13:42:58, TS bitrate: 8,999,958 bits/s, net bitrate: 60,000 bits/s
* bitrate_monitor: 2023/08/12 13:43:00, TS bitrate: 9,000,004 bits/s, net bitrate: 60,000 bits/s
* bitrate_monitor: 2023/08/12 13:43:02, TS bitrate: 8,999,893 bits/s, net bitrate: 59,999 bits/s
* bitrate_monitor: 2023/08/12 13:43:04, TS bitrate: 8,998,055 bits/s, net bitrate: 59,987 bits/s
* bitrate_monitor: 2023/08/12 13:43:06, TS bitrate: 9,000,040 bits/s, net bitrate: 60,000 bits/s
* bitrate_monitor: TS average bitrate: 8,999,927 bits/s, average net bitrate: 59,999 bits/s
~~~

But this is a CBR stream. Maybe VBR will give a less reliable result.

AWhateley, 2023-08-16:
@lelegard Hi, I have tried this fix out and I am not seeing enough drift to cause a problem, so at least up to 10mb/s this looks good to me.
Yes, I agree VBR will be less reliable, but that's not really unexpected or an issue.
Appreciate the fast fix, will close the issue.

closed 2023-08-16 by AWhateley