tsduck/tsduck #1283: solve my error when i converted udp to udp[BUG]

udhayNuk, 2023-08-11:
@lelegard  
solve my error when i converted udp to udp service name was renamed and scrambled.

tsp -v -m --synchronous-log -I ip -r $INPUT_URL \
 -P svrename -i 10 \
 -P scrambler --ignore-scrambled --cw 0000000000000001 --partial-scrambling 7 10 \
 -P filter --pid 0x0014 --pid 0x0010 --pid 0x0001 --negate \
 -P pmt --increment-version --add-ca-descriptor "0x4b19/0x03e8" \
 -P analyze -i 30 -o report.txt \
 -O ip --ttl 2 -l $OUTPUT_INTERFACE $OUTPUT_URL \

#### Errors and logs:
* [MON] 2023/08/09 20:08, stopping stable monitoring messages to avoid infinitely large log files
* svrename: service id 0x002E not found in PAT, will still update NIT, BAT, EIT's
* Error: scrambler: service id 0x000A (10) not found in PAT
* ip: leaving multicast group 237.1.1.4 from local address 192.168.0.23
* ip: leaving multicast group 237.1.1.4 from local address 192.168.47.6
* ip: leaving multicast group 237.1.1.4 from local address 192.168.48.5
* ip: leaving multicast group 237.1.1.4 from local address 192.168.49.4
* ip: leaving multicast group 237.1.1.4 from local address 192.168.50.35
* ip: leaving multicast group 237.1.1.4 from local address 192.168.50.36
* [MON] 2023/08/10 00:18, resource monitoring terminated

#### Environment:
 - OS: 
 - OS version:  Ubuntu 20.10 -

please solve this error
thanking you in advance

M-Bala-2612, 2023-08-11:
@udhayNuk 
Is was very simply issue ..
Pls Read the user guide and Solve it .

lelegard, 2023-08-11:
Please provide the _real_ command you used.

From the code, the message _"svrename: service id 0x002E not found in PAT, will still update NIT, BAT, EIT's"_ is displayed only when the _explicitly specified_ service is not found and the command you posted only mentions `-P svrename -i 10` which means "rename the first service in the PAT, whichever it is", without explicitly specifying the service to rename.

udhayNuk, 2023-08-11:
tsp -v -m --synchronous-log -I ip -r $INPUT_URL
-P svrename -i 10
-P scrambler --ignore-scrambled --cw 0000000000000001 --partial-scrambling 7 10
-P filter --pid 0x0014 --pid 0x0010 --pid 0x0001 --negate
-P pmt --increment-version --add-ca-descriptor "0x4b19/0x03e8"
-P analyze -i 30 -o report.txt
-O ip --ttl 2 -l $OUTPUT_INTERFACE $OUTPUT_URL \

this is the script i have used and runned the script after several hours it has stopped. please give the what is the issue occurs and please solve this error
the occured error was this only

[MON] 2023/08/09 20:08, stopping stable monitoring messages to avoid infinitely large log files
svrename: service id 0x002E not found in PAT, will still update NIT, BAT, EIT's
Error: scrambler: service id 0x000A (10) not found in PAT
ip: leaving multicast group 237.1.1.4 from local address 192.168.0.23
ip: leaving multicast group 237.1.1.4 from local address 192.168.47.6
ip: leaving multicast group 237.1.1.4 from local address 192.168.48.5
ip: leaving multicast group 237.1.1.4 from local address 192.168.49.4
ip: leaving multicast group 237.1.1.4 from local address 192.168.50.35
ip: leaving multicast group 237.1.1.4 from local address 192.168.50.36
[MON] 2023/08/10 00:18, resource monitoring terminated

lelegard, 2023-08-11:
> runned the script after several hours it has stopped

OK, is this means that the only possible scenario is the following:

- The command starts with "rename the first service in the PAT"
- The first PAT is received and its first (or only) service is 0x002E.
- Service 0x002E is now the service to rename. It is renamed as 10 (0x000A) and scrambled for hours.
- At some point, the PAT changes and the new PAT no longer contains service 0x002E.
- The plugin `svrename` can no longer rename the service in the PAT (but will continue to rename it in the NIT, BAT and EIT's as indicated in the message).
- The new PAT is no longer modified and is passed to the next plugins.
- The scrambler plugin receives the new PAT and does not find service 10 in it. It terminates the processing with an error.

This is not a bug, this is the expected behavior. If you want to verify the scenario, add `-P tables --pid 0` before `svrename`. You will see the content of each new PAT in the log.

udhayNuk, 2023-08-11:
@lelegard 
thanking you for your reply

how to solve this adding any other commands?

lelegard, 2023-08-11:
You cannot solve this in one tsp command. This is not the way it should work. If your service id changes regularly, this is the symptom of some weird operational practices.

From a pragmatic perspective, I would simply recommend to write a script where you loop around the tsp command and restart it when it fails. Each time the PAT changes and the service id changes, the command fails and it is automatically restarted. When the command restarts, it uses the new first service as service to scramble.

In that case, no need to use `svrename` (unless you really want to rename it). Simply use an explicit empty name in `-P scrambler "" ...`.

If you want to do something really reliable and resilient, take care of the following operational best practices:
- In your script, compute the interval between two restarts of the command. Enforce a minimum duration between two restarts. If the command fails from the beginning because of some other system or command line error, it would permanently loop on fail/restart without that precaution.
- If you save the tsp output in a log file, create a new log file at each start to be able to compare successive executions.
- But be sure to permanently purge old logs in order to avoid the saturation of the disk with log files. For instance, before restarting the command, only keep the most recent N log files and delete older ones.

But this is just scripting, no longer a TSDuck issue.

udhayNuk, 2023-08-11:
@lelegard 
thanking you for your reply.
how to restart the tsp if its fails please give me.

lelegard, 2023-08-11:
> how to restart the tsp if its fails please give me.

Are you kidding? Your command is supposed to never end. So, if the command terminates, it failed. Just restart it. So, for instance, without the precautions from my previous post:
~~~
while true
do
    tsp ...
done
~~~

If you want to run serious operations with TSDuck, you must master shell scripting.

udhayNuk, 2023-08-11:
@lelegard 
ok thank you.

udhayNuk, 2023-08-14:
> Are you kidding? Your command is supposed to never end. So, if the command terminates, it failed. Just restart it. So, for instance, without the precautions from my previous post:
>
> ```
> while true [...]

#!/bin/bash

STREAM_PID_PATH1=/home/user/BALAJI/udhay/pid_main/stream11-pid

while true; do

tsp -v -m --synchronous-log -I ip -r 224.2.2.1:5000 \
 -P scrambler --ignore-scrambled --cw 0000000000000001 --partial-scrambling 7 "" \
 -P filter --pid 0x0014 --pid 0x0010 --pid 0x0001 --negate \
 -P pmt --increment-version --add-ca-descriptor "0x4b19/0x03e8" \
 -P analyze -i 30 -o testreport.txt \
 -O ip --ttl 2 235.1.1.1:1245 \
</dev/null > /dev/null 2>&1 2>testing.txt & echo $! > $STREAM_PID_PATH1  

    # Capture the PID of the tsp process
    tsp_pid=$!

    # Wait for tsp process to finish
    wait $tsp_pid

    # Print message indicating tsp process stopped
    echo "TSP process (PID: $tsp_pid) stopped. Restarting..."
done

M-Bala-2612, 2023-08-19:
@udhayNuk  
Pls Study the Basic of this User Guide
https://tsduck.io/download/docs/tsduck.pdf

Pls Don't Waste the lelegaurd time , you have Brain so Study this Guide and You Solve this Problem

closed 2023-08-19 by udhayNuk