tsduck/tsduck #1286: [BUG] SRT Listener Accepts Multiple Concurrent Connections on the Same Port

fnwtv, 2023-08-20:
#### Bug description:

When using TSDuck with an SRT listener as input (tsp -I srt --listener 0.0.0.0:5000 -O drop), it is possible to send multiple concurrent streams to the same port simultaneously. As a result, the output stream becomes corrupted due to multiple input streams blending together. In contrast, ffmpeg rejects subsequent connections when one session is already established on an SRT listening port.

#### How to reproduce:

- Start a listener using TSDuck: tsp -I srt --listener 0.0.0.0:5000 -O drop.
- Use any SRT tool to send a stream to the listener on port 5000.
- Without stopping the first stream, send another stream to the same port.

#### Expected behavior:

Once an SRT session is established on the listening port in TSDuck, any further attempts to connect should be rejected until the active session ends.

#### Errors and logs:

Theses errors shows up when it happens
~~~
14:00:16.417439/SRT:RcvQ:worker*E:SRT.c: %248259729:No room to store incoming packet: offset=5751 avail=0 ack.seq=675703979 pkt.seq=675709730 rcv-remain=8191
14:00:16.417500/SRT:RcvQ:worker*E:SRT.c: %248259729:No room to store incoming packet: offset=5752 avail=0 ack.seq=675703979 pkt.seq=675709731 rcv-remain=8191
14:00:16.439903/SRT:RcvQ:worker*E:SRT.c: %248259729:No room to store incoming packet: offset=5774 avail=0 ack.seq=675703979 pkt.seq=675709753 rcv-remain=8191
14:00:16.439963/SRT:RcvQ:worker*E:SRT.c: %248259729:No room to store incoming packet: offset=5775 avail=0 ack.seq=675703979 pkt.seq=675709754 rcv-remain=8191
~~~

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 20.04
 - TSDuck full version: 3.33-3042 
 - Installation type: Rebuilt binary 

#### Additional information:

lelegard, 2023-08-20:
Hi @fnwtv 

> When using TSDuck with an SRT listener as input [...], it is possible to send multiple concurrent streams to the same port simultaneously. As a result, the output stream becomes corrupted due to multiple input streams blending together.

Are you sure about the blending of the streams?

I do not observe this. I just tried and I can see that the second stream is not read by the receiver. More precisely, the second connection is not rejected but the sender seems to wait.

I just did this on macOS with TSDuck v3.35. First, I start a receiver, as you indicate. The plugin `count` will display the number of packets per PID at the end of the reception.
~~~
$ tsp -I srt --listener 5000 -P count -O drop
~~~

Then, from the same system, I start a first sender. It sends all packets with PID 100 at 1 Mb/s:
~~~
$ tsp -I null -P craft -p 100 -P regulate -b 1000000 -O srt --caller localhost:5000
~~~
Then, I start the second sender with all packets on PID 200:
~~~
$ tsp -I null -P craft -p 200 -P regulate -b 1000000 -O srt --caller localhost:5000
~~~
The second client is not rejected. It waits.

I let this run a few seconds. Then, I interrupt the receiver:
~~~
$ tsp -I srt --listener 5000 -P count -O drop
^C
* tsp: user interrupt, terminating...
* count: PID  100 (0x0064):     16,064 packets
~~~

We can see that all received packets have PID 100. Only the first stream was received. No packet from the second stream was received.

Could you please analyze if your two streams are actually received and blended or if the second client simply waits?

lelegard, 2023-08-20:
Hi again,

I think that my analysis is correct. The second sender just waits. If I try to start a third sender, it fails to connect with this error message:
~~~
19:42:16.133098/!W:SRT.cn: @294727451: processConnectResponse: rejecting per reception of a rejection HS response: ERROR:BACKLOG
* Error: srt: error during srt_connect: Connection setup failure: connection rejected, reject reason: Listener's backlog exceeded
~~~

At the same time, the receiver displays this error (without stopping the processing of the first stream):
~~~
19:42:16.133022/!W:SRT.cn: processConnectRequest: rsp(REJECT): 1005 - Listener's backlog exceeded
~~~

It seems that the listener uses a default backlog of 1, meaning one sender waiting its turn. But when another sender wants to connect, the backlog is exceeded and the connection is rejected.

I will see if it is possible to set the listener backlog in libsrt. This is tuning and enhancement. But I see no mix of input streams (which would be a bug if that was true).

lelegard, 2023-08-20:
After looking at the code of the `srt` plugins, I confirm the following points:

- The listener backlog is hardcoded to 1.
- The `SRTO_REUSEADDR` is hardcoded to true.
- After `srt_listen`, `srt_accept` is called exactly once. Only one client socket is accepted and all packets are read from this unique client connection.

The fact that `SRTO_REUSEADDR` is hardcoded to true made me worry a while. Normally, this is set to avoid errors when you restart a listener immediately after it fails. Without this, the port is not immediately reusable and an error is generated. I wondered if this left the possibility to run two simultaneous listeners, with unexpected mix of packets. But it does not. The `SRTO_REUSEADDR` option allows an early reuse of a socket that just broke. But is does not allow to run two simultaneous listeners on the same port. The second listener immediately fails with the following error:
~~~
* Error: srt: error during srt_bind(): Connection setup failure: unable to create/configure SRT socket
~~~

Consequently, I see no way to mix packets from two SRT senders. I may have missed something, so please provide an exact reproducer if you think there is still an error.

fnwtv, 2023-08-20:
Hi @lelegard,

Thank you for your quick and detailed response.

Upon retesting on my end, I've observed two key factors:

- My initial test feed contained corrupted packets.
- The computer I utilized to initiate the two caller feeds had some bandwidth limitations.

When launching two callers simultaneously, the received feed was notably corrupted, likely attributable to the aforementioned issues. Concurrently, I observed log entries on the listener indicating `No room to store incoming packets.` when I have two callers. The two callers also presented SRT statistics suggesting active transmission. This led me to believe that both callers were transmitting, and given the increased number of corrupted packets due to the system's sluggish bandwidth, I mistakenly assumed the feeds were intertwined.

Sorry for the mistake and the time it took you to check it. I should have been more meticulous in my verification process before posting an issue.

I will close the issue now. 
Thanks again

closed 2023-08-20 by fnwtv

lelegard, 2023-08-21:
FYI, I pushed an update to make sure that the second client attempt will be rejected right away with an explanatory message to avoid that kind of confusion in the future.