tsduck/tsduck #1287: how to calculate how many users currently viewing my streams how to calculate?[Q]

udhayNuk, 2023-08-23:
@lelegard 
tsp -v -m --synchronous-log -I ip -r $INPUT_URL
-P svrename -i 10
-P scrambler --ignore-scrambled --cw 120252106510016 --partial-scrambling 7 10
-P filter --pid 0x0014 --pid 0x0010 --pid 0x0001 --negate
-P pmt --increment-version --add-ca-descriptor "0x4b19/0x03e8"
-P analyze -i 30 -o report.txt
-O ip --ttl 2 -l $OUTPUT_INTERFACE $OUTPUT_URL \

i am streaming udp to udp live streams,

i want to calculate how many users currently viewing my streams(output udp) how to calculate please give me?

lelegard, 2023-08-23:
Hi @udhayNuk 

> i want to calculate how many users currently viewing my streams(output udp)

This is impossible by design of the UDP protocol.

If you use a unicast IP address, the answer is 0 or 1. If you use a multicast address, the answer is something between zero and the size of the Internet.

udhayNuk, 2023-09-29:
@lelegard thanking you

closed 2023-09-29 by udhayNuk