tsduck/tsduck #1290: [BUG] merging PID 0x14 does not work

tarelda, 2023-08-30:
#### Bug description:
I tried to inject TDT/TOT from one stream (239.0.0.2) to another (239.0.0.1). I think tsduck just hangs instead of doing something.

#### How to reproduce:
tsp -I ip 239.0.0.1:1234 -P filter -n -p 0x14 -P merge -t --terminate "tsp -I ip 239.0.0.1:1234 -P filter -p 0x14" -P filter -p 0x14 -P tables -P until -s 120 -O file file1.ts

#### Expected behavior:
I expected to get 0x14 PID from stream 239.0.0.2 in file file1.ts. Command never finishes so I expect that something hangs.

#### Errors and logs:
No useful logs generated with -v flag. Only multicast join messages.

#### Environment:
 - OS: Debian
 - OS version: Bookworm
 - TSDuck full version: 3.35-3258
 - Installation type: official .deb

AlfonZ42, 2023-08-30:
How does the behaviour change if you add some input stuffing, e.g. `tsp -v -a 1/10 -I ip ...`, or input 0x14 packets are nulled instead of filtered (`-P filter -s -n -p 0x14`)?

tarelda, 2023-09-01:
> How does the behaviour change if you add some input stuffing, e.g. `tsp -v -a 1/10 -I ip ...`, or input 0x14 packets are nulled instead of filtered (`-P filter -s -n -p 0x14`)?

Adding stuffing or defining constant bitrate produces output.

closed 2024-02-11 by github-actions[bot]