tsduck/tsduck #1293: how to select the one audio from input?[Q]

udhayNuk, 2023-09-09:
@lelegard 

this is my input

Input #0, mpegts, from 'udp://236.1.3.9:1026':
  Duration: N/A, start: 31207.801544, bitrate: N/A
  Program 109
    Metadata:
      service_name    : Hungama
      service_provider: STAR
    Stream #0:0[0x47f]: Video: h264 (High) ([27][0][0][0] / 0x001B), yuv420p(tv, bt470bg, top first), 720x576 [SAR 12:11 DAR 15:11], 25 fps, 50 tbr, 90k tbn, 50 tbc
    Stream #0:1[0x480]: Audio: mp2 ([4][0][0][0] / 0x0004), 48000 Hz, stereo, s16p, 192 kb/s
    Stream #0:2[0x481]: Audio: mp2 ([4][0][0][0] / 0x0004), 48000 Hz, stereo, s16p, 192 kb/s
    Stream #0:3[0xef7]: Unknown: none (CUEI / 0x49455543)

tsp -I ip 236.1.3.9:1026 -P zap Hungama --audio-pid "0x481" -O ip -l 127.0.0.1 235.1.1.2:1234  
this is the command iam using extracting only one audio from input but my audio is not played in output

please solve my issue 
thank you in advance

lelegard, 2023-09-09:
Hi @udhayNuk 

I suggest that, as a test, you add `-P analyze` after `-P zap` to check that the audio PID is actually present and with some bandwidth. Also try `-O play` instead of `-O ip` to check if the audio is played with the video (be sure to have VLC installed).

udhayNuk, 2023-09-09:
@lelegard 
thank you for your reply
i have attached the report with adding zap extracting one audio and without extracting the audio to output.

without extracting the one audio [tsp -I ip 236.1.3.9:1026 -O ip -l 127.0.0.1 235.1.1.2:1234 ] the output was playing the default audio]
[using zap with selecting one audio.txt](https://github.com/tsduck/tsduck/files/12565525/using.zap.with.selecting.one.audio.txt)
[without selecting audio pid.txt](https://github.com/tsduck/tsduck/files/12565526/without.selecting.audio.pid.txt)

udhayNuk, 2023-09-09:
@lelegard 
any other option in tsduck to select one audio from input
please reply

lelegard, 2023-09-09:
Hum, this is a bug, a regression which appeared at some point. The options `--audio` and `-subtitles` which select components by language still work (and there was a non-regression test for that). But the options `--audio-pids` and `--subtitles-pid` which select components by PID were broken (and there was no regression test for that).

I fixed the issue (and also added a non-regression test for it).

To get a fixed version, you may pull the current state of the repo and rebuild. You may also wait until tomorrow and grab the nightly build.

In the meantime, the solution could be to use option `--audio` which selects components by language. However, it seems that the audio components in your stream have no language set.

The other solution is to explicitly remove the _other_ audio PID's from the PMT, for instance:
~~~
tsp ... -P zap Hungama -P pmt -r 0x0480 -P rmorphan -O ip ...
~~~

udhayNuk, 2023-09-09:
@lelegard 
thank you for reply

udhayNuk, 2023-12-16:
@lelegard 
how to select the single to output.
i have the two audio track, but in one track there is two audio, how to select only single audio using pid.
[image]
please reply.

lelegard, 2023-12-16:
Please do not reopen old and unrelated issues. If you have a new question, open a new issue.

> i have the two audio track, but in one track there is two audio, how to select only single audio using pid.

I don't understand. You have two audio PID's. Each audio PID contains only one audio track. What makes you think there are "two audio" in the same PID?

udhayNuk, 2023-12-19:
sorry @lelegard to reopen the issue.
thanking you for your reply.
> What makes you think there are "two audio" in the same PID?
@lelegard 
yes, there are two audio in same pid.
when played in vlc track1 selected from audio, there are two audio played. and selected left, right option in audio stereo mode, one audio played.
how to map the one audio.
please reply

lelegard, 2023-12-19:
> when played in vlc track1 selected from audio, there are two audio played. and selected left, right option in audio stereo mode, one audio played.

I think that there is confusion between audio _track_ and audio _channel_. A stereo audio track is one single _track_ with two _channels_. 5:1 audio tracks have 6 channels, some modern audio codecs have even more channels for more loudspeakers in the room.

A stereo track is one single track because all channels are supposed to be played at the same time on distinct loudspeakers. The way multiple channels are mixed inside the audio track is codec-dependent. TSDuck can separate tracks in distinct PID's, not channels inside the same track. You need an audio transcoding tool to do this, ffmpeg for instance.

udhayNuk, 2023-12-19:
@lelegard 
thank you for replying.
there are two track, one track1 have single language audio.
second track2 have multiple laguages in a single track.

lelegard, 2023-12-19:
> second track2 have multiple laguages in a single track.

I don't think that this is possible with MPEG-2 audio. Please provide a dump of the PMT of the service (`tstables -p 0x049C`)

closed 2024-05-19 by github-actions[bot]