tsduck/tsduck #1294: [BUG] tsmux really slow for no reason when remuxing ts and adding null padding

trollcop, 2023-09-13:
#### Bug description:

I'm trying to remux a ~13mbps TS into 52.17Mbps ts for transmitting via DekTek modulator.
It's running using the command line below, but the disk and CPU usage are barely even touching system's capabilities.
It's almost (?) as if it's running at real-time? Not sure.

#### How to reproduce:

`tsmux -v --bitrate 52170000 -I file bstokyo_orig.ts -O file bstokyo_52.ts`

#### Expected behavior:

Remuxing 50gb of .ts at something faster than realtime speed.

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.35-3258
 - Installation type: portable binary from github

#### Additional information:

lelegard, 2023-09-14:
Hi,

`tsmux` is not a final supported command. It is just experimental and unfinished works that I started long ago and could have evolved into some kind of software mux but I never found the time to do so.

- It is not documented in the user's guide
- It is not in the Linux binary packages, not installed using `make install`.
- It is not in the macOS homebrew package.
- It is not in the Windows binary installers.
- I just discovered (from your report) it accidentally slipped into the Windows portable archive. It should not be there and will be removed from the next portable packages.

trollcop, 2023-09-14:
Haha, I see.
Well, but is there a way to achieve what I'm trying to do  with some other tsduck tool, simply to remux an existing stream at higher bitrate with null padding?

SonnyWalkman, 2023-09-22:
I realtime TSmuxer I smell?

trollcop, 2024-01-30:
Allright, I'll go ahead and close this.

closed 2024-01-30 by trollcop