tsduck/tsduck #1295: [BUG]

neo7530, 2023-09-18:
#### Bug description:

Scrambler sends wrong CRC within CW using dvb-csa2 with entropy reduction.
#### How to reproduce:

Log-Output of my ecmg shows CWs. Calculating the checksum ( (cw[0] + cw[1] + cw[2]) % 256 = cw[3] / (cw[4] + cw[5] + cw[6]) % 256 = cw[7]

```
what i get from scrambler => ecmg:  CW1: 48 9e d5 2f aa 7b 85 4c 
what i should get from scrambler => ecmg CW1: 48 9e d5 bb aa 7b 85 aa
```

#### Expected behavior:

Some CI modules does the calculation / correction internaly, some don't. So with this (wrong) cw, i can't get any descrambled picture on the half of my ci modules.

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: Windows
 - OS version: 10/11
 - TSDuck full version: 3.35-3258
 - Installation type: official binary 

#### Additional information:

lelegard, 2023-09-18:
Hi @neo7530 

You provide no details about the ECMG and the scrambler. So I assume you use the TSDuck `scrambler` plugin and whatever ECMG.

In TSDuck, the entropy reduction is handled at cryptographic level. This means that, for a purely random 8-byte CW, the entropy reduction is applied just before encryption/decryption. The "official" CW is still the complete original 8-byte CW.

This is why the CW which is passed to the ECMG is not reduced.

This is what is done in TSDuck. That being said, the way entropy reduction shall be processed was never standardized and was very CAS-dependent (believe me, I worked for 4 different CAS vendors in the past). So, which equipment is aware of entropy reduction and what kind of information is passed in the ECMG and in the ECM's has always been CAS dependent.

First, the CAS needs to know (somehow) that entropy reduction is applied. In the early days of DVB, this was implicit since entropy reduction was legally mandated. Then, in the last 10 years, when operators started to realize that 48-bit DVB-CSA2 was no longer secure, they decided to restore the full 64-bit CW. But how to do this was never specified. This created a period of operational mess. After which each vendor implemented its own solution.

This could be:
- A parameter in ECMG to force the reduction of CW before insertion in ECM.
- A signalization parameter which is received by the smart card or secure SoC to reduce the CW before the decryption.
- A signalization parameter which is applied by the CAS firmware to the descrambler.

In short, there is no rule. You just have to understand what you are doing and apply the right policy (as CAS vendor).

Slightly off-topic, note that entropy reduction shall no longer be used these days. Cracking 48-bit DVB-CSA2 on the fly, in real time (with only a small time shift) is too easy. No operator should use entropy reduction in 2023.

neo7530, 2023-09-18:
Ah, thank you for this excelent description. So then i will steal some bits of the access criteria to do the signalization for the smart card to recalculate the reduction of entropy. 

Many thanks.

Kind regards
Marco.

closed 2023-09-19 by neo7530