tsduck/tsduck #1297: [FEAT] Randomised PCR Adjust

AWhateley, 2023-09-20:
#### Is your feature request related to a problem?
It is helpful to be able to test streams with randomised PCR jitter for certain debugging. Generating such streams on the fly can be difficult as many playout solutions will attempt to correct or recompute the PCR for a given CBR rate.

#### Proposed solution:
The PCR edit plugin argument -add-pcr currently applies a fixed +/- to all PCR, being able to specify a range and modify each outgoing PCR with a random value in this range would get the right behaviour.

#### Alternatives:
If a stream is CBR with null packets on 8191, it could be possible to drop random numbers of null packets between each PCR and simulate jitter this way.

#### Additional information:

lelegard, 2023-09-20:
This is not random, for sure. But a video PID in a CBR stream is usually a VBR PID, typically using statmux. Moreover, the number of packets in the PID between two PCR is not constant, and certainly not a multiple of 1000 (or whatever suppression interval you use).

To have a truly random drop of packet or alteration of PCR, you need to write a dedicated plugin. Quite easy if you master C++. A bit challenging otherwise.

AWhateley, 2023-09-20:
That's true, mostly the video content will be variable, unless it is some static image. You are right that this method is worth testing, I will try it out tomorrow and update you if it gets to the right ballpark.

> The PCR edit plugin argument -add-pcr currently applies a fixed +/- to all PCR, being able to specify a range and modify each outgoing PCR with a random value in this range would get the right behaviour.
But this method we have not mentioned, does implementing a range and some pseudorandom selection require a large overhaul of the plugin?
Unfortunately I am not a master of C++, maybe I should start learning...

lelegard, 2023-09-23:
Option `--random` has been added to `pcredit`.

AWhateley, 2023-09-25:
@lelegard I have tried the improvised method with filter and also the new pcredit solution today. Both work well but the better control with pcredit is much appreciated.
Thank you again for such fast improvements. I will close the ticket now.

closed 2023-09-25 by AWhateley