tsduck/tsduck #1298: How store the accurately 10 minutes content

Developer-Balaji, 2023-09-26:
Hello @lelegard 

```
tsp  -v -m --synchronous-log -I http --ignore-errors --receive-timeout 300000 --infinite $INPUT_URL \
 -P analyze -i 30 -o report.txt  \
 -P regulate \
 -O hls --duration 2 --live 300 -p index.m3u8 index.ts
```

I was use this script Tried to Store the 10 minutes content , but stored content was not accurately 10 minutes ( 10minutes and 1 minutes extra ),
so i was check and played the all `.ts` files manually, Some` .ts` files was accurately 2 seconds and some files was 1 seconds only ,
How store the accurately 10 minutes content

lelegard, 2023-09-28:
> How store the accurately 10 minutes content

You use a quite complicated way of doing this. Moreover, the processing will never stop and you have to kill the command after you have 300 segment files.

When you write `-O hls --duration 2`, you request to fill each segment with approximately 2 seconds of contents. By default, 2 seconds means the number of TS packets for 2 seconds according to the measured bitrate. The bitrate is evaluated from the PCR. If your stream is VBR, then there is a shift in time between the PCR and the evaluation process (which is an average). So, the computed bitrate used over 2 seconds is not exactly 2 seconds. Depending on the VBR dynamics, the difference can be significant. For each segment, restarting from 0 at each segment. The inaccuracy being accumulated over 300 segments, the probability that you get exactly 10 minutes of contents over the 300 segments is close to zero.

If your input is regulated, you get 10 minutes of contents using:
~~~
tsp .... -P until --seconds 600 -O file out.ts
~~~

Developer-Balaji, 2023-09-29:
Okay @lelegard  Thankyou ..

`tsp .... -P until --seconds 600 -O file out.ts
`
But I need HLS Live stream , to store continuously 10 minutes without stopping the stream

Purpose ( I want store the 10 minutes live content , to play live content after 10 minutes (10 minutes delay live stream) )

lelegard, 2023-09-29:
If you want time shift, have a look at the `timeshift` plugin.

Developer-Balaji, 2023-09-29:
Thankyou @lelegard

`-P timeshift --time milliseconds`

I want to time shift 3.5 Hours incase Streaming server was Off or Reboot , the time shift was not working correctly
please give me a solution.

lelegard, 2023-09-29:
> I need time shift 3.5 Hours

You wrote you needed 10 minutes. Anyway, convert it to milliseconds.

Otherwise, I have no other solution.

Developer-Balaji, 2023-09-29:
3.5 Hours time shift Its Not Possible ?

> You wrote you needed 10 minutes. Anyway, convert it to milliseconds.

Okay @lelegard 10 minutes time shift , incase  Streaming server was Off or Reboot , the time shift was not playing correctly

Developer-Balaji, 2023-10-03:
Thankyou @lelegard

closed 2023-10-03 by Developer-Balaji