tsduck/tsduck #1299: [FEAT] ECM for fixed CW

sichunqin, 2023-09-28:
Currently, ECM could not be generated when fixed CW is used. 
For test purpose, I would like to use fixed CW and ECM. Any idea to solve the problem? Thanks!
#### Is your feature request related to a problem?

#### Proposed solution:

#### Alternatives:

#### Additional information:

lelegard, 2023-09-28:
This is not currently possible because this is not a realistic use case.

It depends on what you want to test and why. If using a fixed CW and ECM is really required, you can generate the ECM offline once using `tsgenecm`. Then, you inject this ECM using the plugin `inject` and update the PMT using the plugin `pmt`.

sichunqin, 2023-10-07:
It helps. Thanks for the reply!

sichunqin, 2023-10-07:
As suggested, Using plugs can solve the problem.

closed 2023-10-07 by sichunqin