tsduck/tsduck #1300: Source code Compilation Error

Developer-Balaji, 2023-09-28:
@lelegard 

I was tried to compile the Current TSDuck version Source code 
But i get Error 

```
base/network/tsSRTSocket.cpp:352:10: fatal error: srt/access_control.h: No such file or directory
  352 | #include <srt/access_control.h>

make[3]: *** [../../Makefile.inc:599: /home/user/tsduck-master2/bin/release-x86_64-user-ProLiant-DL360-Gen9/objs-libtsduck/tsSRTSocket.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:37: default] Error 2
make[1]: *** [Makefile:61: default] Error 2
make: *** [Makefile:235: deb-tools] Error 2

```

[image]

How to Solve this Error to create Bin files

Developer-Balaji, 2023-09-29:
`Which version of the TSDuck source code do you use?
`
tsduck-3.36-3292

`Which operating system version do you use?
`
Description:    Ubuntu 20.04.5 LTS
Release:          20.04
Codename:     focal

`Which version of SRT is installed?
`
ii  libsrt-dev:amd64                           1.4.0-1build1                                 amd64        Secure Reliable Transport UDP streaming library

lelegard, 2023-09-29:
OK, so you use the current state of the TSDuck source code with an old version of SRT. That won't work.

You have to decide which is more important to you:
- the version of the operating system (and accept the version of SRT which comes with it)
- the version of SRT (and accept the version of Ubuntu on which it is available)
- the version of TSDuck, for whatever new feature you might like (and find the versions of Ubuntu and SRT on which it works)

Based on your choice, you will have to pull the corresponding two others. An arbitrary combination of versions of the three may work, or not, as you experience.

If Ubuntu 20 is a requirement for you, we have to find the corresponding older version of TSDuck which proved to work on it. Browsing back in time in the TSDuck release area, the most recent version which was built on Ubuntu 20 is [3.26](https://github.com/tsduck/tsduck/releases/tag/v3.26-2349), from Apr 2021. You may try to build more recent versions of TSDuck on Ubuntu 20 (and SRT 1.4.0). It may work. Or not, like current version 3.36. I no longer have any Ubuntu 20 to test. Current versions of TSDuck are built on Ubuntu 23.

lelegard, 2023-09-29:
I found an old Redhat 8 VM with SRT 1.4.1 and I implemented a compilation workaround in the TSDuck source code. You may try with SRT 1.4.0.

Developer-Balaji, 2023-09-29:
Okay @lelegard 

Thankyou !!!

closed 2023-09-29 by Developer-Balaji