tsduck/tsduck #1301: [BUG] Compilation error/warning

lars18th, 2023-09-29:
#### Bug description: && Errors and logs:

```
  [CXX] base/xml/tsxmlDocument.cpp
base/xml/tsxmlDocument.cpp: In copy constructor ‘ts::xml::Document::Document(const ts::xml::Document&)’:
base/xml/tsxmlDocument.cpp:49:1: error: base class ‘class ts::StringifyInterface’ should be explicitly initialized in the copy constructor [-Werror=extra]
 ts::xml::Document::Document(const Document& other) :
 ^~
cc1plus: all warnings being treated as errors
../../Makefile.inc:598: recipe for target '/home/user/tsduck-3.35-3258/bin/release-x86_64-debian/objs-libtsduck/tsxmlDocument.o' failed
make[2]: *** [/home/user/tsduck-3.35-3258/bin/release-x86_64-debian/objs-libtsduck/tsxmlDocument.o] Error 1
Makefile:37: recipe for target 'default' failed
make[1]: *** [default] Error 2
Makefile:61: recipe for target 'default' failed
make: *** [default] Error 2
```

#### How to reproduce:

```
# gcc --version
gcc (Debian 6.3.0-18) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

#### Environment:
 - OS: Linux
 - OS version: Debian 7
 - TSDuck full version: 3.35-3258
 - Installation type: build from source

#### Additional information:

I know that you don't want to maintain compatibility with _all_ ancient/legacy compilers. However, I feel this warning-error perhaps requires attention.

Regards.

lars18th, 2023-09-29:
And identical error compiling with GCC 4.9 (that is supported).

lelegard, 2023-09-29:
Hi @lars18th 

This was reported in #1285 and #1288. This is a GCC bug. A workaround was added in the source code of TSDuck. Please use the latest state of the repo.

lars18th, 2023-09-29:
Hi @lelegard ,

> This was reported in #1285 and #1288.

Sorry! 😄

closed 2023-09-29 by lars18th