tsduck/tsduck #1302: [BUG] Another compilation error with GCC 4.9 --> missing initializer for member ‘std::array<ts::PcapStream::Stream, 2ul>::_M_elems’

lars18th, 2023-09-29:
#### Bug description && Errors and logs:

```
In file included from base/network/tsPcapStream.cpp:30:0:
base/network/tsPcapStream.h:191:41: error: missing initializer for member ‘std::array<ts::PcapStream::Stream, 2ul>::_M_elems’ [-Werror=missing-field-initializers]
         std::array<Stream, 2> _streams {};
                                         ^
  [CXX] plugins/infra/tsPluginEventData.cpp
cc1plus: all warnings being treated as errors
../../Makefile.inc:599: recipe for target '/home/user/tsduck-3.36-3279-git/tsduck-master/bin/release-x86_64-debian/objs-libtsduck/tsPcapStream.o' failed
make[2]: *** [/home/user/tsduck-3.36-3279-git/tsduck-master/bin/release-x86_64-debian/objs-libtsduck/tsPcapStream.o] Error 1
```

#### How to reproduce:

```
# make GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOCURL=1 NOPCSC=1 NOEDITLINE=1 NOSRT=1 NORIST=1
```

#### Environment:
 - OS: Linux
 - OS version: Debian 7
 - TSDuck full version: git clone at 2023-09-29
 - Installation type: build from source

#### Additional information:

I've followed this suggestion...

>  A workaround was added in the source code of TSDuck. Please use the latest state of the repo.
https://github.com/tsduck/tsduck/issues/1301#issuecomment-1740774999

lelegard, 2023-09-29:
Hi @lars18th 

This one is a bit different. The others were bugs in GCC 8.5 which missed default constructors in initialiser lists.

This one is in a default member initialiser list. I have started a background refactoring of the TSDuck code to adapt to more modern C++ features and styles. This is a required condition to keep the code alive for the future. It seems that GCC 4.9 has an issue here. I see no obvious workaround without reverting the modernisation of the code, which I refuse to do.

I have not found any binary package for GCC 4.9 for modern distros. I tried to rebuild GCC 4.9 from sources and, after some time, it does not even compile on modern compilers. So, I cannot even test it.

GCC 4.9 is 10 years old now and, at some point, we need to upgrade.

I cannot support it. If you are forced to use an old OS and an old compiler, you will face many restrictions and you will probably be forced to use old products. TSDuck is no exception. There are older versions of TSDuck which compiled with GCC 4.9, although I do not remember having using it. I think most people jumped from 4.8.5 to 5.x or 6.x a long time ago.

Sorry, but there won't be any TSDuck 3.36 with GCC 4.9, simply not the same generations of software.

lars18th, 2023-09-30:
Hi @lelegard ,

I agree with your point of view. The idea of going forward is pragmatic. The only reason to use GCC 4.9 is because years ago you've commented of this version as a "legacy" compiler. In my "old" environments the compiler is GCC 6.3 (Debian 7). This version has created troubles with older versions of TSDuck. However, now it compiles without troubles.

In summary:
- GCC 4.9: Last version of TSDuck that compiles is v3.34-3197.
- GCC 6.3: Current master branch compiles with it.

Perhaps (or not) you will want to maintain a list of compatible compilers and versions.
Thank you. You can close the issue when you want. 😉

lelegard, 2023-09-30:
> Perhaps (or not) you will want to maintain a list of compatible compilers and versions.

I will not. I would be obliged to maintain a set of compiler binaries and make tests all the time. I consider this as a waste of time. Just move forward, not backward.

> GCC 6.3: Current master branch compiles with it.

Good to know it compiles with GCC 6 but I am more interested in GCC 12 and 13 😉

closed 2023-09-30 by lelegard