tsduck/tsduck #1303: how to encrypt using the aes and read the key from a file[Q]

udhayNuk, 2023-09-30:
@lelegard 
this is my command line script for streaming hls
i want encrypt using aes plugin and read the key from file.
KEY=head - key (this is not working, how to read the key from a file).
tsp -v -m --synchronous-log -I http --ignore-errors --receive-timeout 300000 --infinite $INPUT_URL \
 -P analyze -i 30 -o report.txt  \
 -P aes --key $KEY --cbc "service" \
 -O hls --duration 2 --live 15 -p index.m3u8 index.ts \
suggest any methods to read the key from a file. 
its possible or not.

lelegard, 2023-09-30:
Hi @udhayNuk 

This is a shell syntax and usage issue, not a TSDuck issue.
~~~
$ tsp -P aes --help
...
  -k hexa-data
  --key hexa-data
      Specifies a fixed and constant AES key for all TS packets. The value must
      be a string of 32 or 64 hexadecimal digits. This is a mandatory
      parameter.
~~~

So, just make sure that the environment variable contains 32 (AES-128) or 64 (AES-256) hexadecimal digits. The real question is what does your file contain? Binary? Hexa? Something else? In case of doubt, just use `xxd` on it. There are plenty of solutions and commands to convert data formats on Linux.

udhayNuk, 2023-09-30:
@lelegard 
thanking you for reply 
iam using hexa-data in a file(key)  9c63891e2d63e8169c63891e2d63e816  .
how to read the key from a file for encryption.
please reply.

lelegard, 2023-09-30:
In that case, use `-P aes -k $(cat keyfile.txt)`.
It works with bash. Other shells may have slightly different syntaxes.
There are other forums which are more appropriate for discussing shell usage.

udhayNuk, 2023-09-30:
@lelegard  thanking you

closed 2023-09-30 by udhayNuk