tsduck/tsduck #1304: ffmpeg TS to DVB including SDT, NIT TDT & TOT [Q]

SonnyWalkman, 2023-09-30:
I have a ffmpeg generated from OBS studio record output a 4.5M CBR TS multicast to TSduck.  With a simple 
```
tsp ^
-I 224.0.0.1:9999 ^
-O localhost:5000  --packet-burst 7 --enforce-burst
```
Works fine..  

Once I start modifying and adding NIT, TDT and TOT tables missing (TDT/TOT) in ffmpeg. I monitor the output with VLC udp://@:5000, the feed is jerky and hangs. In addition using -O Play does the same thing?
 
Am I implementing inject incorrectly? 

Has anyone successfully used TSduck to make ffmpeg DVB compliant before feeding into a dektec DTA-115 or a Hides UT-100C USB?
Once I have my TS working, I'll swap the -O ip with -O dektec --frequency ...

Here's my tsmux.bat script.

```
@echo off
tsp  --add-input-stuffing 1/50 --bitrate 4500000 ^
    -I ip 224.0.0.1:9999 ^
    -P sdt --service-id 1 --name VK3RGL-1 --provider vk3atl.org --type 0x01 ^
    -P nit --create --build-service-list-descriptors --network-id 1 --network-name "Amateur-DATV" --increment-version ^
    -P inject C:\temp\tables\TDT.xml=500 C:\temp\tables\TOT.xml=1000 --pid 20 -s -b 15000 ^
    -P analyze -i 30 -o spts.txt ^
    -O ip localhost:5000 --packet-burst 7 --enforce-burst
pause
```

My TOT.xml and TDT.xml renamed to txt to allow uploading.

[TDT.txt](https://github.com/tsduck/tsduck/files/12775064/TDT.txt)
[TOT.txt](https://github.com/tsduck/tsduck/files/12775065/TOT.txt)

Here's my analysis file.

[spts.txt](https://github.com/tsduck/tsduck/files/12775042/spts.txt)

lelegard, 2023-10-01:
Hi @SonnyWalkman 

It should work, I see no reason why the second command introduces delays.

Concerning TOT and TDT, I usually inject fixed XML files with a dummy date. Then, I add `-P timeref --system-synchronous` and it works fine.

You may try `tsp --max-flushed-packets 7` to make sure that no internal delays are added.

The option `--bitrate` does not seem necessary, although it shouldn't hurt.

If you still have an issue, then start from the first command (which works smoothly) and add each plugin one by one to find which one creates the issue.

SonnyWalkman, 2023-10-02:
Hello @lelegard,
I've taken your advice and injected XML files statically and used `timeref` plugin to update both TDT and TOT.  
Both TDT and TOT update every second from analysing TDT and TOT PSI's using dvbinspector. 👌🏻

The tsp `--max-flushed-packets 7` didn't improve VLC stutter of both  video and sound (lost packets) after 3-4 seconds. Seems    VLC is exceptionally picky on playing my TS? 

I used **ffplay udp://127.0.0.1:5000**, which fortunately plays perfect therefore, I feel my windows VLC is the issue here??  

**Another question if I may?**  
How do I correctly use TOT to change time for Eastern DST?  Day Light Savings time here in Victoria Australia begins at 2am on the first Sunday in October and finishes or ends at 2am (which is 3am Daylight Saving Time) on the first Sunday in April?  `--local-time-offset minutes --next-change value` and `--next-time-offset minute`

Now I'll look at adding a simple EIT..

**Realtime muxer and Teletext?**  🙄

I'm deeply interested in working on **teletext injection** as my final endeavor. While I recognise that the current merge functionality isn't tailored for real-time multiplexing, I remain hopeful that in the future, a real-time merger can be introduced. This merger would adeptly handle the merging of multiple transport streams with diverse PCRs, efficiently buffering and transitioning each SPTS to a unified MPTS PCR. With such advancements, TSDuck could truly emerge as the premier choice for TS manipulation.

Recognising the potential and importance of this feature, I am prepared to provide financial support for its development, given there's a genuine interest in crafting this solution. It's clear from numerous inquiries that many perceive TSDuck primarily as a tool and not for its real-time muxing capabilities. My research indicates that the only open-source multiplexer available is Opencaster, which sadly, isn't actively maintained anymore. On the other hand, proprietary solutions such as Dektec's MuxXpert and MainConcept's MPEG-2 TS Broadcast Delivery Plugin for FFmpeg are prohibitively expensive for Amateur DATVer such as I?

In my quest for alternatives, I stumbled upon [this repository](https://github.com/gfto/mptsd). It appears to hold promise. Could it potentially act as a foundation for converting from C to C++?

lars18th, 2023-10-05:
Hi @SonnyWalkman ,

To solve your troubles when playing with VLC you can try:

1. Play directly the origin TS with VLC. I discovered that some encoders (including FFmpeg) generate incompatible streams for VLC.
2. Remove the `--bitrate` parameter, it's unnecessary. You're are reading from a CBR stream, and TSDuck could calculate it.
3. Try first with a simple `tsp -r --add-input-stuffing 1/50 -I ip 224.0.0.1:9999 -O ip localhost:5000 --packet-burst 7`. The realtime processing of a DVB TS is very sensittive to delays and jitter.
4. Ensure that your encoder is adding the PCR marks at the regular interval (40ms for DVB).

More or less from my experience I see that opensource encoders have troubles generating Broadcast Streams. As a consequence the TSDuck can't do anything to solve it. It's preferable to first generate a _clean_ CBR TS and after "edit/tweak" it using TSDuck. I'm sure that others (like @lelegard) will agree with that strategy.

I hope it helps.

SonnyWalkman, 2023-10-05:
Hi @lars18th ,
Thanks for your valuable input..

> Play directly the origin TS with VLC. I discovered that some encoders (including FFmpeg) generate incompatible streams for VLC.

If you read my first sentence.. 
> I have a ffmpeg multicast TS generated from OBS studio record. The TS 4.5Mb/s CBR TS multicast which TSP is listening to.

With this simple example:-

```
tsp ^
-I 224.0.0.1:9999 ^
-O localhost:5000  --packet-burst 7 --enforce-burst
```

Works fine.. No PTS and DTS issues..

ffmpeg plays perfectly fine through VLC using the multicast address upd://@224.0.0.1:9999 Even the basic in and out with TSduck above works without a hitch as to be expected and rules out ffmpeg being the issue.

Over the last few days, I have changed the process and have successfully been able to get my application working PTS and DTS discontinuity free and able to inject tables, although my chosen inject rates are possibly incorrect.. Guidance appreciated?

```
@echo off
set BITRATE=10053476
set /a REGBITRATE=%BITRATE%*97/100
set TABLEPATH=C:\temp\tables\

tsp ^ 
-b %BITRATE% ^
-I null ^
-P regulate -b %REGBITRATE% ^
-P merge ^
-t "tsp -I ip 224.0.0.1:9999" ^
-P sdt --service-id 1 --name YOUR-SERVICE --provider YOUR-PROVIDER  --type 0x01 ^
-P nit -c --build-service-list-descriptors --network-id 1 --network-name YOUR-NETWORK ^
-P inject %TABLEPATH%TDT.xml=1000 %TABLEPATH%TOT.xml=1000 --pid 20 -s -b 1504 ^
-P timeref --system-synchronous --eit --local-time-offset 600 --next-change "2024-04-14 03:00:00" --next-time-offset 660 --only-country AUS ^
-P eitinject -f %TABLEPATH%EIT.xml --terrestrial --time system -b 1504 ^
-P analyze -i 30 -o ts.txt ^
-O hides -f 690500000 --gain 0 -c QPSK -b 7 -h 5/6 -g 1/32 
pause
```

Like any forum, to be able to be guided and receive usable feedback benefits everyone who want to know and learn how use this marvellous tool.  I've worked many hours to get this fa by experimentation and head scratching.

However, I feel there is more to be done with TSduck real world examples. 

Not everyone here has worked in broadcast industry and has a full understanding of DVB transport system. I worked for a local broadcaster back in 1992-1998 which was all analogue CVBS. however, I've chosen to continue to learn and experiment in my passion for Amateur DATV here in Australia and continue to understand what goes on in the DVB transport system.
I have the TSduck users guide under my pillow eveynight. 

To be honest, I've gain more information trawling this forum than reading the user guide. 

 I understand @lelegard has done an excellent job and are grateful for his many hours 1000's of work and effort and shall continue to support this great Tool.. Only if there was a Real time muxer plugin TSduck would raise from a tool to a full DVB suite.

closed 2023-10-05 by SonnyWalkman

lars18th, 2023-10-06:
Hi @SonnyWalkman ,

Great if you can generate now your broadcast without troubles. The first learned point is that you (or anyone else) needs to check (or start) with a simple PIPE in TSDuck (like the `tsp -I 224.0.0.1:9999 -O localhost:5000  --packet-burst 7 --enforce-burst`). The main reason is to discard any network/system miss configuration or incorrect/incompatible infrastructure. From this point you can start to _process_ the TS using the different filters.

But, regarding your specific current processing:
- I don't understand why you are producing and empty TS (nulls) with a CBR (regulated) and "merging" the original TS. This strategy works (of course). However, I feel it's more robust the other strategy of consuming the original CBR TS and add the padding stuff (with `--add-input-stuffing`).

Perhaps @lelegard could plain why the first works fine and the second can produce errors. Any idea?
Regards.

SonnyWalkman, 2023-10-06:
Hello @lars18th 

At first I tried the null stuffing approach trying different percentages 1/20 was a good compromise. However, we merging the ffmpeg stream, and analysing the results there where many discontinuities which was visible with and audible clicks. So far my approach, I calculate the modulation bitrate with tsterinfo to get what us required for 7Mhz QPSK..

I had to adjust ffmpeg video encoder for 25 key frames.

I'm not expert, however after doing MORE reading on closed issues, I found @lelegard response to a question to use regulate closest to the -O output.
I'm using a Hides UT-100C however, I have a Dektec DTA-115 which fortunately is able to bitstuff setting to regulate the bitstream to fill in the caps if I set the full bitrate lower by 2% - 3%

I may try going back to using bit stuffing the incoming udp TS as per your suggestion and give it another try.

So far I'm happy to have something working. Furthermore, I teletext working developing a little golang program to manage a carousel of txt pages dynamically generating t42 bin stream and adding the nectary PES headers. I'm working on text pages updating header clock and printing PC stats and squirt into the TS at a defined time (yes it creates it own PCR time stamps) using another TSduck merge with sone settings -b 4700 PCR begin 0 and count 3600 each time and how many pages.
Be nice to have real-time mux?

It works. Here's a picture to prove it. 

[image]

SonnyWalkman, 2023-10-06:
Furthermore,
Is there any ratio calculation to determine the injection bitrate to inject tables knowing their a fix section size?
Be great to know the maths so I can use it?
 I'm using -b 1504 which maybe incorrect?

What's the deal? Opencaster explains to some degree however nothing mention in TSduck guide?  It's not just a 'pick a figure and hope for the best'??