tsduck/tsduck #1309: how to create a ts segment name added with current time?[Q]

udhayNuk, 2023-10-04:
Ridsys-RStreamer_Engine -v -m --synchronous-log -I ip -r 231.1.1.44:1026 \
 -P analyze -i 30 -o report.txt  \
 -P regulate \
 -O hls --duration 2 --live 200 -p index.m3u8 index.ts
 
 my index.ts is creating like index -00001.ts.
 i want to create the index.ts with current time like index -12:01ts.
 @lelegard please reply me.
 is there any way to create the hls segments like this.

SonnyWalkman, 2023-10-04:
Hello @udhayNuk ,

Filenames generally shouldn't have colons (:) in them because this character has a special meaning in many file systems and can cause issues. Instead, you might want to use another character, like a hyphen (-) or an underscore (_).

```
#!/bin/bash

while true; do
    # Wait for the next segment to be created
    inotifywait -e create /path/to/hls/folder

    # Rename segments
    for file in /path/to/hls/folder/index-*.ts; do
        mv "$file" "/path/to/hls/folder/index-$(date +%H-%M).ts"
    done
done
```

This script utilises inotifywait (from the inotify-tools package) to wait for file creation events in the specified folder. When a new segment is created, it immediately renames it using the current hour and minute.

However, be cautious: this approach might have race conditions or could miss files in high throughput scenarios

lelegard, 2023-10-04:
@udhayNuk 

> my index.ts is creating like index -00001.ts.
> i want to create the index.ts with current time like index -12:01ts.

There is no possibility to do this in TSDuck. The name of the segment files are technically irrelevant. They are referenced in the playlist. Only the name of the master playlist is important because it is used as URL for the stream.

There is no technical reason, other than aesthetics, to specify the time in a segment file name. The usage is to use a fixed prefix and incrementing numbers.

@SonnyWalkman 

Please do not recommend that. It breaks the HLS structure. The master playlist will then point to non-existent files and the clients will play nothing.

udhayNuk, 2023-10-04:
@SonnyWalkman 
i dont want shell scripts.
i want in tsduck any another methods to write the current time also in ts segments.

udhayNuk, 2023-10-04:
@lelegard 
thanking you for reply

closed 2023-10-04 by udhayNuk