tsduck/tsduck #1310: [Q] Adding EIT (EPG) to a TS? Not showing on SONY TV Any advice from the PG experts.

SonnyWalkman, 2023-10-04:
Hi all TSduck  aficionado's,

I'm currently developing of my Windows 11 laptop but plan to transition to Linux once I'm satisfied with my code. I'm using Go which I've coded to run on either Windows or Linux OS.

The multicast IP, is the output from OBS studio which uses ffmpeg API to interact  ffmpeg directly. The transport stream produce plays flawlessly on both VLC and ffplay. Example: `ffplay udp://244.0.0.1:9999`

I'm using a batch file for testing however; my Go routine creates a shell to instigate TSduck.

The transport stream from ffmpeg provides the foundational PSI's, including valid PMT, PAT, and SDT. My code enhances it by adding a service name and provider, something OBS record output has yet to accommodate. I add an NIT and TDT and TOT which then is updated with `timeref`

At startup, I employ `tsterinfo` to gauge the stream bitrate for modulation parameters, rounding it to a integer, resulting in 10053476 bps which I've hardcoded in my batch file for testing.

My strategy is to first produce a stream of Null packets, followed by table injections, and finally merge the incoming ffmpeg stream, replacing the Null packets. This approach effectively prevents? discontinuity errors. My first method of not conforming to the modulation bitrate requirements was a lesson learnt.

I alternate -O  between file storage for analysis and routing through a Hides UT100-C.

I'll be upgrading to a more robust solution, likely my Dektec DTA-115, when I shift to the Linux system since the UT100-C is Mickey Mouse 🐭.

Upon startup, my program auto-generates TDT.xml, TOT.xml, and EIT.xml based on the current system time, scheduling a 30-minute interval for the next event fora test scenario. The original_network_id I use is "0xFF01", sourced from what ffmpeg allocated . 

My code updates the EIT every 30 minutes interval with new event text for the program.

Any assistance is welcome on how to get the EPG working on my SONY TV.

Here's my tsduck.bat file..

`@echo off
set BITRATE=10053476
set TABLEPATH=C:\temp\tables\

tsp ^ 
-b %BITRATE% ^
-I null ^
-P regulate -b %BITRATE% ^
-P merge ^
-t "tsp -I ip 224.0.0.1:9999" ^
-P sdt --service-id 1 --name VK3RGL-1 --provider vk3atl.org --type 0x01 ^
-P nit -c --build-service-list-descriptors --network-id 1 --network-name Amateur-DATV ^
-P inject %TABLEPATH%TDT.xml=1000 %TABLEPATH%TOT.xml=1000 --pid 20 -s -b 1504 ^
-P timeref --system-synchronous --eit --local-time-offset 660 --next-change "2024-04-14 03:00:00" --next-time-offset 600 --only-country AUS ^
-P eitinject -f %TABLEPATH%EIT.xml --terrestrial --time system -b 1504 ^
-P analyze -i 30 -o ts.txt ^
-O hides -f 690500000 --gain 0 -c QPSK -b 7 -h 5/6 -g 1/32 
pause`

Here's my generated EIT.xml file.

```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
    <EIT actual="true" current="true" last_table_id="0x50" transport_stream_id="0x0001" original_network_id="0xFF01" service_id="0x01" type="pf" version="6">
        <!-- Present event -->
        <event event_id="0x0001" start_time="2023-10-05 07:39:16" duration="00:30:00" running_status="running" CA_mode="false">
            <short_event_descriptor language_code="eng">
                <event_name>Test Event</event_name>
                <text>My Test message</text>
            </short_event_descriptor>
        </event>
        <!-- Following event -->
        <event event_id="0x0002" start_time="2023-10-05 08:08:16" duration="00:30:00" running_status="not-running" CA_mode="false">
            <short_event_descriptor language_code="eng">
                <event_name>Next Test Event</event_name>
                <text>My Test message - Up next!</text>
            </short_event_descriptor>
        </event>
    </EIT>
</tsduck>

```
Other outputs for anaylsing the transport stream.

[image]
[image]
[image]

SonnyWalkman, 2023-10-06:
Hello All,
I had to include injection of a RST.xml  Running Status Table)  alongside the EIT.xml table to make the TV show my EPG.. 
Not sure why?

SonnyWalkman, 2023-10-23:
Any takers on how to set 24 hour period of EPG schedule?  An example EIT.xml could help.

SonnyWalkman, 2023-10-24:
Hello again!

First off, let me say that it's absolutely okay to ask questions. You're diving deep into the world of MPEG-TS and DVB standards which are, to put it mildly, vast and intricate. While the TSDuck toolset provides immense flexibility, it does assume a level of understanding of the underlying standards. So, your feedback about the documentation is certainly valid. However, it's also worth noting that the development and maintenance of such a toolkit is no small feat, and often the documentation may not cover all possible use-cases or real-world scenarios.

Now, onto your questions:

Inconsistencies between TV's and Set-top boxes: This is, unfortunately, a common issue. Manufacturers might not always interpret or implement DVB standards in the same way. The same EIT data might be parsed differently by different devices. This isn't really a TSDuck issue, but more of a broader standard implementation problem.

File Updates in eitinject: TSDuck continuously monitors the file for changes. As for what qualifies as a "change", it typically uses file system notifications. So, you'd need to make sure your changes are being saved in a way that triggers these notifications. Typically, a write or save operation on the file would be enough.

Options: Yes, the EIT type and last_table_id are some of the options that can control the epg . Is see 0x4E, 0x50. Do these determine which kind of EIT table you're generating (e.g., "present/following" or "schedule").  type="p/f" or type="0"?

Event ID:   (event_id) in the EIT table is a 16-bit field, do I need to keep tabs on the actual event ID range 1 to 65,536. Can I simply reset it every 24 hours on a new day schedule or does the ID increment until it reaches the max and then roll it over back to 1. 
I suspect receivers might not react well if they see event IDs changing unpredictably. It might be safer to let it roll over naturally. I'm of course open to what I need to be doing here. I might not implementing/reading the DVB specifications correctly.

Lastly, regarding my epg.xml, 

Could you agree or disagree if it adheres to the expected format? 
Are my fields valid values? 

Feel free to share a working epg I could learn from?  

```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
    <EIT actual="true" current="true" last_table_id="0x50" transport_stream_id="0x0001" original_network_id="0xFF01" service_id="0x01" type="pf" version="0">    
	<event event_id="1" start_time="2023-10-09 06:00:00" duration="00:00" running_status="not-running" CA_mode="false">
		<short_event_descriptor language_code="eng">
			<event_name>"WakeUp"</event_name>
				<text>"Start of the daily schedule"</text>
		</short_event_descriptor>
		<parental_rating_descriptor> 
			<country country_code="AUS" rating="0x05"/> 
		</parental_rating_descriptor> 
	</event>
	<event event_id="2" start_time="2023-10-09 07:00:00" duration="00:10" running_status="not-running" CA_mode="false">
		<short_event_descriptor language_code="eng">
			<event_name>"Play Carousel"</event_name>
				<text>"A short slide show on how repeater equipment and operation"</text>
		</short_event_descriptor>
		<parental_rating_descriptor> 
			<country country_code="AUS" rating="0x05"/> 
		</parental_rating_descriptor> 
	</event>
	<event event_id="3" start_time="2023-10-09 07:10:00" duration="00:10" running_status="not-running" CA_mode="false">
		<short_event_descriptor language_code="eng">
			<event_name>"Tower Camera"</event_name>
				<text>"View of Mt Anakie and the surrounding of Geelong"</text>
		</short_event_descriptor>
		<parental_rating_descriptor> 
			<country country_code="AUS" rating="0x05"/> 
		</parental_rating_descriptor> 
	</event>
	<event event_id="4" start_time="2023-10-09 07:20:00" duration="00:40" running_status="not-running" CA_mode="false">
		<short_event_descriptor language_code="eng">
			<event_name>"Repeater Test Charts"</event_name>
				<text>"The Philips PM5644 is the HD Philips Test Pattern"</text>
		</short_event_descriptor>
		<parental_rating_descriptor> 
			<country country_code="eng" rating="0x05"/> 
		</parental_rating_descriptor> 
	</event>
    </EIT>
</tsduck>
```

lelegard, 2023-10-24:
> Inconsistencies between TV's and Set-top boxes: This is, unfortunately, a common issue. Manufacturers might not always interpret or implement DVB standards in the same way. The same EIT data might be parsed differently by different devices. 

Yes, definitely.  There are no two identical receivers, even though the DVB standards are pretty precise.

I am not sure that _parsing EIT_ is an issue. However, the processing of the parsed EIT is certainly different between receivers.

> File Updates in eitinject: TSDuck continuously monitors the file for changes. As for what qualifies as a "change", does it typically uses file system notifications. 

Because of the differences between operating systems, TSDuck does not use file system notifications. The `eitinject` plugin polls the presence of files. By default, the wildcard specification for input files is checked every 500 ms. A change of existing file is detected when the last modification time or the file size has changed. In that case, the file is not processed immediately. Instead, once a file is declared as "changed", it must remain "stable" (same size and modification time) for another 500 ms. The idea is to avoid reading a modified file while it is updated.

So, an added or updated input file is loaded at most one second after being modified.

> Options: EIT type and last_table_id are some of the options that can control the epg . Is see 0x4E, 0x50. Do these determine which kind EPG is generating (e.g., "present/following" or "schedule"). type="p/f" or type="0"?

This is true for EIT in broadcast. As explained before (and I do recommend to read the `eitinject` chapter in the TSDuck user's guide), in the input XML files (or binary, or JSON) that you provide, the EIT envelope is discarded and ignored (except the original network id and transport stream id of course). Only the `<event>` substructures are kept and stored internally. You can use any EIT type, we don't care.

Then, the `eitinject` plugin will generate the right EIT's on time with the right events. As long as an event is not close, it will be sent in EIT schedule, actual or other, with the appropriate EIT table type. When the event is close, then current, it will be inserted in EIT p/f. This is automatic, this is the job of the `eitinject` plugin. Just provide the events in advance, the plugin will do the rest.

> Event ID: (event_id) in the EIT table is a 16-bit field, do I need to keep tabs on the actual event ID range 1 to 65,536. Can I simply reset it every 24 hours on a new day schedule or does the ID increment until it reaches the max and then roll it over back to 1.

The problem with resetting the event id every 24h hours is that some receivers may react very badly. An event is identified by 4 values: original network id + transport stream id + service id + event id. An EPG usually contains more than one day. This means that, in one service, distinct events (on distinct days) have the same 4 id values. Expect troubles with some receivers.

If you simply allocates event ids sequentially and wraps from 65535 to 0, you won't have problems. The maximum capacity of an EPG is 64 days by design, much less in practice. So, unless you have a full 64 days EPG and more than 1024 events per service, per day, you will never have an event id clash.

See the [description of the EITGenerator C++ class](https://tsduck.io/doxy/classts_1_1EITGenerator.html#details) for additional details.

> Lastly, regarding my epg.xml, Could you agree or disagree if it adheres to the expected format? Are my fields valid values?

As mentioned earlier, the EIT structure does not matter. Only the original network id and transport stream id shall be accurate. The rest (actual/other, current, table id) is ignored.

The event structures may contains anything, any descriptor. They are passed without change in the generated EIT's. What matters is what kind of descriptor is expected / accepted / mandatory / optional in each event for your receivers. This may be specified in some Australian OP document. And this may be more or less correctly implemented in the various receivers. 

For this, I cannot help. This is a local issue.

SonnyWalkman, 2023-10-26:
Thank again @lelegard,
> The problem with resetting the event id every 24h hours is that some receivers may react very badly. An event is identified by 4 values: original network id + transport stream id + service id + [...]

> If you simply allocate event ids sequentially and wraps from 65535 to 0, you won't have problems. The maximum capacity of an EPG is 64 days by design, much less in practice. So, unless you have a full 64 days EPG and more than 1024 events per service, per day, you will never have an event id clash.

I have opted to save the `event_id` value in an external event_id.xml file after each new day EPG generation to preserve the count.  If the program happens to crash or shut down between EPG generations retreving the count is always correct? Not sure how other programs handle counts?
I created a function to roll over the `event_id` count back to 1 after uint16 max value.  Hopefully as you have indicated, happing duplicate events now should be issue of the past?

Could you confirm, I have both TDT and TOT generation using UTC time NOT local. Using local created issues with receivers. Reading the DVB Spec's again indicates all TDT and TOT to use UTC! 
Each receiver has the ability to adjust UTC for local time i.e UTC + 11 for Victoria Australia. 

With EPG do all the start time need to adjusted to UTC or Local time?  I'm assuming the earlier (MUST BE  UTC)

Regards

SonnyWalkman

lelegard, 2023-10-26:
Within DVB, all times are UTC: TDT, TOT, EIT.

The local time offset is given in the TOT which contains a list of country codes (3 letters) / region id (one byte).

What the receiver should know is its region id. This is particularly important for large countries such as Australia which spans over several time zones. The receiver then computes its local time from the data in the TOT. It is important to make the difference because the region id is constant and configured once for good in the receiver, while the time offset may change twice a year with DST.

It is interesting to notice from the TOT of the streams you provided that some regions of Australia change time with DST twice a year, while other regions don't change. Isn't this difficult to manage in every day's life?

~~~
* TOT, TID 115 (0x73), PID 20 (0x0014)
  Short section, size: 120 bytes
  UTC time: 2023/10/21 11:02:36
  - Descriptor 0: Local Time Offset (0x58, 88), 104 bytes

    Country code: AUS
    Region id: 1 (0x01), polarity: east of Greenwich
    Local time offset: 10:00
    Next change: 2038/01/01 16:00:00  <=== dummy date
    Next time offset: 10:00  <============ no change

    Country code: AUS
    Region id: 2 (0x02), polarity: east of Greenwich
    Local time offset: 11:00
    Next change: 2024/04/06 16:00:00  <=== actual change date (UTC)
    Next time offset: 10:00  <============ change = -1h

    Country code: AUS
    Region id: 3 (0x03), polarity: east of Greenwich
    Local time offset: 11:00
    Next change: 2024/04/06 16:00:00
    Next time offset: 10:00

    Country code: AUS
    Region id: 4 (0x04), polarity: east of Greenwich
    Local time offset: 10:00
    Next change: 2038/01/01 16:00:00
    Next time offset: 10:00

    Country code: AUS
    Region id: 5 (0x05), polarity: east of Greenwich
    Local time offset: 10:30
    Next change: 2024/04/06 16:30:00
    Next time offset: 09:30

    Country code: AUS
    Region id: 6 (0x06), polarity: east of Greenwich
    Local time offset: 08:00
    Next change: 2038/01/01 16:00:00
    Next time offset: 08:00

    Country code: AUS
    Region id: 7 (0x07), polarity: east of Greenwich
    Local time offset: 11:00
    Next change: 2024/04/06 16:00:00
    Next time offset: 10:00

    Country code: AUS
    Region id: 8 (0x08), polarity: east of Greenwich
    Local time offset: 09:30
    Next change: 2038/01/01 16:00:00
    Next time offset: 09:30

  CRC32: 0x9160CE78 (OK)
~~~

DVB was specified in Europe where the local time offset remains small (-2 to +3h). Countries on the other side of the planet may find it inconvenient and break the rules. Japan's ISDB is a mixture of DVB and specific stuff. Many DVB tables and descriptors are directly reused in ISDB, but they change the representation of strings and dates. So, most TDT/TOT/EIT contain UTC, except in Japan where they use JST (Japan Standard Time). To add to the confusion, other ISDB countries in South America and Africa chose their own representation of strings and dates, neither DVB nor Japan's ISDB, but still using the DVB syntax for the tables and descriptors.

Fortunately, from the streams you provided, it seems that Australia follows the DVB rules.
Note that some countries

SonnyWalkman, 2023-10-27:
Hello @lelegard,
I've finally have everything working after taking onboard you notes..  I've learnt a few things on the way and here is my breakdown to assist with other whom fall on my forum question in future.

1. Keep everything UTC.   Both TDT and TOT must be set correctly for any receiver to process the EPG .  My application takes my daily schedule.yaml  file in Local time and offsets the start time to UTC. (-11 hours here in Victoria Australia).  EPG now lines up perfectly on Sony TV, Set top boxes and PC DVB-T software. 
3. EPG <event_id> count cycles the full uint16 max then rolls back to 1.  I store my ID in an external event_id.xml file and call it when creating the next 24 hours event.  All 4 receivers are happy and produce solid EPG. Don't assume resetting <event_id>back to 1 every 24 hours works.  Sony TV worked however, both set top boxes and PC software didn't like the idea of know about a previous event.
4. eitinject -f <filepath> plugin needs the full path.  I simulated on the command line with TSP -v to work out I have problems with paths.  

Happy to assist if anyone need clarity in EPG generation.

lelegard, 2023-10-27:
> eitinject -f plugin needs the full path. 

This is safer. The event input files are accessed as any other file, meaning that relative names are interpreted relatively to the current directory. If you put everything in a script, then the execution would depend on the invocation context of that script if you use relative names. So, it is safer to use full names indeed.

The other possibly that I usually favor is building full names from the relative location of the script, when that makes sense. Thus, if the complete file tree of an application is moved somewhere else, the script still works when input files are moved with the script.

Also do not forget to use quotes around wildcards to avoid early shell interpretation. Assuming you produce files named `events-NNN.xml`, with `NNN` being some number, assuming that an initial file `events-000.xml` already exists, then using `-P eitinject -f events-*.xml` is incorrect. The shell would expand the wildcard and the actual command becomes `-P eitinject -f events-000.xml`, using the first file only.

The right way is `-P eitinject -f 'events-*.xml'`. The quotes prevent the shell from expanding the wildcard and the plugin receives the string `events-*.xml`, matching any future `events-NNN.xml` file.

closed 2024-03-31 by github-actions[bot]