tsduck/tsduck #1311: [Q] Why stuffing packages are generated in scrambled TS.

sichunqin, 2023-10-08:
I use the following command to scramble one service (C8) of TS file. The scrambled TS file includes stuffing packets, but the original TS file does not. I wonder why those stuffing packets are generated. 

tsp -v -I file tnt-uhf25-506MHz-2019-01-22.ts  -P scrambler C8 --dvb-csa2 --cw-file cw-dvb-csa2.txt -O tnt-uhf25-506MHz-2019-01-22.ts.scr

tsanalyze tnt-uhf25-506MHz-2019-01-22.ts.scr
|  0x0014  TDT/TOT ...................................... C           74 b/s  |
|  0x0015  Network Synchronization ...................... C        2,727 b/s  |
|  0x1FFF  Stuffing ..................................... C      595,906 b/s  |
|=============================================================================|

tsanalyze tnt-uhf25-506MHz-2019-01-22.ts.scr
|  0x0014  TDT/TOT ...................................... C           74 b/s  |
|  0x0015  Network Synchronization ...................... C        2,727 b/s  |
|=============================================================================|

lelegard, 2023-10-08:
Hi @sichunqin

When a service must be scrambled, we must not let any part the service in the clear. Therefore, when the `scrambler` plugin starts, it nullifies all unidentified packets, until it gets a PMT for the service to scramble (also meaning getting the PAT first and then the SDT if the service is specified by name). Once the PMT is received, the PID's to scramble are identified and other PID's (typically from other services) are passed again.

The null packets you see are all grouped at the beginning of the output stream, and result from the above mentioned process.

This is quite easy to demonstrate:
~~~
$ alias tstest='tsp -I file tnt-uhf25-506MHz-2019-01-22.ts -P scrambler C8 --dvb-csa2 --cw 0123456789123456 -O drop'
$
$ tstest -P filter --pid 0x1FFF -P count
* count: PID 8191 (0x1FFF):     24,255 packets
$
~~~

We have 24255 null packets. Let's skip the first 24340 first output packets and see how many null packets are left: none.

~~~
$ tstest -P skip 24340 -P filter --pid 0x1FFF -P count
$
~~~

Note: The "frontier" at 24340 was determined after a couple of dichotomical tests.

And if you look at the first 24340 packets, they contain only the global signalization (PID < 0x20) and null packets resulting from the initial nullification of all packets which could, maybe, belong to the service to scramble.

~~~
$ tstest  -P until --packet 24340  -P count
* count: PID    0 (0x0000):         15 packets
* count: PID    1 (0x0001):          4 packets
* count: PID   16 (0x0010):          1 packets
* count: PID   17 (0x0011):          1 packets
* count: PID   18 (0x0012):         62 packets
* count: PID   21 (0x0015):          2 packets
* count: PID 8191 (0x1FFF):     24,255 packets
$
~~~

And BTW, the original DVB-T stream had stuffing, just like any broadcast stream, but the tuner I used to capture it silently drops input null packets. Although it makes sense if just want to watch TV, this is a very bad idea for TV engineers making tests...

sichunqin, 2023-10-09:
Thanks a lot for the reply！

lelegard, 2023-10-09:
> Instead of nullifies all unidentified packets, the plugin pmt removes those packages before get PMT?

Exactly, see [comment here](https://github.com/tsduck/tsduck/blob/3d5b13dcf7b6afd70f0b19ba7b8af1b8bcd3ef79/src/tsplugins/tsplugin_pmt.cpp#L766).

I must admit that this is not very consistent with `scrambler`, nullify vs. drop. This is the price to pay for a flexible framework where plugins were developed one by one, over years.

sichunqin, 2023-10-10:
Understand. Thanks much for the quick reply!

closed 2023-10-10 by sichunqin