tsduck/tsduck #1313: [Q] SDT, Station Logo's

SonnyWalkman, 2023-10-12:
Hello Again,
Does anyone know the table these new receivers are able to decode the Station Logo. Logo are displayed on the EPG list and channel info.  I thought it was searched from the internet however, I'm thinking it's a descriptor in either SDT or TSDT tables?  
I'm yet to sniff my local FTA TS data to seelout the logo data.  I'm sure somebody know how it done and where it lives?

[image]

SonnyWalkman, 2023-10-12:
While I'm at it. Can the Channel assignment LCN (LCT) be manipulated somehow? There doesn't seem to be any information on how to? Is it doable in TSduck?

lelegard, 2023-10-12:
> I'm yet to sniff my local FTA TS data to seek out the logo data.

Starting from the reality is always better than speculation, yes, this is the right way to go. This is the main reason why the `dvb` input plugin exists.

> Can the Channel assignment LCN (LCT) be manipulated somehow?

Logical Channel Numbers are not addressed by DVB. They are implemented using private descriptors, usually in the NIT. However, since DVB does not standardize them, each operator or country regulator defines its own flavor of private logical channel descriptor and rules of utilization. In TSDuck, the LCN descriptors are defined for EACEM/EICTA, DTG (UK), NorDig, Sky and Eutelsat. You have to check in which area you are. On European FTA networks, EACEM prevails, except at the "borders of the empire" (DTG, NorDig).

Again, look into the real transport streams.

AlfonZ42, 2023-10-12:
Quickly searching the spec, I think _Image Icon Descriptor_ might be intended for this station logo purpose.

SonnyWalkman, 2023-10-12:
Thanks again @lelegard,
I'll setup to syphon some of the stream and analyse and report back so others can enjoy the fruits.  Obviously, what I found defines what the Australian broadcasters have implemented. 

> Logical Channel Numbers are not addressed by DVB. They are implemented using private descriptors, usually in the NIT. However, since DVB does not standardize them.

I would have through broadcasters to follow the DVB specifications.  

> Quickly searching the spec, I think Image Icon Descriptor might be intended for this station logo purpose. Thank @AlfonZ42 
Again, another no standard DVB implementation however, receiver manufactures are in on the implementation to their products for it to work?

lelegard, 2023-10-13:
>  Question to you is does TSduck support the addition of a descriptor for an image?

Yes, it does. There are many ways to inject new tables or modify existing tables on the fly. XML files can be used to create or update (see "XML patch files") the signalization. The image icon descriptor is supported in TSDuck. See appendix D.8.50 in the user's guide. Most other MPEG, DVB, ISDB, ATSC or private descriptors are supported. If one is missing, it can be added if some proper specification is available.

> > Logical Channel Numbers are not addressed by DVB. They are implemented using private descriptors, usually in the NIT. However, since DVB does not standardize them.
>
> I would have thought broadcasters to follow the DVB specifications here? 

Precisely, DVB specifies nothing about logical channel numbers. The concept does not exist in DVB. It is entirely up to private descriptors.

In all countries, the public terrestrial network is usually controlled by a national regulator which publishes some specification or guidelines document on how to manage local or private aspects of the streams. It can be a huge spec (UK D-Book) or a just a few pages (France), but there is something. You should look for the Australian equivalent. Please post a link to it when you find it.

>  I guess I'll find out during sniffing the stream.

The [TSDuck stream repository](https://tsduck.io/streams/) proposes snapshots of real-life transport streams. They are useful to everyone to understand the TS structure of various countries and operators. There is a snapshot of all terrestrial streams for France, Italy, the Netherlands, UK, USA, Brazil, Russia, Japan. Except the French streams that I captured myself, all of them were provided by users from these countries. But we have nothing from Australia.

It would be great if you could capture a sample of each terrestrial stream in your area, let's say 100 to 200 MB for each, and post them on some public large file transfer site. I would then add them to the repository.

SonnyWalkman, 2023-10-13:
@lelegard,

> In all countries, the public terrestrial network is usually controlled by a national regulator which publishes some specification or guidelines document on how to manage local or private aspects of [...]

Here's the link to all you need to know about Australian FTA TV  - Specs on AIT, NIT and TDT and TOT 

https://www.freetv.com.au/resources/operational-practices/

I'll get some transport stream files.

lelegard, 2023-10-14:
From the link you provided, LCN are explained here:
https://www.freetv.com.au/wp-content/uploads/2019/08/OP-41-LCN-Descriptor-Issue-8-July-2016.pdf

The funny part is this: _"Within the Australian implementation of the DVB-T system, the Logical Channel Descriptor is treated as a public descriptor."_ I assume that this mean that the LCN descriptor (tag 0x83) is not treated as a private descriptor and is not preceded with a private data specifier descriptor. This is illegal in DVB terms. So, the _"Australian implementation of the DVB-T system"_ is not fully DVB-compliant. Anyway, I have seen worse... 🥲 

The description of the LCN descriptor is identical to the DTG version of it (UK), not the EACEM one (continental Europe). So, if you need to create your own LCN descriptors, use the `dtg_logical_channel _descriptor`, see the TSDuck user's guide, appendix D.9.3.

We will see from your streams if there is a private data specifier descriptor or not.

orryverducci, 2023-10-15:
I came across this and can shed some insight.

The specifications for the Australian DVB-T platform are based on the UK specifications (DTG D-Book). Therefore as discovered the private descriptors used (e.g. LCN) are the same as those used on the UK DTT platform.

As for the logos, unfortuantely like most brands Sony TV's retrieve these from the internet.

SonnyWalkman, 2023-10-16:
Hello @orryverducci  

> The specifications for the Australian DVB-T platform are based on the UK specifications (DTG D-Book). Therefore as discovered the private descriptors used (e.g. LCN) are the same as those used on the UK DTT platform.

Thank-you for confirming use of DTG D-Book descriptors.  @lelegard has confirmed such. I've read through most of the OP docs and are more familiar with specifications. 

As Thierry has indicated, the Australian Broadcaster have utilised _public descriptor_  for LCN which is not compliant with the DVB specifications?

I'm setting up my TBS-6922H card to record the TS stream to find what I can find. 
My EPG EIT seems to be inconsistent which I'm work through, 

Start of a new day I'm generating 24hours of events spaced 1 hour apart.
I'm injecting a `TDT `and `TOT `then `timeref` followed by `eitinject`? Not sure if the order is correct. I assume time is important for correct EPG? I'm not sure why it's not working as I would have expected. 
Happy for anyone to correct my wrong misunderstandings.

Attached is my EIT.xml 

> As for the logos, unfortunately like most brands Sony TV's retrieve these from the internet.

I'm assuming the URL is used over the embedded png/jpg thumbnail image.? Regardless, I could implement a link to the club's webpage to pull out an image to testing. I'm the web master therefor trivial to point a URL to a page containing the channel logo image.  Again, I'll have to sift through NIT or AIT to see how it's done.

[EIT.zip](https://github.com/tsduck/tsduck/files/12911423/EIT.zip)

lelegard, 2023-10-16:
> I'm starting each new day event_id="1" incrementing each event by one to end of day. Resetting with new date starting form 0x001 again

The event id uniquely identifies an event in a TS, independently of the day/time. Having two events in the same TS with the same id in the same EPG is illegal and may confuse receivers. Unless, of course, some local regulation specifies otherwise (just like private descriptors becoming public against all rules 😄).

Within each TS, the event generator may simply allocate event ids sequentially and wrap back to zero after 65535, assuming that there are no more than 65536 events at the same time in the same TS, within the EPG depth.

> version="6" , tried version="0"

If this is an XML file which is used by `eitinject`, this does not matter. These XML files are just "event databases". Event are extracted from their EIT envelope, which is then discarded. Proper EIT sections are rebuilt later by the plugin.

> I'm injecting a TDT and TOT then timeref followed by eitinject? Not sure if the order is correct. I assume time is important for correct EPG? 

Exactly. This is the right order.

lelegard, 2023-10-16:
> I think what Thierry was wondering, based on their "public" description, is if they don't bother sending the private data specifier 

Exactly !

Let's wait for the streams to look into them.

lelegard, 2023-10-16:
About LCN and private descriptors, see an example of the DTG LCN descriptor on the UK Freeview network: https://tsduck.io/streams/uk-freeview/482000000.psi.txt

Search for the NIT:
~~~
* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 3, sections: 2, total size: 1683 bytes
  - Section 0:
    Network Id: 12339 (0x3033)
    Network information:
    - Descriptor 0: Network Name (0x40, 64), 6 bytes
      Name: "London"
    [....]
    Transport Stream Id: 4164 (0x1044), Original Network Id: 9018 (0x233A)
    [....]
    - Descriptor 2: Private Data Specifier (0x5F, 95), 4 bytes
      Specifier: 0x0000233A (DTT UK (OFCOM, formerly ITC))
    - Descriptor 3: UK DTT Logical Channel (0x83, 131), 100 bytes
      Service Id:  4164 (0x1044), Visible: 1, Channel number:   1
      Service Id:  4352 (0x1100), Visible: 1, Channel number: 231
      Service Id:  5760 (0x1680), Visible: 1, Channel number: 707
      Service Id:  6275 (0x1883), Visible: 1, Channel number: 719
      Service Id:  6848 (0x1AC0), Visible: 1, Channel number: 703
      Service Id:  4544 (0x11C0), Visible: 1, Channel number:   9
    [....]
~~~

We see a private descriptor (tag 0x83) which is correctly preceded with a private data specifier descriptor identifying the private organization which defines the subsequent private descriptor, here DTT UK / OFCOM / ITC.

SonnyWalkman, 2023-10-21:
> Exactly !

> Let's wait for the streams to look into them.

Hello there Thierry @lelegard,

I have good news, 
I have written a bash script and have successfully pulled analysis and MPTS for each of the Melbourne channels here in Victoria Australia using my TBS-6522H Mult-standed dvb card with TSduck.  https://www.tbsdtv.com/products/tbs6522h-multi-standard-quad-tuner-pci-e-card.html
I receive each via a UHF translator which as it says, translates from VHF to UHF to service Highton area which is a gully and services here in Leopold. 

The files are for the Seven, Nine, Ten, ABC and SBS Networks with 60 seconds of MPTS for each.   

Do you have direct link to share with me so I can upload the MPTS files. Each network is approx 180M of TS file!

My email is vk3eht(at)gmail(dot)com

Attached here are to analysis files.
[MelbourneAustralia.zip](https://github.com/tsduck/tsduck/files/13060631/MelbourneAustralia.zip)

lelegard, 2023-10-21:
Hi @SonnyWalkman 

Great to have Australian streams. I don't provide public access to upload. In similar cases, people use public transfer sites such as wetransfer.com or similar.

SonnyWalkman, 2023-10-21:
Send me your email address and I'll send you a link to my Dropbox to upload from.

orryverducci, 2023-10-21:
I was interested so just had a look at those PSI analysis files.

Looks like they do use a private data specifier, but not the UK DTG one. They're using the specifiers allocated to the Australian Terrestrial Television network. So at the moment TSDuck is not picking up the LCN descriptor as being the DTG one.

Looking at their [DVB SI Codes document (OP40)](https://www.freetv.com.au/wp-content/uploads/2019/08/OP-40-Allocation-of-DVB-Service-Information-Codes-Issue-4-Sept-2017-v2..pdf) it seems they use different private data specifiers for different TV networks instead of a single one like most places, so any value between 0x3200 and 0x320F should be treated as Australian DTT. In practice it seems only the ABC are using the one assigned to them, and the other networks are using the generic 0x3200 specifier.

As for logos, as I suspected I don't see anything in there that would point to logos, so unfortunately I think that confirms the TV is finding them online as it does here in the UK.

SonnyWalkman, 2023-10-21:
Hello @orryverducci,
Yes agree.. No png/jpg data to be seen. It is interesting to navigate to the CRID data in the SDT.  example below of [CRID://melbourne22.abc.net.au](url)
I'm assuming these CRID's are the link for the TV to pull the logo. It be interesting how its done to play with it. Easy enough to set up similar on the Clubs website.

```
 Service Id: 562 (0x0232), EITs: yes, EITp/f: yes, CA mode: free
    Running status: running
    - Descriptor 0: Service (0x48, 72), 20 bytes
      Service type: 0x01 (Digital television service)
      Service: "ABCKids/TVPlus", Provider: "ABC"
    - Descriptor 1: Default Authority (0x73, 115), 29 bytes
      0000:  43 52 49 44 3A 2F 2F 6D 65 6C 62 6F 75 72  CRID://melbour
      000E:  6E 65 32 32 2E 61 62 63 2E 6E 65 74 2E 61  ne22.abc.net.a
      001C:  75                                         u
```

orryverducci, 2023-10-21:
The CRID's are nothing to do with logos. They're the TV Anywhere Content Reference ID's. They're used to link programmes on schedules to other things. For example in the UK we use to enable recording series of programmes (series link). Sometime they're real URL's, sometimes they're not.

As I said before there is no standardised way to point to a channel logo via a DVB multiplex, and your samples confirm that there's nothing like that on the Australian terrestrial platform. Sony TV's get their logo by looking at the service name (e.g. ABC TV) and then asking Gracenote's online service if they have a logo for a channel with that name. Samsung and LG TV's have similar mechanisms.

That unfortuantely means you won't be able to achieve what you're trying to achieve.

SonnyWalkman, 2023-10-22:
Hello sorry,

Yes I understand now..
However there is one question.
If smart TVs know to lookup service ID and look at Gracenote there must be
a mechanising for such?

There obviously programmed into the TV firmware to do this?

Sonny

orryverducci, 2023-10-22:
Yes it's in the TV firmware.

Different manufacturers do it in different ways, so I couldn't say exactly how it works other than it uses online services and appears to use the channel name and the country to the TV is located in to look up the logos.

I know Sony use Gracenote as their logo provider. Other manufacturers may use different services (I suspect Samsung have their own).

lelegard, 2023-10-22:
To check how your TV set does this, I would start with Wireshark. The easiest setup is a small Ethernet switch with a monitoring port (usually one of the ports which is configured as such). Assuming that the TV set uses Ethernet and is plugged on that switch, then a PC on the monitoring port can sniff all the IP traffic to and from the TV set.

lelegard, 2023-10-22:
@SonnyWalkman, thanks a lot of the streams. I am going to publish them on the TSDuck streams repository.

Which area were they captured from? More precisely, which part of the Australian Free TV network is this? From the screenshot in your original post, I would say Melbourne. Could you confirm?

lelegard, 2023-10-22:
The streams were posted there: https://tsduck.io/streams/?name=australia-dttv

@SonnyWalkman, let us know here if you'd like to update the description.

lelegard, 2023-10-22:
@orryverducci,

> Looking at their [DVB SI Codes document (OP40)](https://www.freetv.com.au/wp-content/uploads/2019/08/OP-40-Allocation-of-DVB-Service-Information-Codes-Issue-4-Sept-2017-v2..pdf) it seems they use [...]

Yes you are right, this is a strange setup. I suspect that there were some confusion in the mind of the guys who defined this.

A PDS (private data specifier) is allocated to a technical provider which defines its own private signalisation syntax. This is not for a network. Network ids were defined for that.

In the [DVB-SI allocation document](https://www.dvbservices.com/identifiers/private_data_spec_id), the range 0x3200-0x320F is allocated to "Australian Terrestrial Television Networks". So, the Australian authorities have the right to distribute these 16 values to local technical providers.

Distributing them to networks is strange. Probably an interpretation error in the mind of the original Australian regulators. Anyway, this is still legal.

Where things become weird is the usage of the PDS descriptor. In principle, a PDS is the root of several private descriptors (tags in the range 0x80-0xFE). Looking the other way round, one private descriptor is defined in the context of one single PDS (descriptor tag 0x83 is not the same when the PDS is from EACEM, UK DTG, Australian DTTB, etc).

However, in Australia, descriptor tag 0x83 is equally used after PDS 0x3201 (ABC) and 0x3200 (all others). In theory, this means that the two descriptors are different. While they seem to be the same.

In DVB theory, the situation should be:

1. Before the LCN descriptor, the PDS descriptor should specify the PDS for UK DTG since this is the organization which defined its structure.
2. If some national Australian private descriptors are defined at national level by the Australian authorities, these private descriptors should be preceded by PDS 0x3200.
3. If the operator ABC defines its own private descriptors for its own usage, then they should be preceded by PDS 0x3201. Same principle for other operators.

Let's assume that some national pride prevents the usage of a UK PDS. It could be legitimate to define that the LCN descriptor becomes an Australian one (with the same structure as the UK one) and it is preceded by PDS 0x3200. So, let's drop case 1.

Nonetheless, case 3 remains. The usage of PDS 0x3201 before the LCN descriptor seems a mistake from ABC. All others are right.

For the defense of ABC, the national specification is unclear, imprecise, and confusing, since the OP-41 document says that _"Within the Australian implementation of the DVB-T system, the Logical Channel Descriptor is treated as a public descriptor."_ Not only, this is illegal, but operators apparently use a PDS descriptor anyway, in a nice effort to be more standard. However, on top of that, they don't use the PDS descriptor in a consistent way...

Anyway, seen from Europe, where DVB originates from, you guys are walking upside down 😆 🤣  (just kidding, no offense).

I won't be surprised if most TV sets on the Australian market simply ignore the PDS descriptor and blindly consider descriptor 0x83 as the Australian LCN descriptor.

I have seen that on European TV sets in the French market. Here, we use the EACEM version of the LCN descriptor, which is consequently valid after an EACEM PDS only. Then, some 13 years ago, some guy (me...) defined a private signalisation for a specific Push-VoD service. All these private descriptors where sequentially allocated starting from 0x80, after a _properly registered PDS value_ which was allocated to my company. Guess what? On day 1, all Sony TV sets immediately screwed up all over the test region... They ignored the PDS descriptors and interpreted all descriptors 0x83 as EACEM-defined LCN descriptors. Since the national regulator estimated it was inconvenient to ask Sony to fix their bugs, they asked us to reallocate our private descriptor tags in a less crowed range. Which I did...

Now, to implement that in TSDuck, we have a problem. Each descriptor class and XML representation is attached to _one single PDS_ (zero for public descriptors). There is no way to attach a descriptor class to several PDS. I will think about it.

lelegard, 2023-10-22:
OK, I realised that the same issue of a misused PDS already occurred a long time ago in France when TPS, a now defunct satellite operator, used its own PDS instead of the EACEM PDS for logical channel descriptor.

A trick was found at the time to support this in TSDuck. I applied the very same trick for the Australian LCN descriptor.

I implemented what seems to me the rational way of working of PDS:

First, since Free TV Australia OP-41 documents the structure of the LCN descriptor, this defines a private descriptor for the Free TV Australia PDS (0x3200). This new private descriptor is named `<australia_logical_channel_descriptor>` in XML files. Its official PDS is 0x3200. Its structure is the same as its DTG counterpart.

Second, as a trick, all PDS in ranges 0x3201-0x3205 and 0x320A-320F are considered as alternate (although invalid) PDS for the same descriptor.

Thus, the signalisation of the Australian streams you provided is correctly interpreted.

SonnyWalkman, 2023-10-23:
Hello.

> @SonnyWalkman, let us know here if you'd like to update the description.

The stations are transmitted for Mt Dandenong outer hills of Melbourne. I'm located in Geelong which is 75Kms Northwest of Melbourne and receive these signals through  the Montpellier Service Basin Highton translator. Melbourne metro is VHF for ABC, Seven, Nine and Ten and UHF Channel 29 for SBS to UHF

https://ozdigitaltv.com/transmitters/VIC/588-Highton

Seven   HSV46   655.500 Mhz 
Nine     GTV47   662.500 Mhz 
Ten       ATV48   669.500 Mhz 
ABC     ABC49    676.500 Mhz 
SBS      SBS50    683.500 Mhz

[image]

closed 2023-10-23 by SonnyWalkman