tsduck/tsduck #1316: [BUG] Build error referring to deleted move-constructor

MorelloCherry, 2023-10-13:
#### Bug description:
Compile fails using gcc (Debian 10.2.1-6) 10.2.1 20210110 with the error (Errors and logs)

#### How to reproduce:
1. Clean checkout from GIT
2. Run `make NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOCURL=1 NOPCSC=1 NOSRT=1 NORIST=1 NOEDITLINE=1 NOGITHUB=1 BINDIR=$(pwd -P)/bin/native -j 8`

#### Expected behavior:
The build successfully completes.

#### Errors and logs:
~~~
  [CXX] tsplugin_scrambler.cpp
tsplugin_scrambler.cpp: In constructor ‘ts::ScramblerPlugin::ScramblerPlugin(ts::TSP*)’:
tsplugin_scrambler.cpp:220:19: error: use of deleted function ‘ts::ScramblerPlugin::CryptoPeriod::CryptoPeriod(ts::ScramblerPlugin::CryptoPeriod&&)’
  220 |     _pzer_pmt(duck)
      |                   ^
In file included from /home/user/tsduck/src/libtsduck/base/types/tsUChar.h:16,
                 from /home/user/tsduck/src/libtsduck/base/types/tsUString.h:15,
                 from /home/user/tsduck/src/libtsduck/base/report/tsReport.h:15,
                 from /home/user/tsduck/src/libtsduck/base/app/tsArgs.h:15,
                 from /home/user/tsduck/src/libtsduck/plugins/plugins/tsPlugin.h:15,
                 from /home/user/tsduck/src/libtsduck/plugins/plugins/tsInputPlugin.h:15,
                 from /home/user/tsduck/src/libtsduck/plugins/infra/tsPluginRepository.h:15,
                 from tsplugin_scrambler.cpp:14:
tsplugin_scrambler.cpp:94:23: note: declared here
   94 |             TS_NOCOPY(CryptoPeriod);
      |                       ^~~~~~~~~~~~
/home/user/tsduck/src/libtsduck/base/cpp/tsPlatform.h:1004:9: note: in definition of macro ‘TS_NOCOPY’
 1004 |         classname(classname&&) = delete;            \
      |         ^~~~~~~~~
make[2]: *** [../../Makefile.inc:576: /home/user/tsduck/bin/native/objs-tsplugins/tsplugin_scrambler.o] Error 1
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:39: default] Error 2
~~~

#### Environment:
 - OS: Linux
 - OS version: Debian GNU/Linux 11 (bullseye)
 - TSDuck full version: Git commit hash (a7de2278d)
 - Installation type: None, trying to compile struck from source

#### Additional information:
N/A

lelegard, 2023-10-14:
Hi @MorelloCherry 

This is not a bug in TSDuck, this is a bug in gcc 10 : the default constructor of a class shall not use the move constructor of one of its fields when initialised by default.

It works with gcc 11, 12, 13 (bug fixed then) as well as clang and msvc, all versions.

I do not have any gcc 10. However, I have an old RedHat 8 with gcc 8.5 where I reproduce the same gcc bug. I have added a simple workaround in the code so that gcc no longer attempt to use move constructor of its field. So, you may pull the repo and recompile.

As a general rule, when a software such as TSDuck evolves, we need to maintain the code simple and clean and ready for future use and future compilers (next revisions of the language). Therefore, I regularly refactor the code to more modern forms.

At some point, old operating systems and old compilers may no longer be supported. For instance, now, TSDuck requires C++11. Old compilers not supporting it can no longer be used. This may also trigger bugs in old versions of a compiler, as you just got one. If I can implement a simple workaround which does not clutter the code, I do it, as I just did. Otherwise, I do not commit to support old systems and compilers with new versions of TSDuck at all cost.

MorelloCherry, 2023-10-16:
Thank you for confirming that and making GCC 10 work for me.

closed 2023-10-16 by lelegard