tsduck/tsduck #1317: Is TSDuck still compartible with Windows 7?

sergeyshulga, 2023-10-14:
Hello. Is TSDuck still compartible with Windows 7?
On last version (Win64-3.35-3258) i got error. "The procedure entry point SetThreadDescription could not be located in the dynamic link library KERNEL32.dll"
Last version  working OK - Win64-3.30-2710

sergeyshulga, 2023-10-14:
Sorry. My fault. After installing latest vc_redist.x64.exe problem is gone,

lelegard, 2023-10-14:
Hi @sergeyshulga 

There is no guarantee that the latest version of TSDuck may run on Windows 7 (and, from your experience, it doesn't).

New versions are built on recent operating systems, with recent compilers. They use features of these recent systems and compilers. I do not have the resources to maintain compatibility with zillions of obsolete systems and compilers.

If version 3.30 is latest one which works on Win7, I am afraid you will have to stay at his level until you upgrade Windows. 

BTW, you must have a very very strong reason to stay on Windows 7. This is an obsolete, unsupported and security-vulnerable version of Windows.

sergeyshulga, 2023-10-14:
Thanks for answer. I stay with Win7 on some isolated from Internet machines, and dont fear about any vulnerabilities. ))
But probably TSDuck not also works on Windows Server 2016, Windows 10 LTSB 2016 and Windows 10 version 1607 as described in https://learn.microsoft.com/en-us/windows/win32/api/processthreadsapi/nf-processthreadsapi-setthreaddescription
Its so recent operating systems, but also probably not working... Maybe need to check compiler settings?

lelegard, 2023-10-14:
Yes, exactly. According to the link you found, we need at least Windows 10, version 1607 or  Windows Server 2016. We are at end of 2023 and requiring an OS not older than 7 years behind does not seem too demanding.

In all cases, the rule is : all versions of operating systems work only with old versions of applications. New versions of application move ahead, toward the future, not backward to the past.

closed 2024-03-17 by github-actions[bot]