tsduck/tsduck #1320: How to change the audio and video pid value?[Q]

udhayNuk, 2023-10-16:
@lelegard 
tsp -v -m --synchronous-log -I ip 236.1.3.85:1026 \
 -P pmt -m 0x02C2/0x200 \
 -P analyze -i 30 -o report.txt \
 -O hls --duration 3 --live 15 -p index.m3u8 index.ts
 
 i have changed the audio pid using pmt, but audio is not playing in hls.
 please solve my question.
 and is there any other method to change the audio pid value and video pid value.
 this is my input analysed
 Input #0, mpegts, from 'udp://236.1.3.85:1026':
  Duration: N/A, start: 3746.559744, bitrate: N/A
  Program 6
    Metadata:
      service_name    : service
      service_provider: service tv
    Stream #0:0[0x25e]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuv420p(tv, bt470bg, top first), 720x576 [SAR 12:11 DAR 15:11], 25 fps, 50 tbr, 90k tbn, 50 tbc
    Stream #0:1[0x2c2](eng): Audio: mp2 ([4][0][0][0] / 0x0004), 48000 Hz, stereo, s16p, 96 kb/s

lelegard, 2023-10-16:
> `-P pmt -m 0x02C2/0x200`

You have changed the reference in the PMT but you did not change the PID itself. So, the PMT now points to a non-existent PID 0x0200. The PID 0x02C2 still exists, with its audio content, but it is no longer referenced anywhere.

The plugin `pmt`  performs local modification inside the PMT only.

See the plugin `remap` for a fully consistent PID remapping, in the PID and in all known signalization tables which reference that PID.

udhayNuk, 2023-10-16:
@lelegard 
thanking you for replying.
how to change all the pid values like SDT/BAT, PMT, PAT.
i want to change all pid values, suggest methods to change pids value with taking the default pid values and modify the pid.
please reply me.

udhayNuk, 2023-10-17:
@lelegard 
thanking you for replying.
i have used this script to change the pids pmt, video and audio pid value.
   -P remap 0x2c2=0x1222 0x25e=0x1223 -P regulate \
how to take the default input pids value (PMT, AUDIO and VIDEO Pids value) from input and change the pids value.
is there any way.
please reply me.

lelegard, 2023-10-17:
> how to take the default input pids value (PMT, AUDIO and VIDEO Pids value) from input and change the pids value.

I don't understand the question.

Do you mean that
- you don't initially know the PID for PMT, audio and video
- you want an automatic way of finding these PID values
- and then apply these input values in the `remap` plugin.

If this is what you want, this is a scripting issue (bash or Python for instance). You can't do that in one command. TSDuck is script-friendly: it provides the tools to extract information and automate processing in scripts that _you have to write_.

In that case, I suggest a two phases approach. In the first phase, use the plugin `-P tables --tid 2 --max-tables 1` (and output `-O drop`) to collect the first PMT, and the command will stop. You can get the PMT in textual form, XML or JSON. Choose the format you understand the best and know how to analyse. You then find the PMT PID and all its components. Be aware that, in the general case, there can be several audio PID and, why not, no video PID. Once you have found the PID values, you can generate the final `remap` command.

udhayNuk, 2023-10-18:
@lelegard 
thank you

closed 2023-10-18 by udhayNuk