tsduck/tsduck #1321: [BUG] DTA-2162 as input (TSoIP) does not work

sstaty, 2023-10-16:
#### Bug description:

TSDuck TSP process does not work well with DTA-2162. This is card that does TSoIP (i.e. Transport Stream Over IP). TSP with input from DTA-2162 does not work at all, i.e., no data is received. On the other hand, TSP with output from DTA-2162 does work.

#### How to reproduce:

Send TSoIP stream from TSoIP generator, for example with tsp -ip plugin:
```
tsp I file mux1rai.ts \
    -P regulate \
    -O ip --rtp 192.168.120.201:5004
```

Listen to the TSoIP traffic on receiver side:
```
tsp -I dektec --device 0 --channel 0 --io-standard IP \
 --ip4 192.168.120.201:5004 \
 --smpte-2022-fec \
-O file out.ts 
```

#### Expected behavior:

The file out.ts should be getting TS packets, but instead it is 0 bytes.
If, instead, one uses dektec software DtRecord on the receiver side, instead of tsp, it will correctly get the ts traffic (i.e. out.ts is not empty)

`./DtRecord out.ts -t 2162 -ipa 192.168.120.201:5004`

#### Errors and logs:

There is no error, but debug from the tsp command is attached below (no issues detected)
~~~
dev@DSA-2:~/pikalovd/ts-over-ip$ tsp -d -I dektec --device 0 --channel 0 --io-standard IP  --ip4 192.168.120.201:5004 -O fi
le out.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec --device 0 --channel 0 --io-standard IP --ip4 192.168.120.201:5004
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file out.ts
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 0, input channel 0 (DTA-2162 port 1)
* Debug: dektec: attaching to device DTA-2162 serial 0x0000000080DD89AC
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting RxControl, mode: 0
* Debug: dektec: clearing FIFO and flags
* Debug: dektec: getting FIFO max size
* Debug: dektec: max FIFO size: 134,217,728 bytes
* Debug: dektec: using FIFO size: 134,217,728 bytes
* Debug: dektec: setting IO config of port 1, group: 1, value: 42, subvalue: -1
* Debug: dektec: setting IP parameters: DtIpPars2 = {
* Debug: dektec:   m_Ip = C0 A8 78 C9 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec:   m_Port = 5004
* Debug: dektec:   m_Gateway = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec:   m_VlanId = 0
* Debug: dektec:   m_VlanPriority = 0
* Debug: dektec:   m_Ip2 = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec:   m_Port2 = 0
* Debug: dektec:   m_Gateway2 = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec:   m_VlanId2 = 0
* Debug: dektec:   m_VlanPriority2 = 0
* Debug: dektec:   m_TimeToLive = 0
* Debug: dektec:   m_NumTpPerIp = 7
* Debug: dektec:   m_Protocol = 2
* Debug: dektec:   m_DiffServ = 0
* Debug: dektec:   m_FecMode = 0
* Debug: dektec:   m_FecNumRows = 0
* Debug: dektec:   m_FecNumCols = 0
* Debug: dektec:   m_Flags = 0x00000000
* Debug: dektec:   m_Mode = 0
* Debug: dektec:   m_IpProfile.m_Profile = 0
* Debug: dektec:   m_IpProfile.m_MaxBitrate = 0
* Debug: dektec:   m_IpProfile.m_MaxSkew = 0
* Debug: dektec:   m_IpProfile.m_VideoStandard = 0
* Debug: dektec: }
* Debug: dektec: setting RxMode, mode: 17
* Debug: dektec: setting RxControl, mode: 1
~~~

#### Environment:
 - OS: Debian 11.7
 - OS version: Debian 11.7
 - TSDuck full version: 3.35-3258
 - Installation type: 

#### Additional information:

It is interesting that sending traffic via the same card via TSP works. E.g. this command correctly sends traffic out:
```

tsp -I file mux1rai.ts | tsp -O dektec \
  --device 0 --channel 0 \
  --ip4 192.168.120.201:5004 \
  --rtp \
  --smpte-2022-fec 2d-m2 \
  --smpte-2022-l 5 \
  --smpte-2022-d 5

```

lelegard, 2023-10-16:
@EricDeJong, could you have a look at this one?
I have no such card to test.

sstaty, 2023-10-17:
@lelegard If there is any test / debug log I can provide that could help, please let me know.
I also pinged DekTec support, I am not sure if they are active member here, but in case they are, they might have a look at it.

EricDeJong, 2023-10-24:
> I noticed that for m_IpProfile (the log in the post), there is

I found the problem, the tsDektecInputPlugin tries to read data to fill the complete buffer of 44620 packets, instead of the available data in the dektec driver.
The plugin should check the available bytes with the GetFifoLoad function and then read that number of bytes or when no bytes are available read a single ts packet with a given timeout.
I sent you a modified tsDektecInputPlugin.cpp to try out / evaluation.

Note, for unicast addresses, the Dektec API expects an IP address of 0.0.0.0 or the board own IP address.

Also checked TSDuck source code for this, there is:

    dtpars.m_IpProfile.m_Profile = DTAPI_IP_PROF_NOT_DEFINED;

    dtpars.m_IpProfile.m_MaxBitrate = 0;

    dtpars.m_IpProfile.m_MaxSkew = 0;

    dtpars.m_IpProfile.m_VideoStandard = DTAPI_VIDSTD_TS;

Theoretically MaxBitrate = 0 could throttle all traffic, but not sure how it influences the card;

-

lelegard, 2023-10-24:
Thanks @EricDeJong for the fix.

@sstaty, could you please confirm that the fix solves your problem? So that Eric can submit the PR for the fix.

sstaty, 2023-10-24:
Hi Eric, thanks a lot for the fix!
I just tested with the tsDektecInputPlugin.cpp file from Eric and can confirm it works now. I can receive traffic with DTA-2162. Problem solved :)

lelegard, 2023-10-24:
@sstaty, thanks for the confirmation.

@EricDeJong, do you submit a PR or shall I replace the code you sent us?

lelegard, 2023-10-25:
@EricDeJong, ok, your fix has been pushed.

@sstaty, could you please pull the repo, rebuild, test and confirm that the problem is fixed.

Since you are on Debian, doing `make -j10 installer` builds the binaries and a `.deb` package. The `-j10` option (or any other value depending on your machine) speeds up the build, using 10 parallel compilations.

sstaty, 2023-10-31:
Hi @lelegard and @EricDeJong 
I run into an issue with reading dektec input with tsp from ASI, and I am not sure if this fix could have caused it, so I just wanted to let you know.
Even though I checked that I can receive TSoIP traffic with DTA-2162, I haven't checked if I can receive ASI streams with dektec ASI cards. Now I got the to point where I am using ASI input and I get the following error: 
[image]
This is machine where everything was working with dektec ASI card, now it doesn't (I think the only change that was there was installing the new TSDuck with this fix). Also, I can receive traffic with DtRecord from Dektec, so the card is functional.

sstaty, 2023-10-31:
I can confirm that this issue is caused by the changes.
I reinstalled TSDuck to 3.35 and this issue is not there.

lelegard, 2023-11-01:
@EricDeJong, can you confirm that reading from a Dektec device shall be done differently depending on the class of device? If the fix breaks the other usages of the plugin, I will have to revert to the previous code.

For the anecdote, my initial motivation, back in 2005, to develop what would become TSDuck was the processing of TS coming from a Dektec DTA-100 ASI card. So, I hate the idea of breaking the Dektec ASI input 😉

EricDeJong, 2023-11-02:
The ASI input does not work because the DekTec library expects that a multiple of 4 bytes is read, generating an DTAP_E_INVALID_SIZE error if the read size is not a multiple of 4 bytes. For ASI over IP input, there is no such limitation.
Another problem is that TSDuck expects whole 188-bytes TS packets, giving a sync error when a half packet was already fed to the input.

I fixed the code by reading only whole 188 bytes chunks of data.
As the DekTec input is configured with mode DTAPI_RXMODE_ST188, this will also work for 204-byte TS packets, as the DekTec SDK will sync to and convert this input to 188-byte packets.

I sent a new version of tsDektecInputPlugin.cpp to Martin and Thierry, please try this out.

lelegard, 2023-11-02:
> I sent a new version of tsDektecInputPlugin.cpp to Martin and Thierry, please try this out.

I have integrated Eric's fix into the repo. @sstaty, could you please try it on all input Dektec devices you have and let us know if it works fine everywhere.

@EricDeJong, please resync with the repo. There is a refactoring & code cleanup process in progress and almost source files were slightly impacted.

sstaty, 2023-11-06:
Hi @lelegard,
Sorry for delay. I wanted to test this now. I cloned fresh repo and build (make -j10), however I am getting a compilation error (I am on Debian 11.6).

```
  [CXX] plugins/private/tsRISTPluginData.cpp
  [CXX] dtv/signalization/tsSectionFile.cpp
  [CXX] dtv/demux/tsSectionHandlerInterface.cpp
  [CXX] dtv/demux/tsSectionProviderInterface.cpp
  [CXX] dtv/signalization/tsService.cpp
  [CXX] dtv/signalization/tsServiceDiscovery.cpp
  [CXX] dtv/signalization/tsServiceIdTriplet.cpp
  [CXX] dtv/signalization/tsSectionFileArgs.cpp
  [CXX] base/system/tsSharedLibrary.cpp
  [CXX] base/linux/tsSignalAllocator.cpp
  [CXX] dtv/broadcast/tsSignalState.cpp
  [CXX] dtv/demux/tsSignalizationDemux.cpp
  [CXX] dtv/demux/tsSignalizationHandlerInterface.cpp
  [CXX] dtv/dvbsimulcrypt/tsSimulCryptDate.cpp
plugins/private/tsRISTPluginData.cpp: In member function ‘bool ts::RISTPluginData::addPeers()’:
plugins/private/tsRISTPluginData.cpp:234:50: error: ‘int rist_enable_eap_srp(rist_peer*, const char*, const char*, user_verifier_lookup_t, void*)’ is deprecated [-Werror=deprecated-declarations]
  234 |             const int err = ::rist_enable_eap_srp(peer, config->srp_username, config->srp_password, nullptr, nullptr);
      |                             ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/librist/librist.h:18,
                 from /home/dev/martin/tsduck/src/libtsduck/base/private/tsLibRIST.h:20,
                 from plugins/private/tsRISTPluginData.h:16,
                 from plugins/private/tsRISTPluginData.cpp:9:
/usr/include/librist/librist_srp.h:114:30: note: declared here
  114 | RIST_API RIST_DEPRECATED int rist_enable_eap_srp(struct rist_peer *peer, const char *username, const char *password, user_verifier_lookup_t lookup_func, void *userdata);
      |                              ^~~~~~~~~~~~~~~~~~~
  [CXX] base/network/tsSocket.cpp
  [CXX] dtv/signalization/tsSpliceInsert.cpp
  [CXX] dtv/signalization/tsSpliceSchedule.cpp
  [CXX] dtv/demux/tsStandaloneTableDemux.cpp
cc1plus: all warnings being treated as errors
make[2]: *** [../../Makefile.inc:582: /home/dev/martin/tsduck/bin/release-x86_64-csg/objs-libtsduck/tsRISTPluginData.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:39: default] Error 2
```

lelegard, 2023-11-06:
@sstaty, when you tested the previous fix from @EricDeJong, did you rebuild on this machine?

I cannot reproduce this on a Debian 12 (I have no Debian 11). On my up-to-date Debian 12, the version of `librist-dev` is 0.2.7. In the librist header file, the function is declared as `RIST_API int rist_enable_eap_srp` while yours has an additional `RIST_DEPRECATED` (from the error message).

What is your version of `librist-dev` ? Since your Debian 11 is older than my Debian 12, I assume that your `librist-dev` can only be older than 0.2.7, equal at best. Something valid in current version may be marked as deprecated in _future_ versions, not in an older one.

The calling line in TSDuck has not changed since Oct 2021. So, no recent change in TSDuck may explain this.

To stay focus on the Dektec issue, please try on the same machine where you already rebuilt the previous fix.

lelegard, 2023-11-06:
> Both cards now work properly with tsp -I dektec (recorded .ts and played out in StreamXpert, everything looked alright). All seems good now :)

Thanks for the confirmation. I close this one.

closed 2023-11-06 by lelegard

lelegard, 2023-11-07:
@sstaty, one last thing about the Debian 11.6 system where the compilation failed: please check the installed version of package `librist-dev`. The "standard" version should be 0.2.7 on this system. However, I guess that you have installed a Debian Testing (SID) version, probably 0.2.10.

In version 0.2.8 of librist (unreleased on most Linux distros), there was an API change which causes the error you see. I fixed it using some #ifdef on the librist version. You may try to compile the latest state of the TSDuck repo.