tsduck/tsduck #1322: [Q]  TS-over-IP: options for --smpte-2022-fec, what do they mean?

sstaty, 2023-10-17:
On page 218 of TSDuck manual, for TS-over-IP for dektec devices, there is --smpte-2022-fec options (image attached). From the name of the options, it is not clear what the differences are. From smpte-2022 standard, there should be in options: no FEC, 1D FEC, 2D FEC. 

[image]

lelegard, 2023-10-17:
Hi @sstaty 

These command line options are directly passed to the Dektec DTAPI. See the Dektec documentation "DTAPI Reference - Core Classes.pdf", page 118:

[image]

Generally speaking, the `dektec` plugins are simple (?) command line wrappers on top of DTAPI to read or write a transport stream. Any information on the interpretation of such or such option shall be retrieved from the Dektec documentation. This is a Dektec question, not a TSDuck question.

sstaty, 2023-10-19:
Yes, you are absolutely right - sorry I did not realize this is directly from DTAPI. Thank you for clarification.

closed 2023-10-19 by sstaty