tsduck/tsduck #1323: [FEAT] Add flush method to PESDemux

fennewald, 2023-10-17:
#### Is your feature request related to a problem?

Currently, when demuxing a PES stream, PES Packets at the end of the stream with a length of 0 will not be processed. This is because PES processing is only done when either the expected length is reached, or (in the case of length 0), a start code of a new PES packet is detected. For the final PES packet in a truncated stream, the next start code never occurs.

#### Proposed solution:
Add a method `flush` to `ts::PESDemux` that will trigger PES processing for any pending length-0 PES packets.

For example:
```cpp
void ts::PESDemux::flush(void) {
    for (const auto & it : _pids) {
        auto pc = it.second;
        if (pc.ts->size() <= 6) continue;
        size_t len = GetUInt16(pc.ts->data() + 4);
        if (len == 0) {
            processPESPacket(it.first, pc);
            pc.syncLost();
        }
    }
}

```

#### Alternatives:
Users can also feed a single dummy packet that contains a PES Header to trigger processing of the PES data. This needs to be done in turn for every PID.

Another option to avoid a needlessly expensive iteration over the entire context map would be to introduce a `flushPID` method, which flushes a given PID stream.

#### Additional information:

To generate a file that exhibits the behavior use the following ffmpeg command:
```shell
width=1920
height=1080

ffmpeg \
  -f lavfi \
  -i nullsrc=s=${width}x${height} \
  -filter_complex 'geq=random(1)*255:128:128' \
  -t 0.1 \
  -c:v libx265 \
  -bf 0 \
  -crf 10 \
  test0.ts
```
[example.ts.zip](https://github.com/tsduck/tsduck/files/12961789/example.ts.zip)

The resulting file will have 3 frames (1/10th of a second at 30fps), and each frame will be substantially larger than 2<sup>16</sup> bytes. This means the `pes_packet_length` field for these frames will be set to 0, and tsduck will only process PESPackets with length 0 if it sees another start code. Since there is no start code after the last packet in the file, tsduck will only decode 2 PES packets from this stream.

To reproduce this, the following snippet logs a message for each PES packet it encounters:

```cpp
#include <cstddef>
#include <cstdio>

#include <tsDuckContext.h>
#include <tsPESDemux.h>
#include <tsPESProviderInterface.h>
#include <tsTSPacket.h>

static constexpr size_t k_buff_len = 1000;

class DemuxHandler : public ts::PESHandlerInterface {
	virtual void handlePESPacket(ts::PESDemux & demux, const ts::PESPacket & pkt) {
		printf("Got a packet!\n");
	}
};

int main(int argc, char ** argv) {
	const char * filename = "test0.ts";  // Default target filename to test0.ts
	if (argc >= 2) filename = argv[1];

	DemuxHandler     handler = DemuxHandler();
	ts::TSPacket     buffer[k_buff_len];
	ts::DuckContext  ctx = ts::DuckContext();
	ts::PESDemux     demuxer = ts::PESDemux(ctx, &handler);

	FILE * f = fopen(filename, "r");
	if (f == nullptr) { perror("Couldn't open file"); }

	size_t n_read;
	do {
		n_read = fread(buffer, sizeof(ts::TSPacket), k_buff_len, f);
		for (size_t i = 0; i < n_read; ++i) demuxer.feedPacket(buffer[i]);
	} while (n_read == k_buff_len);

	fclose(f);
}
```
When run on the file provided, it will output `Got a frame!` only twice.

fennewald, 2023-10-17:
As alluded to in the issue, I can think of two approaches to solving this:
1. Adding the `ts::PESDemux::flush`
2. Feeding in a dummy packet with a PUSI after processing

If option 1 is welcome, I will happily submit a PR with an implementation.

If option 2 is the desired approach, that is understandable. This issue can be closed immediately.

fennewald, 2023-10-17:
For an understanding of why this is the behavior, see `ts::PESDemux::processPacket`:

https://github.com/tsduck/tsduck/blob/0a5ae95ee5af173e21c2c5ce4168a4158d794d63/src/libtsduck/dtv/pes/tsPESDemux.cpp#L162-L276

From this snippet we can see that PESPackets are proessed in one of two cases:
1. Current packet has a PUSI and we have data queued to be processed (line 182)
2. Call `ts::PESDemux::processPESPacketIfComplete` if we have just added data and may be done (line 217 and 275)

`ts::PESDemux::processPESPacketIfComplete` will only process packets if they have a defined length, and we've read that entire length:

https://github.com/tsduck/tsduck/blob/0a5ae95ee5af173e21c2c5ce4168a4158d794d63/src/libtsduck/dtv/pes/tsPESDemux.cpp#L320-L334

Given all of this, we can see that neither of these conditions will be satisfied for a packet with undefined length and no following start code.

lelegard, 2023-10-18:
Hi @fennewald 

Your analysis is correct. But the solution is not obvious. Having a flush method would be fine. However, it is difficult to determine when and if it is appropriate to call it. Typically, at the end of a TS, shall we call it or not?

A TS file is rarely fully consistent at both ends (unless it is the direct output of a conversion tool such as ffmpeg). A TS file is usually a fragment of a capture, an HLS segment, a part of a larger stream. There is no guarantee that the file ends on a natural end of unbounded PES packets in all PID's. So, calling the flush method at the end of the file may result in truncated PES packets, without any clue if it is complete or not.

Moreover, even if the last unbounded PES packet in one PID is complete, this may not be the case on other PID's. So, there should be two versions of the flush method: one on a given PID and a general one for all PID's.

I can add this. But there should be no flush by default in TSDuck plugins and commands which use a PES demux. It will be a command line option, to be used on purpose by people knowing what they are doing.

lelegard, 2023-10-18:
I have added two methods `flushUnboundedPES()` and option `--flush-last-unbounded-pes` to plugin `pes`.

closed 2024-03-17 by github-actions[bot]