tsduck/tsduck #1324: [BUG]

jwsmite, 2023-10-18:
#### Bug description:
sdt plugin trying to use --patch-xml to add a component_descriptor. I get an error:
* Error: sdt: Incorrect value for attribute 'language_code' in <component_descriptor>, line 0, contains 0 characters, allowed 3 to 3

my xml has a language tag, 3 characters long 

#### How to reproduce:
tsp -I file myfile.ts -P sdt --patch-xml mypatch.xml -O file mypatchedfile.ts

#### Expected behavior:
Ideally the new component_descriptor would be inserted

#### Errors and logs:
* Error: sdt: Incorrect value for attribute 'language_code' in <component_descriptor>, line 0, contains 0 characters, allowed 3 to 3
* Error: sdt: Error in descriptor <component_descriptor> at line 0
* Error: sdt: error serializing binary table from the patched XML

#### Environment:
 - OS: Ubuntu
 - OS version: 23.04
 - TSDuck full version: 3.35-3258
 - Installation type: official dpkg from site

#### Additional information:
patch xml:
`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <SDT>
    <metadata PID="17"/>
    <service service_id="0x0064" EIT_schedule="false" EIT_present_following="true" CA_mode="false" running_status="running">
      <component_descriptor stream_content="0x09" stream_content_ext="0x00" component_type="0x05" component_tag="0x01" 
language_code="fra" x-node="add"/>
    </service>
  </SDT>
</tsduck>`

lelegard, 2023-10-18:
Hi @jwsmite 

I cannot reproduce your problem. Using an existing TS file, your patch file (changed the service id to match one of mine), same version of TSDuck and Ubuntu: the component descriptor is correctly added.

Could you provide an archive with the exact TS file and XML patch file? The smallest possible TS file, typically PID 17 only, would be preferred.

jwsmite, 2023-10-18:
I've filtered out just the sdt pid and uploaded here: [pid17.ts](https://dolby.box.com/s/d020wj97ce81bb8mvbpw9hj7tdfvn9jw)

jwsmite, 2023-10-18:
you asked for the patch xml too: [sdt_patch_xml](https://dolby.box.com/s/k1jshkvshtvlt6sozu39x5z56hosgwg4)

AlfonZ42, 2023-10-18:
(Originally) non-modified services _0x00C8_ & _0x012C_ in the _pid17.ts_ had empty _language_code_.
```xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <SDT>
    <metadata PID="17" />
    <service service_id="0x0064" EIT_schedule="false" EIT_present_following="true" CA_mode="false" running_status="running">
      <component_descriptor stream_content="0x09" stream_content_ext="0x00" component_type="0x05" component_tag="0x01" language_code="fra" x-node="add" />
    </service>
    <service service_id="0x00C8">
      <component_descriptor x-update-language_code="und"/>
    </service>
    <service service_id="0x012C">
      <component_descriptor x-update-language_code="und"/>
    </service>
  </SDT>
</tsduck>
```

lelegard, 2023-10-18:
As @AlfonZ42 noticed, the problem comes from the original SDT where the existing component descriptors have an empty language code (binary 00 00 00).

In practice, when using XML patch on a table on the fly, the process is the following:
1. Extract the sections until a complete table is found.
2. Deserialize the binary table as a C++ object of that table's class.
3. Convert the C++ object to XML (or at least internal representation of XML, not full text).
4. Apply the XML patch (a generic process which works on any kind of XML structure).
5. Convert the resulting XML into a C++ object.
6. Serialize the C++ object as sections.
7. Inject the sections.

The error occurs in step 5 because the two other services still have an empty language code in their XML structure.

The "line 0" annotation is misleading. The XML is entirely processed in internal representation, there is no XML text or file and consequently no "line".

The solution proposed by @AlfonZ42 works because it also fixes the empty language codes in other services.

As a generalization, I can propose the following patch file:
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <SDT>
    <service>
      <component_descriptor language_code="" x-update-language_code="und"/>
    </service>
    <service service_id="0x0064">
      <component_descriptor x-node="delete"/>
      <component_descriptor stream_content="0x09" stream_content_ext="0x00" component_type="0x05" component_tag="0x01" language_code="fra" x-node="add" />
    </service>
  </SDT>
</tsduck>
~~~
First, it fixes any existing component_descriptor with empty language code in a service and update it with `language_code="und"`.

Second, it deletes any pre-existing component_descriptor in service ID 0x0064.

Finally, it adds a fully specified component_descriptor in service ID 0x0064.

jwsmite, 2023-10-27:
I tried it, fixed an issue with Panasonic TVs not wanting to tune to my stream. Thanks!

closed 2024-03-31 by github-actions[bot]