tsduck/tsduck #1326: [Q] How to extract TS from PCAP files via tspcap ?

stulluk, 2023-10-21:
I have some PCAP files,  captured via Wireshark from IPTV / OTT STB's.

Some of these services broadcast as UDP, some as TCP streams.

When I analyse these pcap files as follows:

```bash
stulluk ~/TSFILES $  tspcap --list-streams X2.Haberturk.Liveplay.pcap 

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
10.XX.YY.74:8027       192.168.1.20:52098     UDP                5             136            2
10.XX.YY.74:32194      192.168.1.20:52097     UDP                1           1,328           15
10.XX.YY.74:32216      192.168.1.20:52097     UDP           11,817      15,691,848      180,278
10.XX.YY.55:33207      192.168.1.20:39554     TCP               15          38,640          444
10.XX.YY.55:33207      192.168.1.20:39580     TCP                7           4,412           51
10.XX.YY.55:33207      192.168.1.20:39582     TCP                7           4,412           51
10.XX.YY.90:6050       192.168.1.20:41834     TCP                2             352            4
10.XX.YY.92:8027       192.168.1.20:52098     UDP                2              80            1
176.XXX.7.11:123       192.168.1.20:34150     UDP                1              48            1
176.XX.7.11:123       192.168.1.20:34236     UDP                1              48            1
176.XX.7.11:123       192.168.1.20:35109     UDP                1              48            1
176.XXX.7.11:123       192.168.1.20:36112     UDP                1              48            1
176.XXX.7.11:123       192.168.1.20:37812     UDP                1              48            1
176.XXX.7.11:123       192.168.1.20:38676     UDP                1              48            1
```

Or another one:

```bash
stulluk ~/TSFILES $  tspcap --list-streams X2.Haberturk.Timeshift.pcap 

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
10.XX.YY.74:8027       192.168.1.20:44703     UDP                6             176            2
10.XX.YY.74:32170      192.168.1.20:44702     UDP            4,786       6,354,868       76,406
10.XX.YY.78:554        192.168.1.20:32768     TCP          115,861     606,781,937    7,295,435
10.XX.YY.55:33200      192.168.1.20:39004     TCP                2               0            0
10.XX.YY.55:33207      192.168.1.20:58674     TCP                5             696            8
10.XX.YY.55:33207      192.168.1.20:58746     TCP               16          40,855          491
10.XX.YY.90:6050       192.168.1.20:47560     TCP                2             352            4
10.XX.YY.92:554        192.168.1.20:49848     TCP                5             678            8
10.XX.YY.92:8027       192.168.1.20:44703     UDP                2              80            1
10.XX.YY.232:443       192.168.1.20:47766     TCP               11           4,592           55
142.TTT.187.110:443    192.168.1.20:50626     TCP               10             359            4
176.ZZZ.7.11:123       192.168.1.20:33425     UDP                1              48            1
176.ZZZ.7.11:123       192.168.1.20:33668     UDP                1              48            1
176.ZZZ.7.11:123       192.168.1.20:34086     UDP                1              48            1
176.ZZZ.7.11:123       192.168.1.20:34094     UDP                1              48            1
176.ZZZ.7.11:123       192.168.1.20:34735     UDP                1              48            1
```
So obviously, some of these transactions carry the TS streams inside.
 
Is there a way to extract the TS from these PCAP files via tsduck ?

Note: I RTFM but couldn't understand how to do it, sorry.
Note2: I tried using gstreamer method [here](https://ask.wireshark.org/question/19448/extract-ts-files-from-pcap-capture/) , it filtered large portion of the PCAP into TS, but generated TS is not a valid Transport Stream anymore.

lelegard, 2023-10-21:
Hi @stulluk 

The input plugin `pcap` extracts a TS from a UDP stream in a pcap file.
The command will be something like:
~~~
tsp -I pcap filename -d 192.168.1.20:44702 -O file outfile.ts
~~~

The `-d` option selects the destination address of the stream to select, if the pcap file contains several streams. The address:port is the example the likely stream in your second example. RTFM for additional options, typically for fine-grain filtering.

The example command simply extracts the TS into a file but any other plugin chain is accepted. The pcap input plugin is just a regular plugin.

stulluk, 2023-10-21:
Hi @lelegard 

Thank you but let me start with a simple question: It extracts TS from only UDP streams or is there a way to extract from TCP streams as well ?

Note: By the way, I confirm above example works but it only extract a very small portion (Only 6mbyte over 650+Mbyte recording) of the overall transaction since most of the playback transaction was on TCP there.

lelegard, 2023-10-21:
The typical use case is multicast for live TV, hence always UDP.

TCP use cases are HTTP or HLS (which is a lot of smaller HTTP sessions) but they are embedded in HTTP transport layer or, worse, in an SSL/TLS session (HTTPS). There is currently no way to exact these (and there will never be for HTTPS of course).

The input plugin `pcap` implements one specific TCP use case for EMMG<=>MUX protocol in packet mode. But this use case only addresses the EMM injection debug.

I don't know any common use case of TS over TCP.

stulluk, 2023-10-21:
Well for the above two examples, liveplay is transmitted as UDP and timeshifted playback was made as TCP. In both cases, the content is unencrypted (public national TV)

I am trying to find a way to filter out HTTP headers from the TCP packets when transmitted in timeshift mode..

Let me study more.

lelegard, 2023-10-22:
> I am trying to find a way to filter out HTTP headers from the TCP packets when transmitted in timeshift mode.

The input plugin `pcap` knows how to extract the binary content of a TCP session from a capture file. This is used to extract EMMG<=>MUX sessions, analyze the DVB SimulCrypt protocol, and extracts the EMM PID from it.

So, in practice, an enhancement of this plugin could extract the HTTP session and get the returned TS. This is just not done. Any volunteer for a PR?

lelegard, 2023-10-27:
Option `--http` has been added to input plugin `pcap`.

To extract or process a TS which is returned by an HTTP server:
First, list all TCP streams in the pcap file:
~~~
$ tspcap --list --tcp test.pcapng

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
192.168.56.1:56071     192.168.56.101:80      TCP            4,955             411          587
192.168.56.101:80      192.168.56.1:56071     TCP              302      14,409,007   20,595,929
~~~

Now, use the `pcap` input plugin to read the TS from the HTTP stream:
~~~
$ tsp -I pcap test.pcapng --http --source 192.168.56.101:80 -O file test.ts
~~~

Here, we only save the TS file. But it can be any other chain of plugins.

stulluk, 2023-10-27:
Hi @lelegard , thank you so much for this amazing open source project and your amazing efforts.

I rebuilt tsduck on my ubuntu 22.04 machine and tried above:

```bash
stulluk ~/TSFILES $  ll X2.Haberturk.Timeshift.pcap 
-rw-rw-r-- 1 stulluk stulluk 613M Eki 22 00:39 X2.Haberturk.Timeshift.pcap
stulluk ~/TSFILES $
stulluk ~/TSFILES $  tspcap --list --tcp X2.Haberturk.Timeshift.pcap 

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
10.XX.51.78:554        192.168.1.20:32768     TCP          115,861     606,781,937    7,295,528
10.XX.72.55:33200      192.168.1.20:39004     TCP                2               0            0
10.XX.72.55:33207      192.168.1.20:58674     TCP                5             696            8
10.XX.72.55:33207      192.168.1.20:58746     TCP               16          40,855          491
10.XX.72.90:6050       192.168.1.20:47560     TCP                2             352            4
10.XX.72.92:554        192.168.1.20:49848     TCP                5             678            8
10.XX.79.232:443       192.168.1.20:47766     TCP               11           4,592           55
142.250.187.110:443    192.168.1.20:50626     TCP               10             359            4
192.168.1.20:5555      192.168.1.25:54056     TCP              662             113            1
192.168.1.20:32768     10.XX.51.78:554        TCP          113,297          11,758          141
192.168.1.20:38876     192.168.1.1:853        TCP                7               0            0
192.168.1.20:39004     10.XX.72.55:33200      TCP               12               0            0
192.168.1.20:43284     10.XX.44.1:853         TCP                7               0            0
192.168.1.20:47560     10.XX.72.90:6050       TCP                4             360            4
192.168.1.20:47766     10.XX.79.232:443       TCP               10           1,250           15
192.168.1.20:48718     10.XX.73.127:33500     TCP                1               0            0
192.168.1.20:49848     10.XX.72.92:554        TCP                7             766            9
192.168.1.20:50626     142.250.187.110:443    TCP                7             428            5
192.168.1.20:56906     10.XX.45.1:853         TCP                7               0            0
192.168.1.20:58674     10.XX.72.55:33207      TCP                6             962           12
192.168.1.20:58746     10.XX.72.55:33207      TCP               26           9,728          117
192.168.1.25:54056     192.168.1.20:5555      TCP              661             683            8

stulluk ~/TSFILES $  tsp -I pcap X2.Haberturk.Timeshift.pcap --http --source 10.XX.51.78:554 -O file test.ts
stulluk ~/TSFILES $  ll test.ts 
-rw-rw-r-- 1 stulluk stulluk 6,7M Eki 28 00:54 test.ts
stulluk ~/TSFILES $  tsp -I pcap X2.Haberturk.Timeshift.pcap --http --source 10.XX.51.78:554 --destination 192.168.1.20:32768 -O file test2.ts
stulluk ~/TSFILES $  ll test*
-rw-rw-r-- 1 stulluk stulluk 6,7M Eki 28 00:55 test2.ts
-rw-rw-r-- 1 stulluk stulluk 6,7M Eki 28 00:54 test.ts
stulluk ~/TSFILES $  tsp --version 
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.36-3350
stulluk ~/TSFILES $
```

As you see, in this capture, the source PCAP is 613Mbyte and I expect output TS file to have a size around 600Mbyte but it extracts only 6.7Megabyte. 

Is this due to a discontiniuty error in the Source PCAP file  or am I misusing the new functionality ?

lelegard, 2023-10-27:
I just fixed an issue if the start of the TCP session was in the capture. Sorry about that, you will have to pull again and rebuild.

stulluk, 2023-10-27:
@lelegard I cloned&rebuilt but result is the same. I am wondering if this is still related to missing packets in the PCAP:

```bash
stulluk ~/TSFILES $  tsp --verbose -I pcap X2.Haberturk.Timeshift.pcap --http --destination 192.168.1.20:32768 -O file test6.ts
* pcap: selected stream 10.31.51.78:554 <-> 192.168.1.20:32768
* pcap: unknown initial input bitrate
* file: creating file test6.ts
stulluk ~/TSFILES $  ll test6.ts 
-rw-rw-r-- 1 stulluk stulluk 6,7M Eki 28 01:23 test6.ts
stulluk ~/TSFILES $  tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.36-3351
stulluk ~/TSFILES $  tspcap --list --tcp X2.Haberturk.Timeshift.pcap 

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
10.XX.51.78:554        192.168.1.20:32768     TCP          115,861     606,781,937    7,295,528
10.XX.72.55:33200      192.168.1.20:39004     TCP                2               0            0
10.XX.72.55:33207      192.168.1.20:58674     TCP                5             696            8
10.XX.72.55:33207      192.168.1.20:58746     TCP               16          40,855          491
10.XX.72.90:6050       192.168.1.20:47560     TCP                2             352            4
10.XX.72.92:554        192.168.1.20:49848     TCP                5             678            8
10.XX.79.232:443       192.168.1.20:47766     TCP               11           4,592           55
142.250.187.110:443    192.168.1.20:50626     TCP               10             359            4
192.168.1.20:5555      192.168.1.25:54056     TCP              662             113            1
192.168.1.20:32768     10.XX.51.78:554        TCP          113,297          11,758          141
192.168.1.20:38876     192.168.1.1:853        TCP                7               0            0
192.168.1.20:39004     10.XX.72.55:33200      TCP               12               0            0
192.168.1.20:43284     10.XX.44.1:853         TCP                7               0            0
192.168.1.20:47560     10.XX.72.90:6050       TCP                4             360            4
192.168.1.20:47766     10.XX.79.232:443       TCP               10           1,250           15
192.168.1.20:48718     10.XX.73.127:33500     TCP                1               0            0
192.168.1.20:49848     10.XX.72.92:554        TCP                7             766            9
192.168.1.20:50626     142.250.187.110:443    TCP                7             428            5
192.168.1.20:56906     10.XX.45.1:853         TCP                7               0            0
192.168.1.20:58674     10.XX.72.55:33207      TCP                6             962           12
192.168.1.20:58746     10.XX.72.55:33207      TCP               26           9,728          117
192.168.1.25:54056     192.168.1.20:5555      TCP              661             683            8

stulluk ~/TSFILES $  
```

lelegard, 2023-10-27:
> I am wondering if this is still related to missing packets in the PCAP:

It depends on what you mean by "missing packets".

If you mean missing TS packets in the transmitted file, then no, it won't make a difference.

If you mean missing network datagrams from the capture, then yes. If the continuity of the TCP stream is broken, then you cannot reassemble it and the extraction probably stops at the missing datagram, looking forward in the stream to find it (datagrams can be transmitted out of order), never finding it and stopping there.

The actual TCP session between client and server correctly exchanged the datagrams, correctly reassembled. But if one or more are missing from the capture, then the TCP stream is broken.

stulluk, 2023-10-27:
This makes perfect sense. So, my final question before closing this issue with a thanks:  From the following log, can we conclude that there is a discontiniuity in the TCP packets ( datagrams ) ?

```bash
stulluk ~/TSFILES $  tsp --debug --verbose -I pcap X4.Haberturk.Timeshift.pcap  --http --destination 192.168.1.46:52358 -O file out1.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug --verbose
* Debug: debug level set to 1
* Debug: trying to load "/usr/bin/tsplugin_pcap.so"
* Debug: /usr/bin/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/pcap.so"
* Debug: /usr/bin/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_pcap.so"
* Debug: /usr/lib64/tsduck/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/pcap.so"
* Debug: /usr/lib64/tsduck/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_pcap.so"
* Debug: /usr/lib64/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/pcap.so"
* Debug: /usr/lib64/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_pcap.so"
* Debug: registering input plugin "pcap", status: ok
* Debug: pcap: debug level set to 1
* Debug: pcap: ====> tsp -I pcap X4.Haberturk.Timeshift.pcap --http --destination 192.168.1.46:52358
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file out1.ts
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: pcap: opened X4.Haberturk.Timeshift.pcap, pcap format version 2.4, little endian
* pcap: selected stream 10.XX.51.78:554 <-> 192.168.1.46:52358
* Debug: pcap: at start of HTTP session, server: 10.XX.51.78:554
* Debug: pcap: response header: RTSP/1.0 200 OK
* Debug: pcap: response header: Server: HMS_V1R2
* Debug: pcap: response header: CSeq: 1
* Debug: pcap: response header: Date: Thu, 28 Sep 2023 10:29:35 GMT
* Debug: pcap: response header: Session: 444708699
* Debug: pcap: response header: Timeshift-Status: 0
* Debug: pcap: response header: Public: DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE
* Debug: pcap: response header: 
* Debug: pcap: initial buffer load: 6 packets, 1,128 bytes
* pcap: unknown initial input bitrate
* file: creating file out1.ts
* Debug: pcap: input thread started
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: file: output thread started
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: debug level set to 1
* Debug: control commands: found 0 packet processor plugins
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 64 bytes (original: 78 bytes), link type: 0
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: no TS packet in message, 65424 bytes
* Debug: pcap: pcap file already in error state
* Debug: pcap: stopping the input plugin
* Debug: file: stopping the output plugin
* Debug: pcap: input thread terminated after 36,218 packets
* Debug: file: output thread terminated after 36,218 packets (36,218 output)
* Debug: Report logging thread terminated
stulluk ~/TSFILES $  ll out1.ts 
-rw-rw-r-- 1 stulluk stulluk 6,5M Eki 28 02:30 out1.ts
stulluk ~/TSFILES $  mediainfo out1.ts 
General
ID                                       : 1 (0x1)
Complete name                            : out1.ts
Format                                   : MPEG-TS
File size                                : 6.49 MiB
Duration                                 : 8 min 9 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 84.8 kb/s

Video
ID                                       : 256 (0x100)
Menu ID                                  : 1 (0x1)
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Main@L4
Format settings                          : CABAC / 2 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 2 frames
Codec ID                                 : 27
Duration                                 : 2 min 58 s
Bit rate mode                            : Variable
Maximum bit rate                         : 7 500 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate                               : 25.000 FPS
Color space                              : YUV
...
```

lelegard, 2023-10-28:
I see this:
~~~
* Debug: pcap: pcap file already in error state
~~~
At some point, reading the pcap file stops because it is considered as erroneous. I don't see an obvious reason for that in the previous traces. Looking at the code, the message _"pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes)"_ is just a warning, an invalid record which is skipped and the file processing goes on. So, the cause must be somewhere else.

If you have the opportunity to upload the pcap file to some file transfer service, I could analyse the issue.

Potentially, another problem (but not his one) can also come from here:
~~~
* Debug: pcap: response header: RTSP/1.0 200 OK
~~~
The server uses RTSP, not HTTP. I originally did not pay attention to the server port, 554, the RTSP port number, and not 80 the HTTP server port. The RTSP response stream may contain interleaves control messages in addition to TS packets. So, this may be another source of issue.

stulluk, 2023-10-28:
@lelegard  I sent you one email, containing the links to the files above. Please let me know if you succesfully received the files or not.

lelegard, 2023-10-28:
@stulluk, after analysing the file and the way the plugin processes it, the file is correct, no TCP segment is missing.

However, the problem is RTSP.

In TCP mode, the `pcap` plugin can only extract a continuous suite of TS packets, after skipping optional HTTP headers.

Here, the client and server use the RTSP protocol. The client sends RTSP commands and the server responds. I see repeated RTSP commands such as DESCRIBE, OPTIONS, SETUP, PLAY, GET_PARAMETER, etc. Depending on the command, there can be different responses formats. Each response starts with headers (in the middle of the TCP stream).

So, the TCP stream from the server to the client is not a simple suite of TS packets as an HTTP server would return and the `pcap` plugin loses its synchronisation at some point.

This kind of TCP session cannot be extracted, sorry.

stulluk, 2023-10-28:
@lelegard  thank you so much looking into it. 

However, I am trying to learn, how does RTSP player/library/software running on the client is extracting and passing raw TS to demux / decoder in this mechanism ? 

If you have any insights, such as if there is a way to pass filtered output to ffmpeg ...etc just let me know.

If not, thank you so much, and we can close this issue.

lelegard, 2023-10-28:
Simply looking the Wikipedia page on RTSP, it seems not too difficult to extract the TS from the TCP stream. In addition to HTTP, we need to be prepared to skip HTTP-like response headers at any time in the reponse stream. Second, RTSP adds an optional '$'-encapsulation of TS packets rather easy to detect and remove.

I have added this in the `pcap` plugin and it know extracts the complete TS from the provided file. Please let us know if it works for you.

stulluk, 2023-10-29:
@lelegard good morning and you are definitely the best.

Here is my test result for X2.Haberturk.Timeshift.pcap

```bash
stulluk ~/TSFILES $  tspcap --list --tcp X2.Haberturk.Timeshift.pcap 

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
10.XX.51.78:554        192.168.1.20:32768     TCP          115,861     606,781,937    7,295,528
10.XX.72.55:33200      192.168.1.20:39004     TCP                2               0            0
10.XX.72.55:33207      192.168.1.20:58674     TCP                5             696            8
10.XX.72.55:33207      192.168.1.20:58746     TCP               16          40,855          491
10.XX.72.90:6050       192.168.1.20:47560     TCP                2             352            4
10.XX.72.92:554        192.168.1.20:49848     TCP                5             678            8
10.XX.79.232:443       192.168.1.20:47766     TCP               11           4,592           55
142.250.187.110:443    192.168.1.20:50626     TCP               10             359            4
192.168.1.20:5555      192.168.1.25:54056     TCP              662             113            1
192.168.1.20:32768     10.XX.51.78:554        TCP          113,297          11,758          141
192.168.1.20:38876     192.168.1.1:853        TCP                7               0            0
192.168.1.20:39004     10.XX.72.55:33200      TCP               12               0            0
192.168.1.20:43284     10.XX.44.1:853         TCP                7               0            0
192.168.1.20:47560     10.XX.72.90:6050       TCP                4             360            4
192.168.1.20:47766     10.XX.79.232:443       TCP               10           1,250           15
192.168.1.20:48718     10.XX.73.127:33500     TCP                1               0            0
192.168.1.20:49848     10.XX.72.92:554        TCP                7             766            9
192.168.1.20:50626     142.250.187.110:443    TCP                7             428            5
192.168.1.20:56906     10.XX.45.1:853         TCP                7               0            0
192.168.1.20:58674     10.XX.72.55:33207      TCP                6             962           12
192.168.1.20:58746     10.XX.72.55:33207      TCP               26           9,728          117
192.168.1.25:54056     192.168.1.20:5555      TCP              661             683            8

stulluk ~/TSFILES $  tsp -I pcap X2.Haberturk.Timeshift.pcap --http --destination 192.168.1.20:32768 -O file rtsp.out.ts
stulluk ~/TSFILES $  ll rtsp.out.ts 
-rw-rw-r-- 1 stulluk stulluk 577M Eki 29 09:53 rtsp.out.ts
stulluk ~/TSFILES $  mpv rtsp.out.ts 
[ffmpeg/demuxer] mpegts: Packet corrupt (stream = 0, dts = 594432XX68).
[ffmpeg/video] h264: error while decoding MB 16 22, bytestream -8
 (+) Video --vid=1 (h264 1920x1080 25.000fps)
 (+) Audio --aid=1 --alang=tur (aac 2ch 48000Hz)
Cannot load libcuda.so.1
[ffmpeg/demuxer] mpegts: Packet corrupt (stream = 0, dts = 594432XX68).
AO: [pulse] 48000Hz stereo 2ch float
Using hardware decoding (vaapi).
VO: [gpu] 1920x1080 vaapi[nv12]
AV: 26:04:44 / 26:12:07 (100%) A-V:  0.000

Exiting... (Quit)
stulluk ~/TSFILES $  tsp --version 
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.36-3353
stulluk ~/TSFILES $  
```

Perfectly extracted, thank you.

And here is my test result for X4.Haberturk.Timeshift.pcap

```bash
stulluk ~/TSFILES $  ll X4.Haberturk.Timeshift.pcap 
-rw-rw-r-- 1 stulluk stulluk 1,5G Eki 28 02:28 X4.Haberturk.Timeshift.pcap
stulluk ~/TSFILES $  tspcap --list --tcp X4.Haberturk.Timeshift.pcap 
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 64 bytes (original: 78 bytes), link type: 0

Source                 Destination            Protocol     Packets      Data bytes      Bitrate
10.XX.51.78:554        192.168.1.46:52358     TCP          378,992     594,993,727    7,265,468
10.XX.72.42:37051      192.168.1.46:56532     TCP               11           1,699           21
10.XX.72.42:37051      192.168.1.46:56652     TCP               11           1,699           21
10.XX.72.55:33200      192.168.1.46:56746     TCP                2               0            0
10.XX.72.55:33200      192.168.1.46:56756     TCP                2               0            0
10.XX.72.55:33200      192.168.1.46:56758     TCP                2               0            0
[... 61 lines cut ...]
192.168.1.46:36908     10.XX.72.55:33207      TCP                8             568            7
192.168.1.46:38050     108.177.126.188:5228   TCP               22             682            8
192.168.1.46:38052     108.177.126.188:5228   TCP                4              56            1
192.168.1.46:38126     85.g.8:8015        TCP               12           6,142           75
192.168.1.46:39252     85.g.8:8015        TCP               12           6,143           75
192.168.1.46:39960     10.XX.72.90:6050       TCP                4             226            3
192.168.1.46:41288     142.251.140.42:443     TCP               11             591            7
192.168.1.46:42180     85.g.8:8015        TCP               12           6,142           75
192.168.1.46:42796     85.g.8:8015        TCP               12           6,141           75
stulluk ~/TSFILES $  # This file is 1.5Gbyte, and most of the transaction happened from 554 and 5555 ports from the source
[... cut ...]
* Error: pcap: content error, neither HTTP reponse headers nor TS packets in TCP stream
stulluk ~/TSFILES $
```

And I suddenly realised that, during this PCAP capture, I connected to Android OTT box via ADB network to capture logcat. And by default, ADB uses port 5555 , so that transaction was definitely not the TS content, it was just the logcat that I was capturing.

Thank you for your amazing efforts & support, and thank you for showing the world how a succesful opensource project can achieve.

Closing the issue with regards & thanks.

stulluk, 2023-10-30:
Hi @lelegard  , sorry, I reopened this issue.

I have been able to extract most of the PCAP files with the latest codebase.

However, when using two PCAP files, I haven't been able to do so.

1st file: "erroneous.pcap" : The extraction starts, and at the end , it shows and error and no file extracted.
2nd file: "unending.pcap" : The extractions starts, and even I wait for about 30 minutes, it still continue but no file extracted.

Both files are about 1.7 Gbyte

I uploaded them to the google drive link I sent to you.

Would it be possible to check these two files, to let me understand if the files has some error or latest tsduck has a bug or not?

lelegard, 2023-10-30:
I will have a look. Could you please specify which command you use? To be sure about the streams you try to extract.

stulluk, 2023-10-30:
Here you are (example for erroneous.pcap):

```bash
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  ll
total 2,8G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 12:58 ./
drwxr-x--- 65 stulluk stulluk 4,0K Eki 30 12:58 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  tspcap --list-streams erroneous.pcap  --no-pager | sed -e '1,2d' | sed 's/,//g' | sort -nrk 5 | head -n10
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
10.31.51.78:554        192.168.1.34:54682     TCP          750326   1222558404    7494668
192.168.1.34:5555      192.168.1.28:50512     TCP          463996     376920013    2310638
192.168.1.28:50512     192.168.1.34:5555      TCP          300348       2989416       18326
10.31.128.36:33207     192.168.1.34:58764     TCP               85         125563          770
192.168.1.34:54682     10.31.51.78:554        TCP          363715          19668          121
192.168.1.34:58764     10.31.128.36:33207     TCP               53          15386           94
192.168.1.34:5353      224.0.0.251:5353       UDP               61          12161           75
192.168.1.1:57128      239.255.255.250:1900   UDP               84          11130           68
142.251.140.46:443     192.168.1.34:46060     TCP               18           9319           57
192.168.1.34:36624     63.35.92.152:443       TCP               18           8452           52
stulluk ~/PCAPS $  tsp -I pcap erroneous.pcap --http --destination 192.168.1.34:54682 -O file test.ts
^C* tsp: user interrupt, terminating...
^C
stulluk ~/PCAPS $  tsp --debug -I pcap erroneous.pcap --http --destination 192.168.1.34:54682 -O file test.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: trying to load "/usr/bin/tsplugin_pcap.so"
* Debug: /usr/bin/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/pcap.so"
* Debug: /usr/bin/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_pcap.so"
* Debug: /usr/lib64/tsduck/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/pcap.so"
* Debug: /usr/lib64/tsduck/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_pcap.so"
* Debug: /usr/lib64/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/pcap.so"
* Debug: /usr/lib64/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_pcap.so"
* Debug: registering input plugin "pcap", status: ok
* Debug: pcap: debug level set to 1
* Debug: pcap: ====> tsp -I pcap erroneous.pcap --http --destination 192.168.1.34:54682
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file test.ts
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: pcap: opened erroneous.pcap, pcap format version 2.4, little endian
* pcap: selected stream 10.31.51.78:554 <-> 192.168.1.34:54682
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Error: pcap: content error, neither HTTP reponse headers nor TS packets in TCP stream
* Debug: pcap: no initial packet read
* Debug: init buffer error
* Debug: Report logging thread terminated
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  ll
total 2,8G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 12:58 ./
drwxr-x--- 65 stulluk stulluk 4,0K Eki 30 12:58 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
stulluk ~/PCAPS $  
```

lelegard, 2023-10-30:
The file erroneous.pcap was not properly handled because of an issue in TCP sequence number wrap-up at 2^32. This wrap-up occurred after approximately 90 MB of TS data. All following TCP packets were incorrectly considered as "packets from the past" and were ignored.

The problem is fixed now.

I see no problem with the file unending.pcap. The 734 MB TS is extracted in 4 seconds. It is unrelated to the previous TCP sequence issue. The extraction is fine with and without the fix.

You didn't mention the command with unending.pcap, only some commands with erroneous.pcap. Are you sure that the command is not somewhat malformed and waits for something, forever?

stulluk, 2023-10-30:
Hi, regarding "unending.pcap" (before upgrading tsduck )

```bash
stulluk ~/PCAPS $  ll 
total 2,8G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 12:58 ./
drwxr-x--- 66 stulluk stulluk 4,0K Eki 30 18:55 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  tspcap --list-streams unending.pcap --no-pager | sed -e '1,2d' | sed 's/,//g' | sort -nrk 5 | head -n10
tspcap: invalid IPv4 datagram in pcap file, 64 bytes (original: 78 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
10.31.51.78:554        192.168.1.34:54682     TCP          474907     772047497    7446833
192.168.1.34:5555      192.168.1.28:50512     TCP          431446     316972292    3057376
192.168.1.28:50512     192.168.1.34:5555      TCP          300645       3157896       30460
192.168.1.34:54682     10.31.51.78:554        TCP          229705          15458          149
10.31.128.36:33207     192.168.1.34:58764     TCP               19          10985          106
192.168.1.34:58764     10.31.128.36:33207     TCP               21           9245           89
192.168.1.34:5353      224.0.0.251:5353       UDP               37           8837           85
192.168.1.34:35848     85.29.13.8:8015        TCP               13           7548           73
192.168.1.34:54062     85.29.13.8:8015        TCP               13           7547           73
192.168.1.34:48574     85.29.13.8:8015        TCP               13           7547           73
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  time tsp --debug -I pcap unending.pcap --http --destination 192.168.1.34:54682 -O file test.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: trying to load "/usr/bin/tsplugin_pcap.so"
* Debug: /usr/bin/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/pcap.so"
* Debug: /usr/bin/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_pcap.so"
* Debug: /usr/lib64/tsduck/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/pcap.so"
* Debug: /usr/lib64/tsduck/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_pcap.so"
* Debug: /usr/lib64/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/pcap.so"
* Debug: /usr/lib64/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_pcap.so"
* Debug: registering input plugin "pcap", status: ok
* Debug: pcap: debug level set to 1
* Debug: pcap: ====> tsp -I pcap unending.pcap --http --destination 192.168.1.34:54682
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file test.ts
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: pcap: opened unending.pcap, pcap format version 2.4, little endian
* pcap: selected stream 10.31.51.78:554 <-> 192.168.1.34:54682
^C* tsp: user interrupt, terminating...
* Debug: aborting all plugins...
* Debug: aborting plugin pcap
* Debug: aborting plugin file
^C

real	0m52,718s
user	0m52,568s
sys	0m0,136s
stulluk ~/PCAPS $ 
stulluk ~/PCAPS $  tsp --version 
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.36-3353
stulluk ~/PCAPS $  
```

Now, after upgrade:

[... 130 lines cut ...]
* Debug: file: ====> tsp -O file test.ts
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: pcap: opened erroneous.pcap, pcap format version 2.4, little endian
* pcap: selected stream 10.31.51.78:554 <-> 192.168.1.34:54682
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Debug: pcap: start in middle of HTTP session, initial read: 19,668 bytes, status: true
* Error: pcap: content error, neither HTTP reponse headers nor TS packets in TCP stream
* Debug: pcap: no initial packet read
* Debug: init buffer error
* Debug: Report logging thread terminated

real	1m2,877s
user	1m2,329s
sys	0m0,532s
stulluk ~/PCAPS $  ll
total 2,8G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 12:58 ./
drwxr-x--- 66 stulluk stulluk 4,0K Eki 30 18:55 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
stulluk ~/PCAPS $ 
```
Situation is unchanged. Can you please let me know, what I am doing wrong here ?

stulluk, 2023-10-30:
Regarding the performance of my Machine:

```bash
stulluk ~/PCAPS $  ll
total 4,2G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 19:15 ./
drwxr-x--- 66 stulluk stulluk 4,0K Eki 30 18:55 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
-rw-rw-r--  1 stulluk stulluk 1,5G Eki 30 19:15 X4.Haberturk.Timeshift.pcap
stulluk ~/PCAPS $  md5sum X4.Haberturk.Timeshift.pcap 
0c2617e478727142e09e16cab1f175c4  X4.Haberturk.Timeshift.pcap
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  tspcap --list-streams X4.Haberturk.Timeshift.pcap   --no-pager | sed -e '1,2d' | sed 's/,//g' | sort -nrk 5 | head -n10
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 64 bytes (original: 78 bytes), link type: 0
10.31.51.78:554        192.168.1.46:52358     TCP          378992     594993727    7265357
176.235.12.68:8050     233.26.166.177:5000    UDP          366919     487268432    5949943
192.168.1.46:5555      192.168.1.57:61163     TCP          362177     310377639    3789963
10.31.51.75:32152      192.168.1.46:55056     UDP            4726       6275188       76625
176.235.12.68:8050     233.26.166.106:5000    UDP            4514       5994592       73199
192.168.1.57:61163     192.168.1.46:5555      TCP          215985       2238834       27338
10.31.72.55:33207      192.168.1.46:60944     TCP               33          41537          507
192.168.1.46:49488     54.77.110.160:443      TCP               30          25188          308
192.168.1.46:52880     142.251.140.42:443     UDP               31          15291          187
192.168.1.57:61158     192.168.1.46:8009      TCP              262          14410          176
stulluk ~/PCAPS $  
stulluk ~/PCAPS $  time tsp --debug -I pcap X4.Haberturk.Timeshift.pcap --http --destination 192.168.1.46:52358 -O file test.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: trying to load "/usr/bin/tsplugin_pcap.so"
* Debug: /usr/bin/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/pcap.so"
* Debug: /usr/bin/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_pcap.so"
* Debug: /usr/lib64/tsduck/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/pcap.so"
* Debug: /usr/lib64/tsduck/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_pcap.so"
* Debug: /usr/lib64/tsplugin_pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/pcap.so"
* Debug: /usr/lib64/pcap.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_pcap.so"
* Debug: registering input plugin "pcap", status: ok
* Debug: pcap: debug level set to 1
* Debug: pcap: ====> tsp -I pcap X4.Haberturk.Timeshift.pcap --http --destination 192.168.1.46:52358
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file test.ts
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: pcap: opened X4.Haberturk.Timeshift.pcap, pcap format version 2.4, little endian
* pcap: selected stream 10.31.51.78:554 <-> 192.168.1.46:52358
* Debug: pcap: at start of HTTP session, server: 10.31.51.78:554
* Debug: pcap: response header: RTSP/1.0 200 OK
* Debug: pcap: response header: Server: HMS_V1R2
* Debug: pcap: response header: CSeq: 1
* Debug: pcap: response header: Date: Thu, 28 Sep 2023 10:29:35 GMT
* Debug: pcap: response header: Session: 444708699
* Debug: pcap: response header: Timeshift-Status: 0
* Debug: pcap: response header: Public: DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE
* Debug: pcap: response header: RTSP/1.0 200 OK
* Debug: pcap: response header: Server: HMS_V1R2
* Debug: pcap: response header: CSeq: 2
* Debug: pcap: response header: Date: Thu, 28 Sep 2023 10:29:35 GMT
* Debug: pcap: response header: Session: 444708699
* Debug: pcap: response header: Timeshift-Status: 1
* Debug: pcap: response header: Content-Length: 134
* Debug: pcap: response header: Content-Type: application/sdp
* Debug: pcap: response header: Content-Base: rtsp://10.31.51.78:554/PLTV/88888888/224/3221227320/9327.smil?rrsip=10.31.72.92&zoneoffset=0&icpid=88888888&limitflux=-1&limitdur=-1&tenantId=9001&GuardEncType=2&accountinfo=%2C%2C%2C20230928102230%2Curn%3AHuawei%3AliveTV%3A001300000008479171%2C%2C-1%2C0%2C1%2C%2C%2C7%2C%2C%2C%2C4%2C%2C2%2C%2C%2C%2C%2C1%2C1%2C9327%2CEND&from=7&hms_devid=5604&online=1695896975/
[... 176 lines cut ...]
* Debug: pcap: readTCP failed, read size: 0 bytes, position in file: 1,516,801,549
* Debug: pcap: readTCP failed, read size: 0 bytes, position in file: 1,516,801,549
* Debug: pcap: stopping the input plugin
* Debug: pcap: max TCP reassembly queue size: 26 data blocks
* Debug: file: stopping the output plugin
* Debug: pcap: input thread terminated after 3,154,956 packets
* Debug: file: output thread terminated after 3,154,956 packets (3,154,956 output)
* Debug: Report logging thread terminated

real	0m1,907s
user	0m1,600s
sys	0m0,637s
stulluk ~/PCAPS $  ll
total 4,8G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 19:16 ./
drwxr-x--- 66 stulluk stulluk 4,0K Eki 30 19:16 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-rw-r--  1 stulluk stulluk 566M Eki 30 19:16 test.ts
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
-rw-rw-r--  1 stulluk stulluk 1,5G Eki 30 19:15 X4.Haberturk.Timeshift.pcap
stulluk ~/PCAPS $  

```

this is a Ryzen 5700G with 64Gbyte DDR + 2Terabyte Samsung 980 Pro NVME.

lelegard, 2023-10-30:
> Regarding the performance of my Machine:

Nice. And even better without the debug traces, I presume.

Concerning your failing or blocking commands, you need to specify the source address at least (with or without destination address, depending if you have several TCP streams from the source). I did all my tests using the source address and it worked.

It is not normal that specifying the destination only ends up in a CPU-bound loop. I will investigate. In the meantime, make sure to specify `--source` (with or without `--destination`).

lelegard, 2023-10-30:
It should be fine with `--destination` only now.

stulluk, 2023-10-30:
Worked perfectly when adding "--source" .. The only issue was, at the end of extraction, it returned following : 

```bash
...
* Error: pcap: content error, neither HTTP reponse headers nor TS packets in TCP stream
```
And the content ( final TS file ) was perfectly correct.

Then I upgrade to 3.36-3358 , and now, it even works without "--source" : 

```bash
stulluk ~/PCAPS $  ll
total 6,1G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 21:33 ./
drwxr-x--- 66 stulluk stulluk 4,0K Eki 30 21:36 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-rw-r--  1 stulluk stulluk 1,2G Eki 30 21:28 erroneous.pcap.TCP.192.168.1.34.54682.ts
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
-rw-rw-r--  1 stulluk stulluk 734M Eki 30 21:32 unending.pcap.TCP.192.168.1.34.54682.ts
-rw-rw-r--  1 stulluk stulluk 1,5G Eki 30 19:15 X4.Haberturk.Timeshift.pcap
stulluk ~/PCAPS $  rm *.ts
stulluk ~/PCAPS $  extract-ts-from-pcap.drj erroneous.pcap 
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
10.31.51.78:554        192.168.1.34:54682     TCP          750326   1222558404    7494668
192.168.1.34:5555      192.168.1.28:50512     TCP          463996     376920013    2310638
192.168.1.28:50512     192.168.1.34:5555      TCP          300348       2989416       18326
10.31.128.36:33207     192.168.1.34:58764     TCP               85         125563          770
192.168.1.34:54682     10.31.51.78:554        TCP          363715          19668          121
192.168.1.34:58764     10.31.128.36:33207     TCP               53          15386           94
192.168.1.34:5353      224.0.0.251:5353       UDP               61          12161           75
192.168.1.1:57128      239.255.255.250:1900   UDP               84          11130           68
142.251.140.46:443     192.168.1.34:46060     TCP               18           9319           57
192.168.1.34:36624     63.35.92.152:443       TCP               18           8452           52
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
Current line is 10.31.51.78:554        192.168.1.34:54682     TCP          750326   1222558404    7494668 
This line is TCP
DATASIZE is 1222558404
Size is larger than 10megabyte
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
* Warning: pcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
Current line is 192.168.1.34:5555      192.168.1.28:50512     TCP          463996     376920013    2310638 
This line is TCP
DATASIZE is 376920013
Size is larger than 10megabyte
* Error: pcap: content error, neither HTTP reponse headers nor TS packets in TCP stream
stulluk ~/PCAPS $  ll
total 5,3G
drwxrwxr-x  2 stulluk stulluk 4,0K Eki 30 21:41 ./
drwxr-x--- 66 stulluk stulluk 4,0K Eki 30 21:36 ../
-rw-r--r--  1 stulluk stulluk 1,7G Eki 30 12:57 erroneous.pcap
-rw-rw-r--  1 stulluk stulluk 1,2G Eki 30 21:41 erroneous.pcap.TCP.192.168.1.34.54682.ts
-rw-r--r--  1 stulluk stulluk 1,2G Eki 30 12:58 unending.pcap
-rw-rw-r--  1 stulluk stulluk 1,5G Eki 30 19:15 X4.Haberturk.Timeshift.pcap
stulluk ~/PCAPS $  extract-ts-from-pcap.drj unending.pcap 
tspcap: invalid IPv4 datagram in pcap file, 64 bytes (original: 78 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
tspcap: invalid IPv4 datagram in pcap file, 46 bytes (original: 60 bytes), link type: 0
10.31.51.78:554        192.168.1.34:54682     TCP          474907     772047497    7446833
192.168.1.34:5555      192.168.1.28:50512     TCP          431446     316972292    3057376
192.168.1.28:50512     192.168.1.34:5555      TCP          300645       3157896       30460
192.168.1.34:54682     10.31.51.78:554        TCP          229705          15458          149
10.31.128.36:33207     192.168.1.34:58764     TCP               19          10985          106
192.168.1.34:58764     10.31.128.36:33207     TCP               21           9245           89
192.168.1.34:5353      224.0.0.251:5353       UDP               37           8837           85
192.168.1.34:35848     85.29.13.8:8015        TCP               13           7548           73
[... 86 lines cut ...]
		# Get destination IP and PORT
		DESTINATION_IP=$(printf "%s" "${line}" | awk -F' ' '{print $2}' | awk -F':' '{print $1}')  #2nd column is the destination IP and port
		DESTINATION_PORT=$(printf "%s" "${line}" | awk -F' ' '{print $2}' | awk -F':' '{print $2}')  #2nd column is the destination IP and port

		if [ "${TRANSMISSION_TYPE}" == "UDP" ]
		then
			# extract UDP TS file
			#tsp -I pcap "${PCAPFILE}" --source "${SOURCE_IP}":"${SOURCE_PORT}" --destination "${DESTINATION_IP}":"${DESTINATION_PORT}" -O file "${PCAPFILE}"."${TRANSMISSION_TYPE}"."${DESTINATION_IP}"."${DESTINATION_PORT}".ts # After tsduck version 3.36-3358 , you don't need --source anymore
[... cut ...]
			tsp -I pcap "${PCAPFILE}" --http  --destination "${DESTINATION_IP}":"${DESTINATION_PORT}" -O file "${PCAPFILE}"."${TRANSMISSION_TYPE}"."${DESTINATION_IP}"."${DESTINATION_PORT}".ts
		fi

	fi

done <<< "$(tspcap --list-streams "${PCAPFILE}" --no-pager | sed -e '1,2d' | sed 's/,//g' | sort -nrk 5 | head -n10)" # this lists largest 10 lines in the pcap analysis result
```

@lelegard  thank you so much for all your amazing effort. 

I will try all my PCAP files ( I have dozens of them ) and then if no critical issue is found, I will close this one.

Thank you.

closed 2023-11-09 by stulluk