tsduck/tsduck #1328: Can’t transmit on DVB-T2 on VATEK output plugin - Suntechtv U3 USB Modulator

WinSta0, 2023-10-24:
Bug description:

Trying to do a screen share with Open Broadcaster Software sending it to FFmpeg then to TSDuck.

Selecting DVB-T2 modulation with the tsp vatek output plugin and sending it to FFmpeg and TSDuck hangs and freezes up at around 30 seconds. Also, to note even before if freezes up no signal comes out at all. Tested this with different computers and still the same results.

How to reproduce:

Note this is a code snippet:

tsp -O vatek --bandwidth 7 --bandwidth-extension --device 0 --fft-mode 16K --frequency 585500000 --modulation DVB-T2 --pilot-pattern 7 --plp0-code-rate 5/6 --plp0-fec-type 64K --plp0-high-efficiency --plp0-modulation 256-QAM --plp0-rotation --t2-guard-interval 1/128 --t2-l1-modulation 64-QAM

This command does work however no signal is generated and FFmpeg and TSDuck freezes at around 30 seconds.

Expected behavior:

Should be able to transmit on DVB-T2 and get a signal on my TV / Set top box and FFmpeg and TSDuck should not to freeze up at all.

Errors and logs:

None: FFmpeg and TSDuck freezes up at around after 30 seconds. No DVB-T2 signal is generated at all.

Environment:

OS: Windows 10 21H2
TSDuck full version: 3.35-3258
Installation type: official binary portable

Additional information:

Besides DVB-T2 modulation also tested it with DVB-T and DVB-C (J83A). With DVB-T and DVB-C no problems were encountered. Everything went perfectly fine (this is with FFmpeg and TSDuck).

I also have the VATEK SDK installed on the computer and tried to transmit DVB-T2 with their application. No problems. Works perfectly fine. So therefore, at this current moment I cannot get a DVB-T2 signal generated from FFmpeg connected with TSDuck.

Would like to know if I am missing something or this is a TSDuck bug?

lelegard, 2023-10-24:
Hi @WinSta0,

Your tsp command only includes the vatek output plugin. Any other plugin? Do you simply pipe ffmpeg output into that tsp command with nothing else than the output plugin?

Richie, @a83912a, could you have a look at this one please?

WinSta0, 2023-10-24:
Thanks for your response.

Here are the commands I use:

Method 1)	
ffmpeg.exe -reconnect 1 -i http://127.0.0.1:8000/live/alex.flv -c:v copy -c:a ac3 -b:a 640k -ac 6 -channel_layout 5.1(side) -ar 48000 -mpegts_service_id 0x400 -streamid 0:0x0462 -streamid 1:0x0463 -muxrate 40000000 -f mpegts -listen 1 http://127.0.0.1:1234

The command above is FFmpeg grabbing a http FLV stream from a server and converting it to a mpegts http stream.

Method 2)	
ffmpeg.exe -re -stream_loop -1 -i file.mp4 -c:v copy -c:a ac3 -b:a 448k -ac 2 -channel_layout stereo -ar 48000 -mpegts_original_network_id 0x7f00 -mpegts_transport_stream_id 0x7f00 -streamid 0:0x0462 -streamid 1:0x0463 -program title=Cockatoo:program_num=0x400:st=0:st=1 -mpegts_flags +system_b -muxrate 40000000 -f mpegts -listen 1 http://127.0.0.1:1234

The command above is FFmpeg grabbing file.mp4 and converting it to a mpegts http stream.

3)	
tsp -v -m --bitrate 40000000 --buffer-size-mb 128 -I http http://127.0.0.1:1234 -P filter --negate -p 0 -s -P filter --negate -p 16 -P filter --negate -p 17 -s -P filter --negate -p 18 -P filter --negate -p 20 -s -P filter --negate -p 4096 -P inject -b 4000 -p 0 -s PAT.xml -P inject -b 4000 -p 16 -s NIT.xml -P inject -b 4000 -p 17 -s SDT.xml -P inject -b 4000 -p 18 -s EIT.xml --eit-normalization -P inject -b 4000 -p 20 -s TDT.xml -P inject -b 4000 -p 4096 -s PMT.xml -P timeref --start "2018/11/20:00:00:00" -P pcrverify -O vatek --bandwidth 7 --bandwidth-extension --device 0 --fft-mode 16K --frequency 585500000 --modulation DVB-T2 --pilot-pattern 7 --plp0-code-rate 5/6 --plp0-fec-type 64K --plp0-high-efficiency --plp0-modulation 256-QAM --plp0-rotation --t2-guard-interval 1/128 --t2-l1-modulation 64-QAM

The command above is TSDuck grabbing FFmpeg Method 1) or Method 2) NOT BOTH and sending it to vatek output for DVB-T2 modulation.

4)	
This is the exact commands that I use with FFmpeg and TSDuck. It dosent matter which FFmpeg method I use: Method 1) or Method 2) TSDuck and FFmpeg freezes.

WinSta0, 2023-10-25:
Hi @lelegard,

Thanks for your response.

Forgive me if I went too detail in my previous post. This time I will answer the questions in a general manner.

Your tsp command only includes the vatek output plugin. Any other plugin? Yes: http, filter, inject, timeref, pcrverify.

Do you simply pipe ffmpeg output into that tsp command with nothing else than the output plugin? No. The output plugins I use when I pipe ffmpeg into that tsp command are: http, filter, inject, timeref, pcrverify and of course vatek plugin.

lelegard, 2023-10-25:
When a problem occurs in the presence of multiple factors, the first thing to do is to eliminate those factors one by one, until the problem disappears, which identifies the problematic one. Alternatively, start with the simplest possible configuration and, assuming it works, add the other factors one by one, until the problem appears.

Therefore, I suggest to start with the simplest structure, a simple pipe with ffmpeg and tsp on the same machine, without any additional processing:
~~~
ffmpeg ... -f mpegts | tsp -O vatek ...
~~~

At a prerequisite, make sure that the requested muxrate in ffmpeg is _**exactly**_ the expected DVB-T2 bitrate, computed from your modulation parameters. 40000000 seems a bit too "round" to be credible. Use an online DVB bitrate calculator if you are unsure about the value.

If the simplest command does not work, this is probably an issue with the TSDuck vatek output plugin.

It it works, only add the intermediate HTTP service. If it works, add other plugins, one by one.

WinSta0, 2023-10-25:
Thanks for your response.

I have just already done a test with FFmpeg and TSDuck with the simplest structure, a simple pipe with ffmpeg and tsp on the same machine, without any additional processing and still the same results.

The FFmpeg muxrate has now been set to 40907500. This value comes from the original Vatek SDK application. Though I have tried with other values in the past before posting the bug in GitHub, I still get the same results (FFmpeg and TSDuck freezing up and no signal coming out at all). I have used multiple DVB bitrate calculator in the past.

It appears from what I have just tested that the simplest command does not work. I have tested this multiple times and still get the same results.

lelegard, 2023-10-25:
OK, so this must be a vatek plugin issue. @a83912a, could you please have a look ?

a83912a, 2023-10-25:
OK, I will test this issue.

a83912a, 2023-10-27:
Hi @WinSta0 ,
Can you please changing the DVB-T2 FFT to 8K and see if it successfully outputs? If all other modulations are working fine and only DVB-T2 is experiencing issues, it might be a problem with our plugin.

WinSta0, 2023-10-27:
Hi @a83912a,

Can you please changing the DVB-T2 FFT to 8K and see if it successfully outputs? Done that and got some good news, works perfectly fine with all the TSDuck plugins. Got signal coming out from the modulator and Video and Audio on the TV.

The 16K FFT however still does not work. I can Imagine that this option is malfunctioning. Probably plugin related. The VATEK SDK default application (vatek_toolkit.exe) can transmit in 16K FFT without any problems, so it’s definitely not the modulator.

a83912a, 2023-10-27:
@WinSta0 
OK, please give me some time to fix this issue. If the fix is completed and deployed, I will notify you. Thank you!

lelegard, 2023-11-29:
Richie, @a83912a, is there anything to fix in the TSDuck plugin? Or in the libvatek?

I plan a new release for December, maybe January. If you have a fix, I will wait for it. Please let me know.

a83912a, 2023-11-30:
Hi Thierry , @lelegard,  I plan to update libvatek to improve the stability and bugs in A3. There are currently no modifications needed for the TSDuck plugin. I anticipate the update to be released in mid-December. Thank you.

lelegard, 2023-11-30:
Thanks Richie, @a83912a. Let us know here when you publish the next binary release of libvatek so that the next release of TSDuck can embed the right version.

lelegard, 2023-12-13:
Hi Richie @a83912a. Does the recently released VATek SDK 3.10 solve the issue? Is this correct to assume that building a new release of TSDuck with libvatek 3.10 fixes the problem?

a83912a, 2023-12-13:
Hi Thierry @lelegard. Yes, this version has addressed the issue with DVB-T2 FFT 16K not working, and it includes some modifications to enhance the stability of A3. Thank you.

lelegard, 2023-12-13:
OK, great, I will probably release TSDuck version 3.36 within a week or so.

closed 2024-05-12 by github-actions[bot]