tsduck/tsduck #1329: [Q] how to extract mpeg-ts

atplsx, 2023-10-24:
please help me
`tsanalyze.exe 0000000100000002STR

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ............. Unknown  |  Services: .............. 0  |
|  Bytes: ....................... 291,848,192  |  PID's: Total: .......... 5  |
|  TS packets: .................... 1,552,384  |         Clear: .......... 0  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 5  |
|     With transport error: ............... 0  |         With PCR's: ..... 1  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 4  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ............... 26,078,944 b/s   28,298,429 b/s  |
|  Selected reference bitrate: ............. 26,078,944 b/s   28,298,429 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ................................... 89 sec (1 min 29 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 1,194, PID's: 1 (clear: 0, scrambled: 1)                       |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. S       20,058 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. S       20,058 b/s  |
|  0x0000  PAT .......................................... S       20,058 b/s  |
|=============================================================================|
|  Unreferenced PID's                                                         |
|  TS packets: 1,551,190, PID's: 4 (clear: 0, scrambled: 4)                   |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Unreferenced PID's ........................... S   26,058,886 b/s  |
|  0x1293  Unreferenced ................................. S       20,058 b/s  |
|  0x12F7  Unreferenced ................................. S   25,694,106 b/s  |
|  0x135B  Unreferenced ................................. S      266,235 b/s  |
|  0x13BF  Unreferenced ................................. S       78,486 b/s  |
===============================================================================

===============================================================================
|  PIDS ANALYSIS REPORT                                                       |
|=============================================================================|
|  PID: 0x0000 (0)                                                       PAT  |
|-----------------------------------------------------------------------------|
|  Global PID                Transport:                Discontinuities:       |
|  Bitrate: .... 20,058 b/s  Packets: ......... 1,194  Expected: ......... 0  |
|  Access: ...... Scrambled  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|  Crypto-Per: .... Unknown  Duplicated: .......... 0  Sections:              |
|  Inv.scramb.: ..... 1,194                            Unit start: ... 1,194  |
|=============================================================================|
|  PID: 0x1293 (4755)                                           Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate: .... 20,058 b/s  Packets: ......... 1,194  Expected: ......... 0  |
|  Access: ...... Scrambled  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|  Crypto-Per: .... Unknown  Duplicated: .......... 0  Sections:              |
|  Inv.scramb.: ..... 1,194                            Unit start: ... 1,194  |
|=============================================================================|
|  PID: 0x12F7 (4855)                                           Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate:  25,694,106 b/s  Packets: ..... 1,529,476  Expected: ......... 0  |
|  Access: ...... Scrambled  Adapt.F.: ........ 8,137  Unexpect: ......... 0  |
|  Crypto-Per: .... Unknown  Duplicated: .......... 0  Sections:              |
|  Inv.scramb.: . 1,529,476  TSrate: . 26,078,944 b/s  Unit start: ... 4,480  |
|                                                                             |
|  Clock values range:                                                        |
|  PCR: ............. 3,688                                                   |
|  from . 1,294,448,517,686                                                   |
|  to ... 1,296,864,661,050                                                   |
|  Leaps: ............... 0                                                   |
|=============================================================================|
|  PID: 0x135B (4955)                                           Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate: ... 266,235 b/s  Packets: ........ 15,848  Expected: ......... 0  |
|  Access: ...... Scrambled  Adapt.F.: .......... 310  Unexpect: ......... 0  |
|  Crypto-Per: .... Unknown  Duplicated: .......... 0  Sections:              |
|  Inv.scramb.: .... 15,848                            Unit start: ..... 310  |
|=============================================================================|
|  PID: 0x13BF (5055)                                           Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate: .... 78,486 b/s  Packets: ......... 4,672  Expected: ......... 0  |
|  Access: ...... Scrambled  Adapt.F.: .......... 465  Unexpect: ......... 0  |
|  Crypto-Per: .... Unknown  Duplicated: .......... 0  Sections:              |
|  Inv.scramb.: ..... 4,672                            Unit start: ..... 467  |
===============================================================================

===============================================================================
|  TABLES & SECTIONS ANALYSIS REPORT                                          |
===============================================================================`

i use this commands
tsp -I file 0000000100000002STR -P descrambler -c e7b8e75466530c3cf7238f577e18618a  -O file output.ts
* Error: descrambler: invalid control word "e7b8e75466530c3cf7238f577e18618a", specify 16 hexa digits
* Error: descrambler: specify either a service or a list of PID's

tsp -I file 0000000100000002STR -P descrambler -c e7b8e75466530c3cf7238f577e18618a  -p 4855 -O file output.ts
* Error: descrambler: invalid control word "e7b8e75466530c3cf7238f577e18618a", specify 16 hexa digits
* Error: descrambler: specify control words to descramble an explicit list of PID's

i have aes-key
0x2fe17e8:      0xe7    0xb8    0xe7    0x54    0x66    0x53    0x0c    0x3c
0x2fe17f0:      0xf7    0x23    0x8f    0x57    0x7e    0x18    0x61    0x8a

lelegard, 2023-10-24:
By default, the descrambler plugins uses the DVB-CSA2 encryption algorithm, which takes a 64-bit key (16 hex digits).

It seems that you provide a 128-bit key, probably an AES key.

The first thing you need to know is the kind of encryption you want to descramble. AES-128, ok, but which one at TS packet level? The descrambler plugin supports AES-CBC, AES-CTR, ATIS-IDSA, DVB-CISSA. All of them are AES-128 but with different modes. There is an option for each of them in the descrambler plugin. With AES-CBC and AES-CTR, be sure to know the initialization vector first (option `--iv`). RTFM for the details.

atplsx, 2023-10-24:
I don't know the -iv value, how can I find it, which of these values ​​could it be? Can you help me?
If I need to specify the pid, which one should I specify?
0x2fe17e8:      0xe7    0xb8    0xe7    0x54    0x66    0x53    0x0c    0x3c
0x2fe17f0:      0xf7    0x23    0x8f    0x57    0x7e    0x18    0x61    0x8a
I actually do this with this program https://github.com/openlgtv/epk2extract, but since the records are divided into pieces, there are corruptions when combining them with ffmpeg concat. That's why I wanted to try this. In this program, the only value is e7b8e75466530c3cf7238f577e18618a I enter this.

atplsx, 2023-10-24:
tsp -I file 0000000100000002STR -P descrambler --aes-cbc e7b8e75466530c3cf7238f577e18618a -O file output.ts
tsp -I file 0000000100000002STR -P descrambler --aes-ctr e7b8e75466530c3cf7238f577e18618a -O file output.ts
tsp -I file 0000000100000002STR -P descrambler --atis-idsa e7b8e75466530c3cf7238f577e18618a -O file output.ts
sp -I file 0000000100000002STR -P descrambler --dvb-cissa e7b8e75466530c3cf7238f577e18618a -O file output.ts
I tried these but the resulting file still contains passwords because it cannot be read.

atplsx, 2023-10-24:
extracted mpeg-ts analyse
===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ...... 43104 (0xA860)  |  Services: .............. 2  |
|  Bytes: ....................... 298,057,268  |  PID's: Total: ......... 12  |
|  TS packets: .................... 1,585,411  |         Clear: .......... 6  |
|     With invalid sync: .......... 1,546,332  |         Scrambled: ...... 6  |
|     With transport error: ......... 776,722  |         With PCR's: ..... 4  |
|     Suspect and ignored: ............ 3,705  |         Unreferenced: ... 6  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ...................... Unknown          Unknown  |
|  Selected reference bitrate: .................... Unknown             None  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ................................................. Unknown  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0001  (unknown) .................................... C          Unknown  |
|  0x3FAC  (unknown) .................................... S          Unknown  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 27, PID's: 1 (clear: 1, scrambled: 0)                          |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C          Unknown  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C          Unknown  |
|  0x0000  PAT .......................................... C          Unknown  |
|=============================================================================|
|  Unreferenced PID's                                                         |
|  TS packets: 6, PID's: 6 (clear: 1, scrambled: 5)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Unreferenced PID's ........................... S          Unknown  |
|  0x02EA  Unreferenced ................................. S          Unknown  |
|  0x0BC9  Unreferenced ................................. S          Unknown  |
|  0x0C5B  Unreferenced ................................. S          Unknown  |
|  0x0DCD  Unreferenced ................................. C          Unknown  |
|  0x0EDB  Unreferenced ................................. S          Unknown  |
|  0x1613  Unreferenced ................................. S          Unknown  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0xA860 (43104), Original Netw: 0x0000 (0)         |
|  Service name: (unknown), provider: (unknown)                               |
|  Service type: 0x00 (Undefined)                                             |
|  TS packets: 1, PID's: 2 (clear: 2, scrambled: 0)                           |
|  PMT PID: 0x00B1 (177), PCR PID: 0x1F00 (7936)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Undefined .................................... C          Unknown  |
|  0x00B1  PMT .......................................... C          Unknown  |
|  0x1F00  MPEG-2 Audio ................................. C          Unknown  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
|=============================================================================|
|  Service: 0x3FAC (16300), TS: 0xA860 (43104), Original Netw: 0x0000 (0)     |
|  Service name: (unknown), provider: (unknown)                               |
|  Service type: 0x00 (Undefined)                                             |
|  TS packets: 32,315, PID's: 4 (clear: 3, scrambled: 1)                      |
|  PMT PID: 0x1293 (4755), PCR PID: 0x12F7 (4855)                             |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Undefined .................................... S          Unknown  |
|  0x1293  PMT .......................................... C          Unknown  |
[... 153 lines cut ...]
[... cut ...]
|  Service: 0x3FAC (16300) (unknown)                                          |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x3FAC (16300)                                     |
|      Repetition: .. 45204  pkt  Section cnt: ....... 28                     |
|      Min repet.: ... 3984  pkt  Table cnt: ......... 28                     |
|      Max repet.: . 260304  pkt  Version: ............ 9                     |
===============================================================================

closed 2023-11-07 by atplsx