tsduck/tsduck #1330: [Q]  Seeking Software-based Stream Validation Alternatives to StreamXpert

popperwin, 2023-10-24:
Hi everyone,

I've been using TSDuck to distribute streams encoded with ffmpeg and overall, it's been working well for me. However, I've received feedback from some entities I distribute to, who vaguely state my streams "do not conform to standards". Yet, they decode these streams without any visual artifacts. Digging deeper, I learned they're using probes like StreamXpert to validate these streams.

I'm keen on addressing this, and I'd like to validate my streams using a similar system to ensure the quality of my feeds. It's worth noting that I do not integrate any Windows systems or Dektec hardware (associated with StreamXpert) in my setup, and I plan to keep it that way.

I'm looking for a fully software-based solution, be it free or paid, for this validation. A SaaS or "Pay as you go" model would be perfect, especially since I run my encoder on AWS. But so far, I haven't found any fitting AWS solutions.

Any suggestions or tools would be greatly appreciated.

Thanks in advance!

popperwin, 2023-10-25:
Thanks for your help.
I have asked them and they told me that my feed is not CBR.
However, using mediainfo  on a .ts extracted, I can see it's CBR.

Here is the transcoding command:

`ffmpeg -fflags nobuffer -i - -vcodec h264_nvenc -flags cgop -cq 1 -rc cbr -minrate 6M -bitrate 6M -maxrate 6M -bufsize 12M -r 25 -g 50 -strict_gop 1 -no-scenecut 1 -sc_threshold 0 -force_key_frames "expr:gte(t,n_forced*2)"" -forced-idr 1 -threads 0 -preset slow -profile:v high -gpu:v 1 -tune hq -zerolatency 1 -acodec libfdk_aac -af aresample=async=1 -b:a 320k -muxrate 7500000 -flush_packets 1 -f mpegts -|tsp --receive-timeout 20000 --verbose --timed-log --max-flushed-packets 7 --max-input-packets 7 --initial-input-packets 7 -I file -O ip 224.0.0.1:5216`

Mediainfo is giving me this output:
```
General
Complete name                            : cbr_or_vbr.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 45.4 MiB
Duration                                 : 1 min 0 s
Overall bit rate mode                    : Constant
Overall bit rate                         : 6 327 kb/s
Writing application                      : Lavf60.3.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4
Format settings                          : CABAC / 2 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 2 frames
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 1 min 0 s
Source duration                          : 1 min 0 s
Bit rate mode                            : Constant
Bit rate                                 : 6 014 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 25.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.116
Stream size                              : 41.2 MiB (91%)
Source stream size                       : 43.0 MiB (95%)
Codec configuration box                  : avcC

Audio
ID                                       : 2
Format                                   : AAC LC
Format/Info                              : Advanced Audio Codec Low Complexity
Codec ID                                 : mp4a-40-2
Duration                                 : 1 min 0 s
Bit rate mode                            : Constant
Bit rate                                 : 320 kb/s
Channel(s)                               : 2 channels
Channel layout                           : L R
Sampling rate                            : 48.0 kHz
Frame rate                               : 46.875 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 2.29 MiB (5%)
Default                                  : Yes
Alternate group                          : 1

```
My guess is that the mpegts container is somehow received in a way that StreamXpert thinks its VBR. However I have also noticed that the SRT logs never report the same number of transmitted packets each interval:

Command:
`tsp --receive-timeout 20000 --verbose --timed-log  -I ip 224.0.0.1:5216 -P continuity --fix  -P pcrbitrate -P regulate -O srt  --latency 2000 --caller X.X.X.X:X  --statistics-interval 10000 --passphrase XXXXXX --peer-idle-timeout 1000`  

Logs:
```
* 2023/10/25 12:08:40 - srt: SRT statistics:
  Total sent: 9,764,580 bytes, 7,504 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
  Interval sent: 9,764,580 bytes, 7,504 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
* 2023/10/25 12:08:50 - srt: SRT statistics:
  Total sent: 19,430,604 bytes, 15,054 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
  Interval sent: 9,666,024 bytes, 7,550 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
* 2023/10/25 12:09:00 - srt: SRT statistics:
  Total sent: 29,165,132 bytes, 22,533 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
  Interval sent: 9,734,528 bytes, 7,479 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
* 2023/10/25 12:09:10 - srt: SRT statistics:
  Total sent: 38,960,620 bytes, 30,184 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
  Interval sent: 9,795,488 bytes, 7,651 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
* 2023/10/25 12:09:20 - srt: SRT statistics:
  Total sent: 48,637,484 bytes, 37,630 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
  Interval sent: 9,676,864 bytes, 7,446 packets, retransmit: 0 packets, lost: 0 packets, dropped: 0 packets
  ```
  
 But when they send me a SRT feed they make, I have always the same number of packets at each interval. 
 I have tried to use tsstuff to force CBR on the mpegts container level, but without any success.

lelegard, 2023-10-25:
> the SRT logs never report the same number of transmitted packets each interval:

Not that much. The numbers of reported packets per 10 s. are 7504, 7550, 7479, 7651, 7446. The max diff is 147, with a smaller diff around 50 in most cases. With an average of 753 packets per second, the differences are equivalent to 0.06 to 0.2 seconds. The report interval is never exactly 10 seconds because this is processed and displayed by a low priority thread. Thus, the differences remain in the interval of uncertainty.

Do you really need to go through UDP between the encoding command `ffmpeg | tsp -O ip` and the send command `tsp -I ip -O srt`? The problem with UDP is that packet loss is included and possible between the two. Why not doing `ffmpeg | tsp -O srt` in one command?

The most important problem I see is the encoding command. I don't know what is the input for ffmpeg. If this is a file or any other unregulated input, you will have a problem since the encoding process is probably faster than the playout speed. Thus, packets will be sent too fast over UDP and packets will probably be lost between the encoding and send commands.

The question is: is this a live stream (to be transmitted at playout speed) or a media content transfer (independent speeds)? In the former case, I recommend to add `-P regulate --packet-burst 7` in the encoding command, before `-O ip`. In the latter case, you must use  `ffmpeg | tsp -O srt` in one single command to avoid packet loss.

popperwin, 2023-10-25:
Thanks for your reply.

Yes I really need to go UDP first, it's the backbone of the architecture - the transcoded feed needs to be avalaible in multicast.

I have already seen that ffmpeg have a lot of issues with UDP, and yes I understand the issues with packet losses.
The input of ffmpeg is another live multicast feed, received with `tsp -I srt -O ip`  

I have experienced `ffmpeg | tsp -O srt`  and they told me that it was more constant, but still not perfect according to their probe.
They are telling that the underlying H264 is not CBR; however `mediainfo`says the contrary, and I can't understand how this transcoding command cannot produce a CBR feed. 

I will continue experiments things, thanks for your help . And if you have a mpegts probe, used by broadcaster to recommend I would be very grateful :)

lelegard, 2023-10-25:
> They are telling that the underlying H264 is not CBR; however mediainfosays the contrary

Mediainfo is well-known and usually reliable. Your tools against their tools. No longer a technical problem. Good luck...

popperwin, 2023-12-19:
Thanks for your help.

After a lot of tests, it seems like doing `-P regulate --packet-burst 14 --pcr-synchronous` is more "compliant" with their tool, and the feed appears to be CBR according to them.

But sometimes the logs on their side shows a lot of CC and PCR errors. 

So I'm trying to avoid this using this command , like this : `-P pcrbitrate -P regulate -P continuity --fix -P pcradjust -P regulate --packet-burst 14 --pcr-synchronous`

Full command is : 

```tsp --receive-timeout 5000 --timed-log --verbose --max-flushed-packets 7 --max-input-packets 7 --initial-input-packets 7 -I ip 224.0.0.1:5217 -P pcrbitrate -P regulate -P continuity --fix -P pcradjust -P regulate --packet-burst 14 --pcr-synchronous -O srt --packet-burst 7 --enforce-burst 7 --latency 1000 --caller X.X.X.X:33990 --statistics-interval 60000```

The idea is to regulate according to feed bitrate before fixing continuity. Once continuity error have been fixed, the idea is to adjust PCR according to the (constant) bitrate found by the first regulate plugin.  Once PCR have been adjusted, I can use "safely" a second regulate plugin with   --pcr-synchronous for the final transmission.  

At the moment the result seems to be more "compliant" with their tool, however I'm not completely sure about the command , hence I'm asking if there is something wrong between what I want to do and what the command is exactly doing.

closed 2024-05-19 by github-actions[bot]